coolsnowwolf/lede #6387: 编译不出v2raycore

qd201211, 2021-02-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：luci-app-ssr-plus勾选v2ray(非xray)且编译完成后发现并没有v2raycore.ipk，固件安装后在添加节点时也没有v2ray的选项
      期间编译无任何报错，关闭goproxy或者开启更换为阿里云和goproxy.io都不报错也不编译。project V里面的v2ray plugin已经 
     勾选，xray-core取消勾选。问题是从goproxy失效,lean更新代码后出现的，不知道是不是把v2ray彻底删掉了？求大佬解答

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：

AmadeusGhost, 2021-02-17:
已经在 https://github.com/coolsnowwolf/lede/commit/51a94cb899b6d2a262003a3d94801672dc6e1c58 这里删掉了

qd201211, 2021-02-17:
> 已经在 https://github.com/coolsnowwolf/lede/commit/51a94cb899b6d2a262003a3d94801672dc6e1c58 这里删掉了

早上看到了，但是我14号开的虚拟机clone的编译出来就有v2core，真是奇怪😅。谢谢啦

souwei168, 2021-02-17:
怪不得编译不了 PASSWALL,提示缺少 V2RAY，怎么才能加上去？

qd201211, 2021-02-17:
> 怪不得编译不了 PASSWALL,提示缺少 V2RAY，怎么才能加上去？
找以前的版本，package/lean/v2ray把他复制粘贴到当前的源码里面，然后make menuconfig里面勾上v2ray就行了

closed 2021-06-17 by coolsnowwolf