coolsnowwolf/lede #6388: 这个编译出错!请问是什么问题？如何解决？谢谢

wlsk888, 2021-02-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：make[3] -C target/linux install
make -r world: build failed

### (2) 路由器型号和固件版本

A：极路由3-7260-5861

### (3) 详细日志

A：make[3] -C feeds/luci/applications/luci-app-ddns compile
 make[3] -C package/lean/default-settings compile
 make[3] -C package/lean/luci-app-autoreboot compile
 make[3] -C package/lean/luci-lib-fs compile
 make[3] -C package/lean/luci-app-filetransfer compile
 make[3] -C package/kernel/mt76 compile
 make[2] package/install
 make[2] target/install
 make[3] -C target/linux install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
**make: *** [world] Error 1**
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/libubox host-compile
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb /workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb.tmp
../dts/mt7620a_youku_yk1.dts:79.3-22: ERROR (duplicate_property_names): /palmbus@10000000/spi@b00/flash@0:broken-flash-reset: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
Makefile:201: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb' failed
make[5]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb] Error 2
make[5]: Leaving directory '/workdir/openwrt/target/linux/ramips/image'
make[4]: *** [install] Error 2
Makefile:25: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[4]: Leaving directory '/workdir/openwrt/target/linux/ramips'
make[2]: *** [target/linux/install] Error 2
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#44.74#12.39#54.78
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make: *** [world] Error 2
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.
[1_build.txt](https://github.com/coolsnowwolf/lede/files/5993767/1_build.txt)

4kerSoo, 2021-02-17:
../dts/mt7620a_youku_yk1.dts:79.3-22: ERROR (duplicate_property_names): /palmbus@10000000/spi@b00/flash@0:broken-flash-reset: Duplicate property name
[/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb] Error 2
删掉一行broken-flash-reset

AmadeusGhost, 2021-02-19:
https://github.com/coolsnowwolf/lede/commit/90bd81e0991046927634bf74830ec1c835d83e85 这边 `broken-flash-reset;` 重复了，去掉一个就好

PR #6408 (2021-02-21): Revert "Add broken-flash-reset flag to Youku YK-L1"
This reverts commit 90bd81e0991046927634bf74830ec1c835d83e85.

The commit 90bd81e is a duplicate of b132179.

Fixes: #6388
Fixes: #6406
commit b484f2ee (2021-02-21): Revert "Add broken-flash-reset flag to Youku YK-L1" (#6408)

This reverts commit 90bd81e0991046927634bf74830ec1c835d83e85.

The commit 90bd81e099 is a duplicate of b132179c89.

Fixes: #6388
Fixes: #6406
closed 2021-02-21 by coolsnowwolf