coolsnowwolf/lede #6389: 大雕的 PLUS+ ,和PASSWALL ，不能同时编译，不然就无法编译成功，就是最后封包时，报错，有没有解决办法？

souwei168, 2021-02-17:
反馈bug/问题模板，提建议请删除

最近发现一个问题，大雕的 PLUS+ ,和PASSWALL ，不能同时编译，不然就无法编译成功，就是最后封包时，报错，有没有解决办法？

Q：是否搜索了issue (使用 "x" 选择)
 [ x] 没有类似的issue

2. 详细叙述
大雕的 PLUS+ ,和PASSWALL ，不能同时编译，不然就无法编译成功，就是最后封包时，报错，有没有解决办法？

 (2) 路由器型号和固件版本

A：X86

 (3) 详细日志

A：

souwei168, 2021-02-17:
只能是，要不就只编译 PULX+,能成功，要不就编译 PASSWALL能成功，同时一起编译就不能成功，都是最后一布封包时报错！

camino13, 2021-02-17:
那就不一起编译呗，反正你只能用一个

WYC-2020, 2021-02-17:
咋不行，我就是同时编译

WYC-2020, 2021-02-17:
自己fork改冲突

souwei168, 2021-02-17:
谢谢，改哪里，求指导！

Myoko, 2021-02-17:
一直两个一起没问题

souwei168, 2021-02-17:
我测试过好几次，你拉最新的源码测试一下！

Ljzkirito, 2021-02-17:
我的就一起编译的，没有问题
[Github Actions](https://github.com/Ljzkirito/Actions-Lean-R2S)
是不是v2ray-plugin与Xray冲突？

czxinshui, 2021-02-17:
Include Shadowsocks V2ray Plugin  这个不要选中试试

souwei168, 2021-02-18:
> 包括Shadowsocks V2ray插件这个不要替换试试
没用，同样编译不出来
 make[3] -C feeds/small/trojan-plus compile
 make[3] -C package/pack/1/openwrt-passwall/trojan compile
 make[3] -C package/lean/rblibtorrent compile
 make[3] -C feeds/kenzo/luci-app-passwall compile
 make[3] -C feeds/kenzo/luci-app-ssr-plus compile
 make[3] -C feeds/kenzo/luci-app-vssr compile
 make[3] -C package/lean/qBittorrent compile
 make[3] -C package/lean/luci-app-qbittorrent compile
 make[3] -C package/lean/luci-app-kodexplorer compile
 make[3] -C package/lean/luci-app-amule compile
 make[2] package/install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/sw/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

souwei168, 2021-02-18:
make[3] -C feeds/small/trojan-plus compile
 make[3] -C package/pack/1/openwrt-passwall/trojan compile
 make[3] -C package/lean/rblibtorrent compile
 make[3] -C feeds/kenzo/luci-app-passwall compile
 make[3] -C feeds/kenzo/luci-app-ssr-plus compile
 make[3] -C feeds/kenzo/luci-app-vssr compile
 make[3] -C package/lean/qBittorrent compile
 make[3] -C package/lean/luci-app-qbittorrent compile
 make[3] -C package/lean/luci-app-kodexplorer compile
 make[3] -C package/lean/luci-app-amule compile
 make[2] package/install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/sw/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

binge8, 2021-02-18:
这两个一直同时编译，从不失败

souwei168, 2021-02-18:
> 这两个一直同时编译，从不失败

你DIY用了哪些插件，能不能发一下，看看我是不是用的不一样？

binge8, 2021-02-18:
> > 这两个一直同时编译，从不失败
> 
> 你DIY用了哪些插件，能不能发一下，看看我是不是用的不一样？

https://github.com/binge8/op

treisrunut, 2021-02-21:
两个都要的话 推荐你用 https://github.com/immortalwrt/immortalwrt 
原 project-openwrt CTCGFW 
这个源码编译 可以 passwall ssrplus OpenClash 一起加 。或者你参考那边的共用的处理方式

closed 2021-02-21 by coolsnowwolf