coolsnowwolf/lede #6394: 大雕的 PLUS+ ,和PASSWALL ，不能同时编译，不然就无法编译成功

souwei168, 2021-02-17:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志
 make[1] world
 make[2] target/compile
 make[3] -C target/linux compile
 make[2] buildinfo
 make[3] diffconfig buildversion feedsversion
 make[2] package/cleanup
 make[2] package/compile
 make[3] -C package/libs/libjson-c host-compile
 make[3] -C package/libs/libubox host-compile
 make[3] -C package/system/opkg host-compile
 make[3] -C package/libs/toolchain compile
 make[3] -C package/libs/libnl-tiny compile
 make[3] -C package/libs/libjson-c compile
 make[3] -C package/utils/lua compile
 make[3] -C package/libs/libubox compile
 make[3] -C package/system/ubus compile
 make[3] -C package/system/uci compile
 make[3] -C package/network/config/netifd compile
 make[3] -C package/firmware/linux-firmware compile
 make[3] -C package/firmware/prism54-firmware compile
 make[3] -C package/kernel/linux compile
 make[3] -C package/system/ubox compile
 make[3] -C package/libs/ncurses host-compile
 make[3] -C package/libs/zlib compile
 make[3] -C package/libs/ncurses compile
 make[3] -C package/utils/util-linux compile
 make[3] -C package/system/fstools compile
 make[3] -C package/system/fwtool host-compile
 make[3] -C package/system/fwtool compile
 make[3] -C package/system/procd compile
 make[3] -C package/system/usign host-compile
 make[3] -C package/system/ucert host-compile
 make[3] -C package/utils/jsonfilter compile
 make[3] -C package/system/openwrt-keyring compile
 make[3] -C package/system/usign compile
 make[3] -C package/base-files compile
 make[3] -C package/boot/grub2 host-compile
 make[3] -C package/boot/grub2 compile
 make[3] -C package/boot/grub2 compile
 make[3] -C package/libs/gettext compile
 make[3] -C package/libs/libiconv compile
 make[3] -C package/devel/binutils compile
 make[3] -C feeds/packages/lang/golang/golang host-compile
 make[3] -C package/system/ca-certificates compile
 make[3] -C feeds/helloworld/xray-core compile
 make[3] -C feeds/packages/libs/libmaxminddb compile
 make[3] -C feeds/kenzo/lua-maxminddb compile
 make[3] -C package/libs/mbedtls compile
 make[3] -C package/kernel/cryptodev-linux compile
 make[3] -C package/libs/openssl compile
 make[3] -C package/libs/ustream-ssl compile
 make[3] -C package/libs/uclient compile
[... 317 lines cut ...]
 make[3] -C package/lean/luci-app-vlmcsd compile
 make[3] -C package/lean/vsftpd-alt compile
 make[3] -C package/lean/luci-app-vsftpd compile
 make[3] -C package/lean/luci-app-webadmin compile
 make[3] -C package/lean/luci-app-xlnetacc compile
 make[3] -C package/lean/luci-app-zerotier compile
 make[3] -C package/lean/luci-lib-docker compile
 make[3] -C package/lean/luci-proto-bonding compile
 make[3] -C package/lean/r8125 compile
 make[3] -C package/lean/r8168 compile
 make[3] -C package/network/services/dropbear compile
 make[3] -C package/network/utils/iperf compile
 make[3] -C package/pack/luci-app-jd-dailybonus compile
 make[3] -C package/system/mtd compile
 make[3] -C package/system/opkg compile
 make[3] -C package/system/urandom-seed compile
 make[3] -C package/system/urngd compile
 make[3] -C package/utils/busybox compile
 make[3] -C package/utils/f2fs-tools compile
 make[2] package/install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/sw/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

A：

WYC-2020, 2021-02-18:
瞎提，不知道已经有人提了？

gsblay, 2021-02-18:
一女二嫁！晕！！！

souwei168, 2021-02-18:
看不懂什么意思 ，能不能解决我的问题呢，这次是把编译输出，给贴出来了，有没有大神能看出是哪里问题？

souwei168, 2021-02-18:
次次都是到这个封包这里，过不去了，不知道是哪里出问题？make[2] package/install
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
/home/sw/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 1

ismethr, 2021-02-19:
#Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on# 这不写的很清楚吗！话说你为什么非要两个都用

closed 2021-02-21 by coolsnowwolf

souwei168, 2021-02-24:
谢谢，已经解决 是大雕，删除 V2RAY源码造成的，编译时，去掉V2RAY，就好，改用X2RAY!