coolsnowwolf/lede #6397: 编译qbtorrent出错，还望大神们不吝赐教，感激不尽！

trombonist852, 2021-02-18:
编译树莓派4b 64位，提示找不到boost的版本 但是在menuconfig里面选中boost之后 还是会跳错误，这个如何破？望大神们帮忙看看 谢谢啦！

这是没选boost之前的错误：
checking for Qt5 qmake >= 5.9.0... /home/trb/21.02/staging_dir/toolchain-aarch64_cortex-a72_gcc-8.4.0_musl/bin/qmake
checking whether QtDBus should be enabled... no
checking for boostlib >= 1.40 (104000) includes in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/include"... yes
checking for boostlib >= 1.40 (104000) lib path in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib/aarch64-linux-gnu"... no
checking for boostlib >= 1.40 (104000) lib path in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib64"... no
checking for boostlib >= 1.40 (104000) lib path in "/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib"... yes
checking for boostlib >= 1.40 (104000)... yes
configure: Boost CXXFLAGS: "-I/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/include"
configure: Boost LDFLAGS: "-L/home/trb/21.02/staging_dir/target-aarch64_cortex-a72_musl/usr/lib"
checking whether the Boost::System library is available... yes
configure: error: Could not find a version of the Boost::System library!
make[2]: *** [Makefile:67: /home/trb/21.02/build_dir/target-aarch64_cortex-a72_musl/qBittorrent-release-4.3.1/.configured_68b329da9893e34099c7d8ad5cb9c940] Error 1
make[2]: Leaving directory '/home/trb/21.02/package/lean/qBittorrent'
time: package/lean/qBittorrent/compile#4.01#0.17#4.09
    ERROR: package/lean/qBittorrent failed to build.
make[1]: *** [package/Makefile:114: package/lean/qBittorrent/compile] Error 1
make[1]: Leaving directory '/home/trb/21.02'
make: *** [/home/trb/21.02/include/toplevel.mk:230：package/qBittorrent/compile] 错误 2

sxx1314, 2021-02-18:
apt install libboost-system-dev

trombonist852, 2021-02-18:
> apt install libboost-system-dev

感谢大神指教 我运行了显示libboost-system-dev已经是最新版了....

sxx1314, 2021-02-18:
> > apt install libboost-system-dev
> 
> 感谢大神指教 我运行了显示libboost-system-dev已经是最新版了....

rm -rf tmp
make clean

trombonist852, 2021-02-18:
> > > apt install libboost-system-dev
> > 
> > 
> > 感谢大神指教 我运行了显示libboost-system-dev已经是最新版了....
> 
> rm -rf tmp
> make clean
哈哈哈，这些我都执行过啦，我刚才重新拉了一遍源码 再试试看。。。谢谢大神！

jinjin327, 2021-02-18:
一样的问题

trombonist852, 2021-02-22:
这是最新编译提示的错误，望大神能帮忙分析一下  谢谢啦
/home/trb/mainn1/staging_dir/toolchain-aarch64_cortex-a53_gcc-8.4.0_musl/aarch64-openwrt-linux-musl/include/c++/8.4.0/memory:116:58: error: cast from 'void*' to 'uintptr_t' {aka 'unsigned int'} loses precision [-fpermissive]
   const auto __intptr = reinterpret_cast<uintptr_t>(__ptr);
                                                          ^
In file included from /usr/include/boost/asio/detail/throw_error.hpp:19,
                 from /usr/include/boost/asio/detail/impl/posix_tss_ptr.ipp:23,
                 from /usr/include/boost/asio/detail/posix_tss_ptr.hpp:76,
                 from /usr/include/boost/asio/detail/tss_ptr.hpp:27,
                 from /usr/include/boost/asio/detail/call_stack.hpp:20,
                 from /usr/include/boost/asio/detail/thread_context.hpp:20,
                 from /usr/include/boost/asio/detail/recycling_allocator.hpp:20,
                 from /usr/include/boost/asio/detail/handler_alloc_helpers.hpp:21,
                 from /usr/include/boost/asio/detail/executor_op.hpp:20,
                 from /usr/include/boost/asio/impl/system_executor.hpp:18,
                 from /usr/include/boost/asio/system_executor.hpp:129,
                 from /usr/include/boost/asio/associated_executor.hpp:21,
                 from /usr/include/boost/asio/detail/bind_handler.hpp:20,
                 from /usr/include/boost/asio/detail/wrapped_handler.hpp:18,
                 from /usr/include/boost/asio/io_context.hpp:24,
                 from /usr/include/boost/asio/detail/io_object_executor.hpp:21,
                 from /usr/include/boost/asio/detail/io_object_impl.hpp:20,
                 from /usr/include/boost/asio/basic_socket.hpp:21,
                 from /usr/include/boost/asio/basic_socket_acceptor.hpp:19,
                 from /usr/include/boost/asio/ip/tcp.hpp:19,
                 from /home/trb/mainn1/staging_dir/target-aarch64_cortex-a53_musl/usr/include/libtorrent/socket.hpp:53,
                 from /home/trb/mainn1/staging_dir/target-aarch64_cortex-a53_musl/usr/include/libtorrent/add_torrent_params.hpp:45,
                 from ./base/bittorrent/session.h:36,
                 from app/application.cpp:65:
/usr/include/boost/system/error_code.hpp: In function 'std::size_t boost::system::hash_value(const boost::system::error_code&)':
/usr/include/boost/system/error_code.hpp:787:55: error: cast from 'const boost::system::error_category*' to 'uintptr_t' {aka 'unsigned int'} loses precision [-fpermissive]
         id = reinterpret_cast<boost::uintptr_t>( &cat );
                                                       ^
In file included from ./base/bittorrent/session.h:38,
                 from app/application.cpp:65:
/home/trb/mainn1/staging_dir/target-aarch64_cortex-a53_musl/usr/include/libtorrent/torrent_handle.hpp: In member function 'uint32_t libtorrent::torrent_handle::id() const':
/home/trb/mainn1/staging_dir/target-aarch64_cortex-a53_musl/usr/include/libtorrent/torrent_handle.hpp:1263:70: error: cast from 'std::__shared_ptr<libtorrent::torrent, (__gnu_cxx::_Lock_policy)2>::element_type*' {aka 'libtorrent::torrent*'} to 'uintptr_t' {aka 'unsigned int'} loses precision [-fpermissive]
    uintptr_t ret = reinterpret_cast<uintptr_t>(m_torrent.lock().get());
                                                                      ^
make[4]: *** [Makefile:9680: application.o] Error 1
make[4]: Leaving directory '/home/trb/mainn1/build_dir/target-aarch64_cortex-a53_musl/qBittorrent-release-4.3.1/src'
make[3]: *** [Makefile:47: sub-src-make_first] Error 2
make[3]: Leaving directory '/home/trb/mainn1/build_dir/target-aarch64_cortex-a53_musl/qBittorrent-release-4.3.1'
make[2]: *** [Makefile:66: /home/trb/mainn1/build_dir/target-aarch64_cortex-a53_musl/qBittorrent-release-4.3.1/.built] Error 2
make[2]: Leaving directory '/home/trb/mainn1/package/lean/qBittorrent'
time: package/lean/qBittorrent/compile#6.81#0.39#7.63
    ERROR: package/lean/qBittorrent failed to build.
make[1]: *** [package/Makefile:114: package/lean/qBittorrent/compile] Error 1
make[1]: Leaving directory '/home/trb/mainn1'
make: *** [/home/trb/mainn1/include/toplevel.mk:230：package/luci-app-qbittorrent/compile] 错误 2

hcym, 2021-02-22:
下午编译了2711的64位，没问题，无线也有，不过插件很少，只有ssrp和haproxy

AmadeusGhost, 2021-02-24:
```
--- a/package/lean/qBittorrent/Makefile
+++ b/package/lean/qBittorrent/Makefile
@@ -25,7 +25,7 @@ define Package/qBittorrent
   CATEGORY:=Network
   TITLE:=bittorrent client programmed in C++ / Qt
   URL:=https://www.qbittorrent.org/
-  DEPENDS:=+qt5-core +qt5-network +qt5-sql +qt5-xml +rblibtorrent       
+  DEPENDS:=+boost +qt5-core +qt5-network +qt5-sql +qt5-xml +rblibtorrent
 endef

 define Package/qBittorrent/description
```
照着改下 qBittorrent 的 makefile 再编译试试

trombonist852, 2021-02-26:
> ```
> --- a/package/lean/qBittorrent/Makefile
> +++ b/package/lean/qBittorrent/Makefile
> @@ -25,7 +25,7 @@ define Package/qBittorrent
>    CATEGORY:=Network
>    TITLE:=bittorrent client programmed in C++ / [...]

感谢大神！ 按照您的提示修改好之后还是会提示错误：（依赖的最后一个包如果是rblibtorrent会提示找不到这个包，所以我改成了libtorrent-rasterbar，其他都是按照您的提示改的）
/home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/x86_64-openwrt-linux-musl/include/c++/8.4.0/ext/string_conversions.h:99:5: note: candidate: 'template<class _String, class _CharT> _String __gnu_cxx::__to_xstring(int (*)(_CharT*, std::size_t, const _CharT*, __va_list_tag*), std::size_t, const _CharT*, ...)'
     __to_xstring(int (*__convf) (_CharT*, std::size_t, const _CharT*,
     ^~~~~~~~~~~~
/home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/x86_64-openwrt-linux-musl/include/c++/8.4.0/ext/string_conversions.h:99:5: note:   template argument deduction/substitution failed:
In file included from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/x86_64-openwrt-linux-musl/include/c++/8.4.0/string:52,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qbytearray.h:52,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qstring.h:49,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qhashfunctions.h:44,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qlist.h:47,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qhash.h:46,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qshareddata.h:46,
                 from /home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qsharedpointer.h:45,
                 from ../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/qpointer.h:43,
                 from ../../../../staging_dir/toolchain-x86_64_gcc-8.4.0_musl/include/QtCore/QPointer:1,
                 from app/application.h:32,
                 from app/application.cpp:30:
/home/trb/mainx86/staging_dir/toolchain-x86_64_gcc-8.4.0_musl/x86_64-openwrt-linux-musl/include/c++/8.4.0/bits/basic_string.h:6613:23: note:   mismatched types '__va_list_tag*' and 'int'
          L"%Lf", __val);
                       ^
make[4]: *** [Makefile:9680: application.o] Error 1
make[4]: Leaving directory '/home/trb/mainx86/build_dir/target-x86_64_musl/qBittorrent-release-4.3.1/src'
make[3]: *** [Makefile:47: sub-src-make_first] Error 2
make[3]: Leaving directory '/home/trb/mainx86/build_dir/target-x86_64_musl/qBittorrent-release-4.3.1'
make[2]: *** [Makefile:66: /home/trb/mainx86/build_dir/target-x86_64_musl/qBittorrent-release-4.3.1/.built] Error 2
make[2]: Leaving directory '/home/trb/mainx86/package/lean/qBittorrent'
time: package/lean/qBittorrent/compile#6.70#0.30#7.47
    ERROR: package/lean/qBittorrent failed to build.
make[1]: *** [package/Makefile:114: package/lean/qBittorrent/compile] Error 1
make[1]: Leaving directory '/home/trb/mainx86'
make: *** [/home/trb/mainx86/include/toplevel.mk:230：package/qBittorrent/compile] 错误 2

closed 2021-06-17 by coolsnowwolf

marleeang1992, 2021-08-01:
求问，旧版本qb总是崩溃，下载新版本后丢失之前全部记录，手里种子文件也删光了
如何把旧下载目录下的文件导入并继续下载，谢谢