coolsnowwolf/lede #6403: PVE/ESXI虚拟机下固件无法升级

star-o0O, 2021-02-19:
大多数教程中
PVE安装openwrt使用qm importdisk命令操作img文件添加磁盘
ESXI安装openwrt使用star wind对img文件做格式转换再添加磁盘
之前有试过，用这两种方法安装openwrt后，无法在WEB页面中升级固件（具体表现为上传刷写后不重启/重启后没升级）
试过许多搜索方式，没能找到原因和解决办法
望大佬们指点方向（大概一周前看到过一篇帖子好像说是可以解决，但一回头就找不到这篇帖子了）

minlang112, 2021-02-20:
直接备份配置文件，然后更换虚拟磁盘文件不是也挺快的吗

yankeecyb, 2021-02-21:
可以啊，我一直pve下升级的

star-o0O, 2021-02-22:
> 可以啊，我一直pve下升级的

怎么装的？

chenyg0911, 2021-02-22:
VirtualBox/ESXi lede一直都是在线升级的。应该是你选择的固件不匹配。

star-o0O, 2021-02-22:
> VirtualBox/ESXi lede一直都是在线升级的。应该是你选择的固件不匹配。

固件自编译的X86-64，img格式文件。如何安装和升级的方便细说吗

chenyg0911, 2021-02-23:
第一次安装，网上有现成的教程。后面编译完以后，就可以在openwrt的web里面升级了。img选用你安装的一样的就行。一种是EFI的，一种不是，用哪个装的升级时选哪个。

xiangfeidexiaohuo, 2021-02-25:
看你固件分区大小吧，有些人的固件，分区很小，有些人的很大。

star-o0O, 2021-03-01:
找到了http://www.wifizoo.net/archives/1995

closed 2021-03-01 by star-o0O

YipSzeho, 2022-02-18:
> 找到了http://www.wifizoo.net/archives/1995

打不开了，请问是怎么解决的呢？