coolsnowwolf/lede #6404: 大家不要用 网关模式/旁路由模式  了,会有路由侧漏问题

z826540272, 2021-02-19:
导致安卓手机/盒子 翻墙经常不稳定.
经过1年的排查,发现是**路由侧漏**, 用着用着 ip转发会直接跳过网关,直通光猫,导致被GFW阻挡.
完全随机性的,开关WIFI后可以暂时缓解,但是过会儿又会这样.

目前Windows下未发现此问题, Android 高通801和845手机都复现了此问题.

hcym, 2021-02-19:
用vlan，都是瞎掰乎

xe5700, 2021-02-21:
不是这个问题，安卓的问题是DNS会多提供一个IP地址，你只提供了一个主要服务器他强行给你加个备用DNS服务器，OPENWRT默认DNS地址只有一个，你给你网关多加个IP旁路由给分配两个IP就好了。

z826540272, 2021-02-21:
不是，你说的都排除了

mcucore, 2021-02-24:
上级路由或光猫的DHCP没关吧？

huang1wen, 2021-02-24:
我一直用树莓派做旁路由，手里三个安卓一点事都没已经大半年了，无论主路由开dhcp旁路由关（手动指定翻墙设备网关指向旁路由ip） 还是旁路由开dhcp主路由关都没问题

z826540272, 2021-02-24:
> 上级路由或光猫的DHCP没关吧？

当然不是.

z826540272, 2021-02-24:
> 我一直用树莓派做旁路由，手里三个安卓一点事都没已经大半年了，无论主路由开dhcp旁路由关（手动指定翻墙设备网关指向旁路由ip） 还是旁路由开dhcp主路由关都没问题

经常用谷歌Play或者youtube的最容易发现

hahach, 2021-02-25:
原来不是我一个遇到这种问题...
我的网络环境没办法，只能用硬路由中继上级路由(本地没有有线接口，上级路由信号太差，用硬路由中继比手机等设备直连好很多)，同时硬路由作为我的wifi，lede跑在nas上。lede，硬路由，硬路由下所有设备以及上级路由均在同一网段，上级路由提供DHCP但我不用，硬路由wifi下的所有设备手动设定网关与DNS指向lede，lede网关指向上级路由，我也不懂我这算网关模式？旁路模式？
最近几个月用起来偶尔会有win10，手机，平板皆无法访问墙外，lede上测试无问题的情况。这时只能lede上重启dnsmasq，还不一定有效。
话说这种情况是软件问题吗？我这只要出现这个问题，不论安卓还是win或者ios都是一起不行的

hcym, 2021-02-25:
都说了一定要有vlan标定的，我是用了半年usb才会的

closed 2021-06-17 by coolsnowwolf

jehovist, 2022-04-06:
> 不是这个问题，这个应该说是安卓系统的问题，安卓的问题是DNS会多提供一个IP地址，你只提供了一个主要服务器他强行给你加个备用DNS服务器，然后这个DNS服务器会被功夫网路污染掉然后你懂的OPENWRT默认DNS地址只有一个，你给你旁路由多加个IP给分配两个IP做DNS服务器就好了。

大佬你好，问个小白问题，给openwrt添加ip具体是怎么操作呢

hbtx99099, 2024-08-26:
我把第二个DNS填写127.0.0.0也就是本机，暂时可用。