coolsnowwolf/lede #6405: 建议SSR+提供本地DNS应用不走翻墙线路的选项

kenyloveg, 2021-02-19:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：解决DNS本地和翻墙路线的解析问题

### (2) 路由器型号和固件版本

A：X86

### (3) 详细日志

A：

本地DNS应用（smartdns或者adguard的udp 53，443，853等端口）不走翻墙路线（不通过SSR+出口），这样就能解决国内网站返回后的结果为国内CDN（解决了淘宝提示VPN，微信刷不出文章等问题），国外网站可能解析到离本地IP较近的CDN（香港，台湾，韩国，日本，新加坡等，大部分常用的节点都是这几个国家，速度影响不大）

目前在用的解决方案是在本地一台不走翻墙线路的下载机上运行adguardhome，效果不错，如果openwrt固件的ssr+能加上此功能就能节约一个设备，谢谢

NagaseKouichi, 2021-02-20:
应该只有GFW列表里面的域名会走SSR+的解析吧？

hcym, 2021-02-20:
昨天编译的x和arm，没问题，mtk的几乎不能生活自理😂

kenyloveg, 2021-02-20:
我的主路由是RouterOS，openwrt的ssr+是全局模式，所有非CN的IP一律走openwrt

hxlls, 2021-02-20:
你做什么“旁”路由之类的的你自己的事啊，设计是直接作为主路由使用，也就是你自己使用的事情，那和这个IPK的设计有啥问题么？

kenyloveg, 2021-02-20:
@hxlls 建议，讨论，所以ssr+才会不断进步，另:你不需要不代表别人都不需要

dongjuanyong, 2021-02-20:
把DNS劫持的这两行注解或删掉就行了https://github.com/coolsnowwolf/lede/blob/73fdfdf6aa62f0b28535b036313266f2ab06e993/package/lean/default-settings/files/zzz-default-settings#L39-L40

kenyloveg, 2021-02-20:
> 把DNS劫持的这两行注解或删掉就行了
> 
> https://github.com/coolsnowwolf/lede/blob/73fdfdf6aa62f0b28535b036313266f2ab06e993/package/lean/default-settings/files/zzz-default-settings#L39-L40

这个步骤刷完固件就先做了，无效

dongjuanyong, 2021-02-21:
那就复杂了呗
把你这些需要进行请求的上级dns服务器地址找出来，在iptable的chain中return target为这些地址的包（就和国内地址一样操作）

kenyloveg, 2021-02-21:
大佬，但凡是公共dns，不管dot/doh/dnscrypt（域名为主的dns服务器）还是纯ip的plain udp 53服务商，都有cdn，这么多ip一次几次几十次都拉不全吧😂

dongjuanyong, 2021-02-21:
？？？
你客户端程序dns设置的是哪个地址，你还不知道吗？
你要是不知道的话，又怎么知道它走了fq路线而不是直连？

而且，dns返回的结果是根据cdn优化的，没听说过dns服务器地址也因cdn不同。即使是doh这种域名地址需要bootstrap的？你抓个包或者nslookup一次不也就够了吗

kenyloveg, 2021-02-21:
就是因为SSR+在全局模式下，本地应用adguardhome会走翻墙线路，所以才提出本地dns应用不走翻墙线路（SSR+全局模式下）的需求（类似passwall的路由器自身tcp udp代理模式）

dongjuanyong, 2021-02-21:
你要是非得让某个程序的流量都不走ssr，我能想到的有这么一招，前提是应用和ssr都跑在同一台设备上：
https://xtls.github.io/documents/level-2/iptables_gid/#3-%E9%85%8D%E7%BD%AE%E6%9C%80%E5%A4%A7%E6%96%87%E4%BB%B6%E5%A4%A7%E5%BC%80%E6%95%B0%E8%BF%90%E8%A1%8Cxray%E5%AE%A2%E6%88%B7%E7%AB%AF
总体思路就是这样：
1. 让应用跑在特定的gid上（非root/nobody）
2. 利用类似`iptables -t mangle(nat) -I XXX(对应的chain) -m owner --gid-owner 23333 -j RETURN`让流量不被重定向

kenyloveg, 2021-02-21:
> 你要是非得让某个程序的流量都不走ssr，我能想到的有这么一招，前提是应用和ssr都跑在同一台设备上： [...]

目前有思路了，就看大佬是否愿意添加代码了，抛砖引玉完成
谢谢

dongjuanyong, 2021-02-21:
别想太多，你这这小众需求只能自己动手丰衣足食
钞票多的话可以找大佬定制

kenyloveg, 2021-06-05:
更新下，提交给adguardhome的issue有了进展
https://github.com/AdguardTeam/AdGuardHome/commit/48c44c29aba472839b25fe1132c6ee8b91a27bba

closed 2021-06-17 by coolsnowwolf