coolsnowwolf/lede #6406: 编译极壹S路由（HC5661）出错

kleisth, 2021-02-20:
反馈bug/问题模板，提建议请删除

 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

2. 详细叙述

 (1) 具体问题

A：编译过程中出错，其他版本固件编译无问题，HC5661A和网件R6220，还有X86都能正常编译，编译HC5661时出错无法完成编译

(2) 路由器型号和固件版本

A：极壹S（HC5661A），最新固件

 (3) 详细日志

A：
../dts/mt7620a.dtsi:119.4-28: Warning (gpios_property): /palmbus@10000000/gpio@600:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:119.4-28: Warning (gpios_property): /palmbus@10000000/gpio@600:ralink,num-gpios: Could not get phandle node for (cell 0)
../dts/mt7620a.dtsi:136.4-28: Warning (gpios_property): /palmbus@10000000/gpio@638:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:125.19-142.5: Warning (gpios_property): /palmbus@10000000/gpio@638: Missing property '#gpio-cells' in node /usbphy or bad phandle (referred from ralink,num-gpios[0])
  also defined at ../dts/mt7620a_xiaomi_miwifi-mini.dts:59.8-61.3
../dts/mt7620a.dtsi:155.4-28: Warning (gpios_property): /palmbus@10000000/gpio@660:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:155.4-28: Warning (gpios_property): /palmbus@10000000/gpio@660:ralink,num-gpios: Could not get phandle node for (cell 0)
../dts/mt7620a.dtsi:174.4-27: Warning (gpios_property): /palmbus@10000000/gpio@688:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:163.19-180.5: Warning (gpios_property): /palmbus@10000000/gpio@688: Missing property '#gpio-cells' in node /palmbus@10000000/intc@200 or bad phandle (referred from ralink,num-gpios[0])
rm -f /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_xiaomi_miwifi-mini.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/arch/mips/boot/dts -I/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/arch/mips/boot/dts/include -I/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/include/ -undef -D__DTS__  -o /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_xiaomi_miwifi-r3.dtb.tmp ../dts/mt7620a_xiaomi_miwifi-r3.dts
/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_xiaomi_miwifi-r3.dtb /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_xiaomi_miwifi-r3.dtb.tmp
../dts/mt7620a.dtsi:119.4-28: Warning (gpios_property): /palmbus@10000000/gpio@600:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:119.4-28: Warning (gpios_property): /palmbus@10000000/gpio@600:ralink,num-gpios: Could not get phandle node for (cell 0)
../dts/mt7620a.dtsi:136.4-28: Warning (gpios_property): /palmbus@10000000/gpio@638:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:125.19-142.5: Warning (gpios_property): /palmbus@10000000/gpio@638: Missing property '#gpio-cells' in node /usbphy or bad phandle (referred from ralink,num-gpios[0])
  also defined at ../dts/mt7620a_xiaomi_miwifi-r3.dts:121.8-123.3
../dts/mt7620a.dtsi:155.4-28: Warning (gpios_property): /palmbus@10000000/gpio@660:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:155.4-28: Warning (gpios_property): /palmbus@10000000/gpio@660:ralink,num-gpios: Could not get phandle node for (cell 0)
../dts/mt7620a.dtsi:174.4-27: Warning (gpios_property): /palmbus@10000000/gpio@688:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:163.19-180.5: Warning (gpios_property): /palmbus@10000000/gpio@688: Missing property '#gpio-cells' in node /palmbus@10000000/intc@200 or bad phandle (referred from ralink,num-gpios[0])
rm -f /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_xiaomi_miwifi-r3.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/arch/mips/boot/dts -I/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/arch/mips/boot/dts/include -I/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/include/ -undef -D__DTS__  -o /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb.tmp ../dts/mt7620a_youku_yk1.dts
/home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb.tmp
../dts/mt7620a_youku_yk1.dts:79.3-22: ERROR (duplicate_property_names): /palmbus@10000000/spi@b00/flash@0:broken-flash-reset: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
make[5]: *** [Makefile:202: /home/kleist/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb] Error 2
make[5]: Leaving directory '/home/kleist/lede/target/linux/ramips/image'
make[4]: *** [Makefile:25: install] Error 2
make[4]: Leaving directory '/home/kleist/lede/target/linux/ramips'
make[3]: *** [Makefile:13: install] Error 2
make[3]: Leaving directory '/home/kleist/lede/target/linux'
time: target/linux/install#52.88#11.97#64.38
make[2]: *** [target/Makefile:25: target/linux/install] Error 2
make[2]: Leaving directory '/home/kleist/lede'
make[1]: *** [target/Makefile:19: /home/kleist/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/kleist/lede'
make: *** [/home/kleist/lede/include/toplevel.mk:224：world] 错误 2

BLX2233, 2021-02-20:
参考这个[https://github.com/coolsnowwolf/lede/issues/6388#issuecomment-781804160](url)

kleisth, 2021-02-20:
> 参考这个[https://github.com/coolsnowwolf/lede/issues/6388](url)

你给的链接是空的
顺便补充句，用github的actions编译也是一样报错的

kleisth, 2021-02-20:
> 参考这个[https://github.com/coolsnowwolf/lede/issues/6388](url)

谢谢

> 参考这个[https://github.com/coolsnowwolf/lede/issues/6388](url)

谢谢，果然是这个问题

PR #6408 (2021-02-21): Revert "Add broken-flash-reset flag to Youku YK-L1"
This reverts commit 90bd81e0991046927634bf74830ec1c835d83e85.

The commit 90bd81e is a duplicate of b132179.

Fixes: #6388
Fixes: #6406
commit b484f2ee (2021-02-21): Revert "Add broken-flash-reset flag to Youku YK-L1" (#6408)

This reverts commit 90bd81e0991046927634bf74830ec1c835d83e85.

The commit 90bd81e099 is a duplicate of b132179c89.

Fixes: #6388
Fixes: #6406
closed 2021-02-21 by coolsnowwolf