coolsnowwolf/lede #6409: PPPOE拨号，断线后不会自动重连！

forever2019, 2021-02-20:
## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述
PPPOE拨号，断线后不会自动重连！

### (1) 具体问题

A：家里使用的联通，每2天运营商强制踢下线，强制更新IP，而openwrt没有自动检测网络状态功能，不会重连pppoe。
能否增加这个功能。
定时功能，每隔多少分钟，检测自定义的域名或IP的连接状态，断线后pppoe重新拨号。

### (2) 路由器型号和固件版本

A：d2550  最新分支

### (3) 详细日志

A：无

ricksuzade-maker, 2021-02-21:
我也是openwrt拨号啊，没遇到这个问题呢？

forever2019, 2021-02-21:
> 我也是openwrt拨号啊，没遇到这个问题呢？

openwrt有没有检测和重播功能?

db-one, 2021-02-21:
正常断线都会自己重拨的啊
不过我倒是遇到过，重拨之后拨号是成功了，也拿到IP了，但是无法上网，不知道是路由自己问题还是运营商那边拨号拿到的那个IP有问题，这时候重启下网卡，让它重新拨号就好了
只遇到过两次，概率不大，不过为了外出时候不遇到这种问题，无法联网回来，我就自己写了一个命令，定时ping外网，ping不通就隔一会再尝试，如果还是不通就判定网络出错直接重启网络

forever2019, 2021-02-21:
我也是这样  所以每天5点重启路由  临时解决了
应该为国内的使用环境 直接开集成检测并重连pppoe功能
希望哪位开发者能做出来  非常感谢

db-one, 2021-02-21:
其实luci-app-serverchan就自带有网络检测的，不同就重启网络，只不过我没编译进去，所以就自己写了SH脚本

forever2019, 2021-02-21:
> 其实luci-app-serverchan就自已通过网络检测的，不同就重启网络，只不过我没编译进去，所以就自己写了SH脚本

我编译个试试   谢谢

minlang112, 2021-02-21:
鄙人用电信，每30天被局端踢下线，会自动重拨……

foxhoundhcm, 2021-02-22:
我的情况是重启后拨号成功，地址已经分配了，但是就是上不了网，不知如何定位问题。一定要重新再点一次连接重新拨号才行

db-one, 2021-02-23:
> 我的情况是重启后拨号成功，地址已经分配了，但是就是上不了网，不知如何定位问题。一定要重新再点一次连接重新拨号才行

你说的情况应该跟我上面说的差不多，我也是这样，不过只遇到过两三次

db-one, 2021-02-23:
> 鄙人用电信，每30天被局端踢下线，会自动重拨……
> 
> 如果遇到重连之后无法上网的情况，怀疑是防火墙的问题

你这个幸福了，我们的都是48小时左右，有的时候多 有时候甚至不到48小时

forever2019, 2021-02-23:
手动拨号也无法上网，必须把链接关闭，重新启用才可以，不知道什么原因。
我看大家都有遇到，说明这个问题确实存在，应该针对国内环境，重点优化下。

minlang112, 2021-02-23:
> 手动拨号也无法上网，必须把链接关闭，重新启用才可以，不知道什么原因。
> 我看大家都有遇到，说明这个问题确实存在，应该针对国内环境，重点优化下。

这个得看下系统日志了吧……一般不会有这种问题的，看看是不是把mwan3开启了，这玩意很坑

forever2019, 2021-02-23:
我没编译过这个东西  
现在换ikuai了  谁遇到这个问题 可以把日志发上来   看看具体原因

db-one, 2021-02-23:
> > 手动拨号也无法上网，必须把链接关闭，重新启用才可以，不知道什么原因。
> > 我看大家都有遇到，说明这个问题确实存在，应该针对国内环境，重点优化下。
> 
> 这个得看下系统日志了吧……一般不会有这种问题的，看看是不是把mwan3开启了，这玩意很坑

我没用过mwan3   也没多拨过，因为出现概率极低，至少我这里是这样，一两年了，一共也只出现过三次，上次出现还是去年大概11月份左右，当时看了日志也没发现问题，就是正常拨号的记录，所以也不清楚具体什么问题，个人猜测可能是软路由网络或者防火墙，或者是联通拨号系统问题？  出现概率太低，不好排除了，太难复现

xiaobaishu1, 2021-02-23:
我这里是正常的 我关闭光猫重启也是能识别到断网会自己重播
[image]

foxhoundhcm, 2021-02-24:
我也没有mwan3，就没编译这个东西进去，还有就是重启完后的系统日志拨号部分与之后手动拨号的日志部分并没有什么不同，也没有什么错误，不知道该如何查找定位这个问题。

Suwipe, 2021-02-25:
我这也遇到每2天运营商强制断线重播后无法上网的问题，重播后IPv4拿到IP，IPv6没拿到IP，但是下面的设备无法上网，需要手动重连，wan链接里设置LCP响应也没用，感觉wan口本身是正常的，估计真可能是本机防火墙的问题

zgc, 2021-02-26:
我也有这个问题，我自己写了个脚本，15分钟检查一次，网络不通就重拨，看日志的话，基本可以踏点重拨，至少家里的设备没有提示断网。。。算是个伪无缝衔接

acooler15, 2021-03-03:
我是每天凌晨3点 `ifup wan`

swbbs1982, 2021-06-07:
我好像发现pppoe 拨号问题的原因了，是usb 网卡的问题，吧usb 网卡做 lan 问题就解决了

laaArmy, 2021-08-25:
我想问问，你问题解决没

d407565739, 2022-12-13:
> 我也有这个问题，我自己写了个脚本，15分钟检查一次，网络不通就重拨，看日志的话，基本可以踏点重拨，至少家里的设备没有提示断网。。。算是个伪无缝衔接

具体要怎么操作啊 大佬

zgc, 2022-12-18:
> > 我也有这个问题，我自己写了个脚本，15分钟检查一次，网络不通就重拨，看日志的话，基本可以踏点重拨，至少家里的设备没有提示断网。。。算是个伪无缝衔接
> 
> 具体要怎么操作啊 大佬

我不是大佬，我也是小白，如果你想知道我的脚本，可以看这里 https://raw.githubusercontent.com/zgc/Actions-LEDE/main/scripts/check_wan_connect.sh
我把脚本加入计划任务 */5 * * * * /etc/check_wan_connect.sh