coolsnowwolf/lede #6412: 编译时遇到以下报错信息

yuos-bit, 2021-02-21:
also defined at ../dts/mt7620a_xiaomi_miwifi-r3.dts:121.8-123.3
../dts/mt7620a.dtsi:155.4-28: Warning (gpios_property): /palmbus@10000000/gpio@660:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:155.4-28: Warning (gpios_property): /palmbus@10000000/gpio@660:ralink,num-gpios: Could not get phandle node for (cell 0)
../dts/mt7620a.dtsi:174.4-27: Warning (gpios_property): /palmbus@10000000/gpio@688:ralink,num-gpios: cell 0 is not a phandle reference
../dts/mt7620a.dtsi:163.19-180.5: Warning (gpios_property): /palmbus@10000000/gpio@688: Missing property '#gpio-cells' in node /palmbus@10000000/intc@200 or bad phandle (referred from ralink,num-gpios[0])
rm -f /home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_xiaomi_miwifi-r3.dtb.tmp
mipsel-openwrt-linux-musl-cpp -nostdinc -x assembler-with-cpp -I/home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/arch/mips/boot/dts -I/home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/arch/mips/boot/dts/include -I/home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/include/ -undef -D__DTS__  -o /home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb.tmp ../dts/mt7620a_youku_yk1.dts
/home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/linux-5.4.95/scripts/dtc/dtc -O dtb -i../dts/ -Wno-unit_address_vs_reg -Wno-simple_bus_reg -Wno-unit_address_format -Wno-pci_bridge -Wno-pci_device_bus_num -Wno-pci_device_reg -Wno-avoid_unnecessary_addr_size -Wno-alias_paths -Wno-graph_child_address -Wno-graph_port -Wno-unique_unit_address  -o /home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb /home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb.tmp
../dts/mt7620a_youku_yk1.dts:79.3-22: ERROR (duplicate_property_names): /palmbus@10000000/spi@b00/flash@0:broken-flash-reset: Duplicate property name
ERROR: Input tree has errors, aborting (use -f to force output)
Makefile:201: recipe for target '/home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb' failed
make[5]: *** [/home/oprenwrt/lede/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7620/image-mt7620a_youku_yk1.dtb] Error 2
make[5]: Leaving directory '/home/oprenwrt/lede/target/linux/ramips/image'
Makefile:25: recipe for target 'install' failed
make[4]: *** [install] Error 2
make[4]: Leaving directory '/home/oprenwrt/lede/target/linux/ramips'
Makefile:13: recipe for target 'install' failed
make[3]: *** [install] Error 2
make[3]: Leaving directory '/home/oprenwrt/lede/target/linux'
time: target/linux/install#666.95#102.88#759.66
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: *** [target/linux/install] Error 2
make[2]: Leaving directory '/home/oprenwrt/lede'
target/Makefile:19: recipe for target '/home/oprenwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install' failed
make[1]: *** [/home/oprenwrt/lede/staging_dir/target-mipsel_24kc_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/home/oprenwrt/lede'
/home/oprenwrt/lede/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

请问这个怎么解决呢

jjit, 2021-02-21:
https://github.com/coolsnowwolf/lede/issues/6388
修改源码的这个文件 target/linux/ramips/dts/mt7620a_youku_yk1.dts
文件里面有 2个 broken-flash-reset; 重复了，去掉一个

yuos-bit, 2021-02-21:
好的，谢谢大哥

yuos-bit, 2021-02-21:
我是编译的小米路由3的固件，请问为啥会出现关于优酷路由宝的报错呢，亲

minlang112, 2021-02-21:
evert "Add broken-flash-reset flag to Youku YK-L1" (#6408)

已修复，git pull一下

yuos-bit, 2021-02-22:
谢谢，已 pull

HUiTHEME, 2021-02-22:
这个问题怎么还一直存在，好多Issues里的好多人都说解决了，可问题依然在啊。。。

closed 2021-02-23 by coolsnowwolf