coolsnowwolf/lede #6414: 编译错误，求助大佬们

madcat0513, 2021-02-21:
## 1.关于你要提交的问题
编译错误
Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：都已经更新到今天最新，貌似是编译wireless-regdb的时候出错了

### (2) 路由器型号和固件版本

A：R7800

### (3) 详细日志

A：touch /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/.prepared_f7366bfb5bbce42d96d52affb445af77_6664517399ebbbc92a37c5bb081b5c53
rm -f /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/.configured_*
rm -f /home/madcat/openwrt/lean/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.wireless-regdb_installed
(cd /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/./; if [ -x ./configure ]; then find /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/ -name config.guess | xargs -r chmod u+w; find /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/ -name config.guess | xargs -r -n1 cp --remove-destination /home/madcat/openwrt/lean/scripts/config.guess; find /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/ -name config.sub | xargs -r chmod u+w; find /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/ -name config.sub | xargs -r -n1 cp --remove-destination /home/madcat/openwrt/lean/scripts/config.sub; AR="arm-openwrt-linux-muslgnueabi-gcc-ar" AS="arm-openwrt-linux-muslgnueabi-gcc -c -Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20=wireless-regdb-2020.11.20 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" LD=arm-openwrt-linux-muslgnueabi-ld NM="arm-openwrt-linux-muslgnueabi-gcc-nm" CC="arm-openwrt-linux-muslgnueabi-gcc" GCC="arm-openwrt-linux-muslgnueabi-gcc" CXX="arm-openwrt-linux-muslgnueabi-g++" RANLIB="arm-openwrt-linux-muslgnueabi-gcc-ranlib" STRIP=arm-openwrt-linux-muslgnueabi-strip OBJCOPY=arm-openwrt-linux-muslgnueabi-objcopy OBJDUMP=arm-openwrt-linux-muslgnueabi-objdump SIZE=arm-openwrt-linux-muslgnueabi-size CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20=wireless-regdb-2020.11.20 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CXXFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -Wno-error=unused-but-set-variable -Wno-error=unused-result -mfloat-abi=hard -ffile-prefix-map=/home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20=wireless-regdb-2020.11.20 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro " CPPFLAGS="-I/home/madcat/openwrt/lean/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/include -I/home/madcat/openwrt/lean/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/include -I/home/madcat/openwrt/lean/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/include/fortify -I/home/madcat/openwrt/lean/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/include " LDFLAGS="-L/home/madcat/openwrt/lean/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/usr/lib -L/home/madcat/openwrt/lean/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/lib -L/home/madcat/openwrt/lean/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/usr/lib -L/home/madcat/openwrt/lean/staging_dir/toolchain-arm_cortex-a15+neon-vfpv4_gcc-8.4.0_musl_eabi/lib -znow -zrelro "   ./configure --target=arm-openwrt-linux --host=arm-openwrt-linux --build=x86_64-pc-linux-gnu --program-prefix="" --program-suffix="" --prefix=/usr --exec-prefix=/usr --bindir=/usr/bin --sbindir=/usr/sbin --libexecdir=/usr/lib --sysconfdir=/etc --datadir=/usr/share --localstatedir=/var --mandir=/usr/man --infodir=/usr/info --disable-nls  ; fi; )
touch /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/.built
touch /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/.built_check
/home/madcat/openwrt/lean/staging_dir/host/bin/python /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/db2fw.py /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/regulatory.db /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/db.txt
bash: /home/madcat/openwrt/lean/staging_dir/host/bin/python: No such file or directory
make[3]: *** [Makefile:35: /home/madcat/openwrt/lean/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/wireless-regdb-2020.11.20/.built] Error 127
make[3]: Leaving directory '/home/madcat/openwrt/lean/package/firmware/wireless-regdb'
time: package/firmware/wireless-regdb/compile#0.21#0.07#0.26
make[2]: *** [package/Makefile:113: package/firmware/wireless-regdb/compile] Error 2
make[2]: Leaving directory '/home/madcat/openwrt/lean'
make[1]: *** [package/Makefile:107: /home/madcat/openwrt/lean/staging_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/madcat/openwrt/lean'
make: *** [/home/madcat/openwrt/lean/include/toplevel.mk:224：world] 错误 2

jjit, 2021-02-21:
看一下这个：https://github.com/coolsnowwolf/lede/issues/6224

minlang112, 2021-02-21:
bash: /home/madcat/openwrt/lean/staging_dir/host/bin/python: No such file or directory

检查/home/madcat/openwrt/lean/staging_dir/host/bin/里面这个叫python的软链接，他指向的位置是不是不存在的，用winscp连进去看如果错误的话是红色图标，自己修改一下

madcat0513, 2021-02-21:
看了下，果然是这个链接的问题，不知道啥时候我给升级成python3.8了......
感谢大佬们！

closed 2021-02-21 by madcat0513