coolsnowwolf/lede #6419: 如果配置https访问路由器

BigDog66, 2021-02-23:
这个固件默认只能通过http(80)访问web管理  
无法通过https访问  
我想开启https后从外网访问 不知道如何操作。 telnet 443端口也是不通的
我之前用的是的koolshar官方的Openwrt 刷完后443端口是通的  替换掉ETC里面的证书 做好端口转发就能在外网通过https访问了
L大这个固件我不知道该如何操作才能实现外网HTTPS访问  望知道的大佬教我如何操作

sxx1314, 2021-02-24:
x86 直接换成nginx即可 配置更方便

hahach, 2021-02-25:
两三年前搞的，当时的web服务器默认是uhttpd，而uhttpd是默认不开启https，不监听443的。
如果现在还是uhttpd的话，你就按这个关键词自己去找修改uhttpd配置文件和替换证书的方法。

如果现在的默认web服务器是Nginx的话，一样的，自己去找找看怎么开启。

alepennyq, 2021-03-04:
> 这个固件默认只能通过http(80)访问web管理
> 无法通过https访问
> 我想开启https后从外网访问 不知道如何操作。 telnet 443端口也是不通的
> 我之前用的是的koolshar官方的Openwrt 刷完后443端口是通的 替换掉ETC里面的证书 做好端口转发就能在外网通过https访问了
> L大这个固件我不知道该如何操作才能实现外网HTTPS访问 望知道的大佬教我如何操作

/etc/config/uhttpd参照修改：
config uhttpd 'main'
	list listen_http '0.0.0.0:80'
	list listen_http '[::]:80'
	list listen_https '0.0.0.0:443'
	list listen_https '[::]:443'
	option redirect_https '0'
	option home '/www'
	option max_requests '3'
	option max_connections '100'
	option cert '/etc/uhttpd.crt'
	option key '/etc/uhttpd.key'
	option cgi_prefix '/cgi-bin'
	list lua_prefix '/cgi-bin/luci=/usr/lib/lua/luci/sgi/uhttpd.lua'
	option script_timeout '60'
	option network_timeout '30'
	option http_keepalive '20'
	option tcp_keepalive '1'
	option ubus_prefix '/ubus'
	list index_page 'cgi-bin/luci'
	option rfc1918_filter '0'

config cert 'defaults'
	option days '730'
	option key_type 'rsa'
	option bits '2048'
	option ec_curve 'P-256'
	option country 'ZZ'
	option state 'Somewhere'
	option location 'Unknown'
	option commonname 'OpenWrt'

证书放/etc/，命名为uhttpd.crt uhttpd.key

closed 2021-06-17 by coolsnowwolf

nevertoday, 2022-12-07:
我的死活访问不了， 说私密连接 ，真是奇怪，貌似没有扫到证书目录一般，如同死星！