coolsnowwolf/lede #6421: 编译错误 看日志是 关于 xray-core 的

bddddd, 2021-02-23:
## 1.关于你要提交的问题
编译固件失败， 代码都已经拉取了最新的
Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译失败 日志如下

### (2) X84

A：

### (3) 详细日志

A：
Makefile:143: recipe for target '/home/bdd/openwrt/build_dir/target-x86_64_musl/Xray-core-1.3.0/.built' failed
make[3]: *** [/home/bdd/openwrt/build_dir/target-x86_64_musl/Xray-core-1.3.0/.built] Error 2
make[3]: Leaving directory '/home/bdd/openwrt/feeds/helloworld/xray-core'
time: package/feeds/helloworld/xray-core/compile#1.17#0.93#0.96
package/Makefile:111: recipe for target 'package/feeds/helloworld/xray-core/compile' failed
make[2]: *** [package/feeds/helloworld/xray-core/compile] Error 2
make[2]: Leaving directory '/home/bdd/openwrt'
package/Makefile:107: recipe for target '/home/bdd/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile' failed
make[1]: *** [/home/bdd/openwrt/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/bdd/openwrt'
/home/bdd/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2

lyx32, 2021-03-05:
我编译了几次，发现几个常见问题
1.**不要用root编译！**这一点比较容易被忽视
2.真实内存+虚拟内存最好大于4G以上
3.如果要编译v2ray，xray，torjan-go之类的，要安装go，并且设置GOPROXY

bddddd, 2021-03-05:
> 我编译了几次，发现几个常见问题
> 1.**不要用root编译！**这一点比较容易被忽视
> 2.真实内存+虚拟内存最好大于4G以上
> 3.如果要编译v2ray，xray，torjan-go之类的，要安装go，并且设置GOPROXY

感谢回复 应该是我梯子的问题 可能有些地方的文件因为网络原因没有下载下来
重做了下环境换了个节点目前是正常的

lyx32, 2021-03-05:
> 
> 
> > 我编译了几次，发现几个常见问题
> > 1.**不要用root编译！**这一点比较容易被忽视
> > 2.真实内存+虚拟内存最好大于4G以上
> > 3.如果要编译v2ray，xray，torjan-go之类的，要安装go，并且设置GOPROXY
> 
> 感谢回复 应该是我梯子的问题 可能有些地方的文件因为网络原因没有下载下来
> 重做了下环境换了个节点目前是正常的

不客气，弄好就行

closed 2021-06-17 by coolsnowwolf