coolsnowwolf/lede #6423: igb-I350-T4网卡问题

kuangdongliang, 2021-02-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
通过exsi虚拟机直通了igb-I350-T4网卡，在lede固件找不到相关驱动，只能有82575，请问佬大能否帮忙加上去呢。

Q：是否搜索了issue (使用 "x" 选择)
* [ x"] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
通过exsi虚拟机直通了igb-I350-T4网卡，在lede固件找不到相关驱动，只能有82575，请问佬大能否帮忙加上去呢。

A：

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：
[image]

dgulan, 2021-02-24:
我也是esxi 直通了。
默认编译固件以后，网卡正常工作的。

Aes64X, 2021-02-24:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> 通过exsi虚拟机直通了igb-I350-T4网卡，在lede固件找不到相关驱动，只能有82575，请问佬大能否帮忙加上去呢。
> 
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [ x"] 没有类似的issue
> 
> ## 2. 详细叙述
> ### (1) 具体问题
> 通过exsi虚拟机直通了igb-I350-T4网卡，在lede固件找不到相关驱动，只能有82575，请问佬大能否帮忙加上去呢。
> 
> A：
> 
> ### (2) 路由器型号和固件版本
> A：
> 
> ### (3) 详细日志
> A：
> [image]

你自己告诉了你自己答案，你自己截图里已经告诉你自己了，igb驱动包含i350

minlang112, 2021-02-25:
5.4内核好像默认勾选igb驱动的吧……

closed 2021-06-17 by coolsnowwolf