coolsnowwolf/lede #6426: Xray v1.24下，原来的v2ray+ws+tls VMESS配置无法使用了

ttmi9002, 2021-02-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue

## 2. 详细叙述
原来的v2ray配置，VMESS协议,ws+tls，配置在原来的v2ray v4.25.1上是能用的

### (1) 具体问题

A：原来的v2ray设置在新的Xray插件中失效，Xray版本v1.24

尝试解决办法，
1. 修改AlterID = 1，原有的v2ray配置的AlterID是0
2. 填写TLS Host，TLS Host填写的内容和服务器地址是一样的，“xxxxxxx.xyz”
以上两个办法都无效

### (2) 路由器型号和固件版本

A：XiaoMi R3G 

固件版本 OpenWrt R20.12.12

### (3) 详细日志

A：
2021-02-24 00:41:41: Main node:Xray 1.2.4 (OpenWrt) Custom (go1.15.6 linux/mipsle) Started!
2021-02-24 00:41:39: ----------start------------
2021-02-24 00:39:16: -----------end------------
2021-02-24 00:39:11: Main node:Xray 1.2.4 (OpenWrt) Custom (go1.15.6 linux/mipsle) Started!
2021-02-24 00:39:09: ----------start------------
2021-02-24 00:38:08: -----------end------------
2021-02-24 00:38:03: Main node:Xray 1.2.4 (OpenWrt) Custom (go1.15.6 linux/mipsle) Started!
2021-02-24 00:38:00: ----------start------------
2021-02-24 00:37:48: -----------end------------

PaulYoung0265, 2021-02-24:
服务端还是v2ray？
改一下吧，也改成xray1.2.4，也支持VMESS+TLS+WS[CDN]模式

ttmi9002, 2021-03-05:
嗯。。服务端还是v2ray
原来是这样。。

closed 2021-03-05 by ttmi9002