coolsnowwolf/lede #6427: 系统报错“arp_cache: neighbor table overflow!”，且占满全部CPU

IMconnor, 2021-02-23:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
一旦在局域网机器中打开比特彗星（版本号：1.74），系统日志就会被类似如下的条目塞满，且占满CPU
`Wed Feb 24 00:26:17 2021 kern.info kernel: [15757.030087] neighbour: arp_cache: neighbor table overflow!`

经初步测试，该问题与UPnP是否开启无关。另，目前编译的版本对utorrent之类的软件支持良好，但是不支持比特彗星或微信语音、B站之类应用的UPnP重定向。

软件 | 版本
-- |  --
libminiupnpc | 2.1.20191224-1
miniupnpd | 2.0.20170421-2

固件的设置几乎就是默认的，没做内核参数之类的更改。

### (2) 路由器型号和固件版本

A：

主机型号 | Intel(R) Celeron(R) CPU J1800 @ 2.41GHz : 2 Core 2 Thread
-- | --
固件版本 | OpenWrt  R21.2.1 / 			LuCI Master (git-21.035.34289-13f96d1)
内核版本 | 5.4.95

### (3) 详细日志

A：

> Wed Feb 24 00:26:22 2021 kern.info kernel: [15762.133381] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:22 2021 kern.info kernel: [15762.134331] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:22 2021 kern.info kernel: [15762.135280] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:23 2021 kern.info kernel: [15762.136902] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:23 2021 kern.info kernel: [15762.140680] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:23 2021 kern.info kernel: [15762.141629] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.warn kernel: [15767.674103] net_ratelimit: 204 callbacks suppressed
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.674106] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.674998] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.677346] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.678129] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.678909] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.679657] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.680431] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.681188] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.682062] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:26:28 2021 kern.info kernel: [15767.682879] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:27:12 2021 daemon.notice miniupnpd[27330]: private/reserved address checking is ignore
> Wed Feb 24 00:27:48 2021 daemon.notice miniupnpd[27330]: private/reserved address checking is ignore
> Wed Feb 24 00:28:19 2021 daemon.err miniupnpd[27330]: Unauthorized to remove PCP mapping internal port 15294, protocol TCP
> Wed Feb 24 00:28:19 2021 daemon.err miniupnpd[27330]: Unauthorized to remove PCP mapping internal port 15294, protocol UDP
> Wed Feb 24 00:28:25 2021 kern.warn kernel: [15885.286399] net_ratelimit: 81 callbacks suppressed
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.286401] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.287828] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.288678] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.290466] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.291304] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.292085] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.292857] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.293041] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.294552] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:25 2021 kern.info kernel: [15885.295451] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 00:28:30 2021 kern.warn kernel: [15890.323057] net_ratelimit: 2836 callbacks suppressed
[... 568 lines cut ...]
> Wed Feb 24 01:30:23 2021 kern.info kernel: [19603.133655] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:23 2021 kern.info kernel: [19603.134251] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:23 2021 kern.info kernel: [19603.134995] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:23 2021 kern.info kernel: [19603.135650] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:23 2021 kern.info kernel: [19603.137738] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.warn kernel: [19608.443198] net_ratelimit: 2411 callbacks suppressed
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.443200] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.443240] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.443916] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.444597] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.445291] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.445968] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.446720] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.447367] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.448047] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:30:29 2021 kern.info kernel: [19608.448739] neighbour: arp_cache: neighbor table overflow!
> Wed Feb 24 01:38:51 2021 daemon.notice netifd: wan (8900): udhcpc: sending renew to 166.111.8.9
> Wed Feb 24 01:38:51 2021 daemon.notice netifd: wan (8900): udhcpc: lease of 59.66.211.27 obtained, lease time 3600
> Wed Feb 24 01:41:11 2021 daemon.notice miniupnpd[21037]: HTTP listening on port 5000
> Wed Feb 24 01:41:11 2021 daemon.notice miniupnpd[21037]: Listening for NAT-PMP/PCP traffic on port 5351
> Wed Feb 24 01:41:11 2021 daemon.notice miniupnpd[21037]: shutting down MiniUPnPd
> Wed Feb 24 01:41:11 2021 daemon.notice miniupnpd[21078]: HTTP listening on port 5000
> Wed Feb 24 01:41:11 2021 daemon.notice miniupnpd[21078]: Listening for NAT-PMP/PCP traffic on port 5351
> Wed Feb 24 01:42:04 2021 daemon.err nlbwmon[3229]: Netlink receive failure: Out of memory
> Wed Feb 24 01:42:04 2021 daemon.err nlbwmon[3229]: Unable to dump conntrack: No buffer space available

coolsnowwolf, 2021-02-24:
内网有 ARP 病毒或者 环路了

IMconnor, 2021-02-24:
> 内网有 ARP 病毒或者 环路了

也许比特彗星比较魔性吧😄，我试了试发现这个软件一打开暴露出来的两个“问题”：占满CPU和报neighbor table overflow，分别对应独立的解决办法，两者好像没什么关联

- 取消勾选防火墙的Fullcone NAT可以使CPU占用恢复正常，
- 按链接 [https://www.cnblogs.com/rongfengliang/p/7141490.html](url) 中提供的思路，把内核相应参数调大就没警告了。
`root@OpenWrt:~# cat /etc/sysctl.conf`
`net.ipv4.neigh.default.gc_thresh1 = 512`
`net.ipv4.neigh.default.gc_thresh2 = 2048`
`net.ipv4.neigh.default.gc_thresh3 = 4096`

NAT1对游戏之类应用有加成，比特彗星这是逼用户在影音和游戏之间二选一吗…😜

IMconnor, 2021-02-25:
更新：
可能我软路由（升腾C92）有些菜，恢复其默认设置，即开启Fullcone NAT时，把比特彗星的高级参数作一下调整就没问题了

[image]

重要的是数值为100的参数，恢复默认则立即占满路由器资源，可以根据自己的情况调整。或者干脆在软件右下角小绿灯那里右键不允许加入DHT网络，这样应该对下载有影响？没测试。按说有tracker也没问题的。

另外，Fullcone NAT这种近乎完全暴露IP的技术虽然能解决IPv4 NAT带来的问题，但是在有UPnP的前提下，其对BT软件并没有什么加成，反而让BT软件最大限度地申请资源，倒过来影响路由器的性能，以致整个网络崩溃… 

所以目前来说，要么关闭Fullcone NAT；要么就调整比特彗星这个“不良”软件的参数。

closed 2021-02-25 by IMconnor