coolsnowwolf/lede #6428: ifdown 命令有时会调用失败

hahaleyile, 2021-02-24:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：有时家里断网，我想重启一下 wan 接口，但是命令不能百分百成功

### (2) 路由器型号和固件版本

A：
[image]

### (3) 详细日志

A：查看了一下日志，发现有时调用命令会返回 `Command failed: Request timed out`
Google 过，没有找到类似问题和解决方法
[image]

sxx1314, 2021-02-24:
- 可以尝试 ``` /etc/init.d/network restart ```
 - ifdown有些机型/网卡不一定可以执行

hahaleyile, 2021-02-24:
>  - 可以尝试 ``` /etc/init.d/network restart ```
>  - ifdown有些机型/网卡不一定可以执行

明白了，谢谢

hahaleyile, 2021-02-25:
>  - 可以尝试 ``` /etc/init.d/network restart ```
>  - ifdown有些机型/网卡不一定可以执行

不好意思，我看了一下 network 的实现，好像也是调用 ifdown 的😂
请问在 web 上点击连接后会调用什么命令吗？

sxx1314, 2021-02-25:
> > * 可以尝试 `/etc/init.d/network restart`
> > * ifdown有些机型/网卡不一定可以执行
> 
> 不好意思，我看了一下 network 的实现，好像也是调用 ifdown 的😂
> 请问在 web 上点击连接后会调用什么命令吗？

- 可以尝试直接用ifup不用ifdown
- 另外也可以尝试/etc/init.d/network reload
- 如果是pppoe可以尝试直接重启pppoe服务

hahaleyile, 2021-02-25:
> > > * 可以尝试 `/etc/init.d/network restart`
> > > * ifdown有些机型/网卡不一定可以执行
> > 
> > 不好意思，我看了一下 network 的实现，好像也是调用 ifdown 的😂
> > 请问在 web 上点击连接后会调用什么命令吗？
> 
> - 可以尝试直接用ifup不用ifdown
> - 另外也可以尝试/etc/init.d/network reload
> - 如果是pppoe可以尝试直接重启pppoe服务
> 
> 

谢谢，我想我使用第一种方法吧，重启 PPPoE 不知道该怎么做

closed 2021-06-17 by coolsnowwolf