coolsnowwolf/lede #6431: 选择luci-app-openvpn-server编译固件时会报错liblzo2.so.2

bleach1991, 2021-02-24:
具体问题
选择luci-app-openvpn-server编译固件时会报错liblzo2.so.2

路由器型号和固件版本
x86 ,r4s-1g,r4s-4g

 详细日志
install -m0644 files/openvpn.options /home/yxw/op/lede/x86/build_dir/target-x86_64_musl/openvpn-openssl/openvpn-2.5.0/ipkg-x86_64/openvpn-openssl/usr/share/openvpn/openvpn.options
install -m0600 files/openvpn.config /home/yxw/op/lede/x86/build_dir/target-x86_64_musl/openvpn-openssl/openvpn-2.5.0/ipkg-x86_64/openvpn-openssl/etc/config/openvpn
install -m0644 files/openvpn.upgrade /home/yxw/op/lede/x86/build_dir/target-x86_64_musl/openvpn-openssl/openvpn-2.5.0/ipkg-x86_64/openvpn-openssl/lib/upgrade/keep.d/openvpn
find /home/yxw/op/lede/x86/build_dir/target-x86_64_musl/openvpn-openssl/openvpn-2.5.0/ipkg-x86_64/openvpn-openssl -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
Package openvpn-openssl is missing dependencies for the following libraries:
liblzo2.so.2
make[3]: *** [Makefile:160: /home/yxw/op/lede/x86/bin/packages/x86_64/base/openvpn-openssl_2.5.0-1_x86_64.ipk] Error 1
make[3]: Leaving directory '/home/yxw/op/lede/x86/package/network/services/openvpn'
time: package/network/services/openvpn/openssl/compile#0.21#0.08#0.27
make[2]: *** [package/Makefile:113: package/network/services/openvpn/compile] Error 2
make[2]: Leaving directory '/home/yxw/op/lede/x86'
make[1]: *** [package/Makefile:107: /home/yxw/op/lede/x86/staging_dir/target-x86_64_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/yxw/op/lede/x86'
make: *** [/home/yxw/op/lede/x86/include/toplevel.mk:224: world] Error 2

jjit, 2021-02-24:
make menuconfig，Libraries，liblzo 有选中吗

bleach1991, 2021-02-25:
选中了，依旧不行

bleach1991, 2021-02-25:
> make menuconfig，Libraries，liblzo 有选中吗

> make menuconfig，Libraries，liblzo 有选中吗
选中了，依旧不行

bleach1991, 2021-02-25:
Network->VPN->openvpn-openssl->Enable LZO compression support
总算搞清楚了，luci-app-openvpn-server如果这里不选的话就会报错

closed 2021-03-14 by bleach1991