coolsnowwolf/lede #6434: esxi下配置虚拟机不正常断电后不能重启，需要修复磁盘！

magic-monkey, 2021-02-25:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

esxi下配置虚拟机不正常断电后不能重启，需要修复磁盘后才能启动。

sxx1314, 2021-02-25:
你应该找esxi

magic-monkey, 2021-02-25:
> 你应该找esxi

问题是之前用koolshare的lede固件不会出现这个情况😂但是自己编译的固件就会，试了不同的编译方式都会

bleach1991, 2021-03-10:
不要用变异生成的vmdk，容易出问题，自己用img去转换

magic-monkey, 2021-03-16:
> 不要用变异生成的vmdk，容易出问题，自己用img去转换

行，我试试，谢谢！

chaihx0857, 2021-06-07:
> 反馈bug/问题模板，提建议请删除
> 
> ## 1.关于你要提交的问题
> Q：是否搜索了issue (使用 "x" 选择)
> 
> * [x]  没有类似的issue
> 
> esxi下配置虚拟机不正常断电后不能重启，需要修复磁盘后才能启动。

我也遇到多次这个问题，爬网后按如下解决
1.vmkfstools -x check /path/to/your/machine.vmdk
先进行将报错的磁盘，进行查看
如果提示「Disk needs repair.」，执行下面的命令修复：
2.vmkfstools -x repair /path/to/your/machine.vmdk
这时候，虚拟机已经可以正常开机了，但是无法迁移操作，治标不治本进行操作磁盘格式转换为双副本。
3.vmkfstools -i source.vmdk destination.vmdk -d thin
该情况出现于所有vsan情况，如果磁盘为直接上传到存储内进行挂载的，均需要重新转换磁盘格式。

也就是说，编译完上传的vmdk，先用上述3转一下再用，我这没再出过问题

感谢lean大

closed 2021-06-17 by coolsnowwolf