coolsnowwolf/lede #6436: gmediarender编译后运行报错Segmentation fault

swq93, 2021-02-25:
反馈bug/问题模板，提建议请删除

## 1.gmediarender编译后运行报错Segmentation fault

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：正常编译gmediarender
Libraries --->
<*> libupnp

Sound --->
<*> alsa-utils
<*> madplay-alsa
-*- mpg123

Multimedia --->
gst1-libav
gstreamer1-libs
gstreamer1-plugins-base
gstreamer1-plugins-good
gstreamer1-plugins-ugly
gstreamer1-utils

Kernel modules --->

Sound Support --->
kmod-usb-audio

USB Support --->
kmod-usb-ohci
kmod-usb-storage
kmod-usb-storage-extras
kmod-usb-uhci

Native Language Support --->
kmod-nls-utf8
插件的选择，包括gmediarender插件。
启动gmediarender -I -f 或者/usr/sbi/gmediarender start 均报错Segmentation fault

### (2) 路由器型号和固件版本

A：x86软路由，D2550 CPU、虚拟机均测试，问题依旧，

### (3) 详细日志

A：gmediarender[4525]:segfault at 0 ip xxx sp xxx error 4 in libglib-2.0.so.5800.3

stock169, 2021-02-27:
这东西作者不维护了，要降低gst版本才行

467815891a, 2021-02-27:
alsa-utils不选吗？

swq93, 2021-02-28:
> alsa-utils不选吗？
Sound --->
<> alsa-utils
<> madplay-alsa
这两个都选了，星号没打上。

swq93, 2021-02-28:
> 这东西作者不维护了，要降低gst版本才行

还有其他的DLNA播放方案吗

swq93, 2021-02-28:
> alsa-utils不选吗？

声音--->
<>的alsa-utils的
<> madplay这样-alsa的
这两个都选了，星号没打上。

swq93, 2021-02-28:
> alsa-utils不选吗？

Sound --->
<> alsa-utils
<> madplay-alsa
这两个都选了，星号没打上

stock169, 2021-02-28:
> > 这东西作者不维护了，要降低gst版本才行
> 
> 还有其他的DLNA播放方案吗
upmpdcli

coolsnowwolf, 2021-03-01:
gst 版本的问题

swq93, 2021-03-01:
> > > 这东西作者不维护了，要降低gst版本才行
> > 
> > 
> > 还有其他的DLNA播放方案吗
> > upmpdcli

 upmpdcli编译报错，选了mdp-full。。。

swq93, 2021-03-01:
> gst 版本的问题

谢谢

stock169, 2021-03-02:
> > > > 这东西作者不维护了，要降低gst版本才行
> > > 
> > > 
> > > 还有其他的DLNA播放方案吗
> > > upmpdcli
> 
> upmpdcli编译报错，选了mdp-full。。。

是会报错还是版本问题，要换新版就能过

sxx1314, 2021-03-02:
[image]  DLNA 不好玩

swq93, 2021-03-03:
> [image] DLNA 不好玩

还有这个插件？有源码吗

swq93, 2021-03-03:
> > > > > 这东西作者不维护了，要降低gst版本才行
> > > > 
> > > > 
> > > > 还有其他的DLNA播放方案吗
> > > > upmpdcli
> > 
> > 
> > upmpdcli编译报错，选了mdp-full。。。
> 
> 是会报错还是版本问题，要换新版就能过

换最新的，果然就过了，谢谢

closed 2021-06-17 by coolsnowwolf