coolsnowwolf/lede #6441: 安装源自带的luci-app-dockerman后luci崩溃

shenmu33, 2021-02-26:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：安装自带源的app-luci-dockerman后，luci出现错误Insufficient permissions to read UCI configuration，几乎所有页面都出错读不出来，在网上查了一下，大概意思是luci版本太低？那应该装哪个版本的app-luci-dockerman?

### (2) 路由器型号和固件版本

A：J1900  R20.4.8-x64-Release

### (3) 详细日志

A：日志忘了保存，已经恢复出厂设置了，哪位大大遇过这个问题，直接告诉我解决方案吧，不想再重新弄一次了

sxx1314, 2021-02-26:
作者的项目下就直接就有预编译ipk。去那下载即可

shenmu33, 2021-02-26:
> 作者的项目下就直接就有预编译ipk。去那下载即可

那该装哪个版本才对？自带源的版本就是作者的最新版本，如果网上说的这个错误的原因是luci不兼容，那我装作者的就不会出错吗？你装的是哪个版本？

sxx1314, 2021-02-26:
- 自带源里并没有 dockerman。
- 另外dockerman自带依赖需要自行安装。

Wilson-coder, 2021-02-27:
直接把作者的git地址添加到feeds.conf.default里面再刷新，就可以在menuconfig里面选择了。另外记得去确认一下docker-ce依赖包是否有选上。

shenmu33, 2021-03-02:
> 直接把作者的git地址添加到feeds.conf.default里面再刷新，就可以在menuconfig里面选择了。另外记得去确认一下docker-ce依赖包是否有选上。

我说的是luci不兼容，不知道你说的是什么意思，你有试过吗？

caliban511, 2021-03-07:
> 
> 
> 直接把作者的git地址添加到feeds.conf.default里面再刷新，就可以在menuconfig里面选择了。另外记得去确认一下docker-ce依赖包是否有选上。

请教一下怎么加入docker-ce？
lean大佬的代码编译固件没有问题。
但编译lienol大什么的代码时，我在feeds.conf.default里加入以下几行，但在选择luci-app的那里，只有dockerman，并没有看到docker-ce：
src-git dockerce https://github.com/docker/docker-ce
src-git libdocker https://github.com/lisaac/luci-lib-docker
src-git dockerman https://github.com/lisaac/luci-app-dockerman
请教一下，正确把docker-ce加进去的方法是什么？

shenmu33, 2021-03-08:
> > 直接把作者的git地址添加到feeds.conf.default里面再刷新，就可以在menuconfig里面选择了。另外记得去确认一下docker-ce依赖包是否有选上。
> 
> 请教一下怎么加入docker-ce？
> lean大佬的代码编译固件没有问题。
> 但编译lienol大佬的代码时，我在feeds.conf.default里加入以下几行，但在选择luci-app的那里，只有dockerman，并没有看到docker-ce：
> src-git dockerce https://github.com/docker/docker-ce
> src-git libdocker https://github.com/lisaac/luci-lib-docker
> src-git dockerman https://github.com/lisaac/luci-app-dockerman
> 请教一下，正确把docker-ce加进去的方法是什么？

请教一下我装的是lean的最新固件，但是安装自带源的软件经常提示大小不一致无法安装，我下到本地强制安装，比如luci-app之类的软件后常常luci崩溃，在网上查了一下说是有些软件用的是新版的luci，而lean的lede是旧版luci，那应该怎么解决呢？要上哪找有旧版luci的luci-app-dockerman或者其他luci的app？

caliban511, 2021-03-08:
> 请教一下我装的是lean的最新固件，但是安装自带源的软件经常提示大小不一致无法安装，我下到本地强制安装，比如luci-app之类的软件后常常luci崩溃，在网上查了一下说是有些软件用的是新版的luci，而lean的lede是旧版luci，那应该怎么解决呢？要上哪找有旧版luci的luci-app-dockerman或者其他luci的app？

换openwrt官方源试试

closed 2021-06-17 by coolsnowwolf