coolsnowwolf/lede #6447: J4125+螃蟹8125B网卡总是link down

liuyanantj, 2021-02-28:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：一旦下载上传负载大了，软路由网卡就开始r8125: eth0: link down  间歇性的断开  eth0对应软路由lan口 然后这个口接后面的硬路由AP。造成间歇断网  不知道为啥了？从bios把asmp那个改成禁用问题依旧。

### (2) 路由器型号和固件版本

A：J4125+8125B众筹软路由

### (3) 详细日志

Sun Feb 28 20:56:47 2021 kern.info kernel: [  683.424348] NETDEV WATCHDOG: eth0 (r8125): transmit queue 0 timed out
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.424858] WARNING: CPU: 3 PID: 0 at dev_watchdog+0x21c/0x220
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.425280] Modules linked in: fast_classifier xt_FULLCONENAT ath9k ath9k_htc ath9k_common rt2800usb rt2800lib pppoe ppp_async iptable_nat ath9k_hw ath5k ath10k_pci ath10k_core ath xt_state xt_nat xt_conntrack xt_REDIRECT xt_MASQUERADE xt_CT rt2x00usb rt2x00lib pppox ppp_mppe ppp_generic nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mac80211 ipt_REJECT cfg80211 ax88179_178a asix amdgpu xt_time xt_tcpudp xt_socket xt_multiport xt_mark xt_mac xt_limit xt_comment xt_TPROXY xt_TCPMSS xt_LOG usbnet ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm slhc rtl8150 r8168 r8152 r8125 pcnet32 nf_tproxy_ipv6 nf_tproxy_ipv4 nf_socket_ipv6 nf_socket_ipv4
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.425371]  nf_reject_ipv4 nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 mmc_spi mlx5_core mlx4_en mlx4_core macvlan iptable_raw iptable_mangle iptable_filter ip_tables igc i40e gpu_sched forcedeth exfat e1000e crc7 crc_itu_t crc_ccitt compat bnx2 asn1_decoder 8139too 8139cp tcp_bbr evdev i2c_dev cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 bonding ixgbe igbvf igb e1000 mdio nls_utf8 shortcut_fe_ipv6 shortcut_fe crypto_user algif_skcipher algif_rng algif_hash algif_aead af_alg sha1_ssse3 sha1_generic authenc arc4 nls_iso8859_1 nls_cp437 vfat fat uas sdhci_pltfm antfs ptp pps_core mii libphy
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.437295] CPU: 3 PID: 0 Comm: swapper/3 Not tainted 5.4.100 #0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.437727] Hardware name: HARDKERNEL ODROID-H2/ODROID-H2, BIOS 5.13 07/27/2020
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.438259] RIP: 0010:dev_watchdog+0x21c/0x220
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.438584] Code: 00 49 63 4d e8 eb 92 4c 89 e7 c6 05 4d b0 b2 00 01 e8 a8 50 fc ff 89 d9 4c 89 e6 48 c7 c7 b0 be 07 82 48 89 c2 e8 fe d6 0f 00 <0f> 0b eb c0 55 48 89 e5 41 57 49 89 d7 41 56 49 89 ce 41 55 4d 89
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.439886] RSP: 0018:ffffc90000120e90 EFLAGS: 00010286
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.440262] RAX: 0000000000000039 RBX: 0000000000000000 RCX: 0000000000000006
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.440770] RDX: 0000000000000007 RSI: 0000000000000002 RDI: ffff888277d97340
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.441283] RBP: ffffc90000120ec0 R08: 0000000000000000 R09: 00000000000002ae
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.441789] R10: 0000000000000007 R11: 0000000000000000 R12: ffff8882730e0000
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.442284] R13: ffff8882730e0478 R14: 0000000000000004 R15: 0000000000000003
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.442794] FS:  0000000000000000(0000) GS:ffff888277d80000(0000) knlGS:0000000000000000
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.443366] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.443775] CR2: 00007f78eb1ce9c3 CR3: 0000000003208000 CR4: 0000000000340ee0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.444287] Call Trace:
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.444477]  <IRQ>
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.444636]  ? dev_reset_queue.constprop.49+0x80/0x80
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.445004]  call_timer_fn.isra.40+0x16/0x80
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.445317]  run_timer_softirq+0x164/0x180
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.445621]  ? lapic_next_deadline+0x27/0x30
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.445933]  ? clockevents_program_event+0x8b/0xf0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.446281]  __do_softirq+0xc9/0x20e
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.446539]  irq_exit+0x6f/0x80
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.446776]  smp_apic_timer_interrupt+0x5e/0x90
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.447106]  apic_timer_interrupt+0xf/0x20
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.447400]  </IRQ>
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.447561] RIP: 0010:cpuidle_enter_state+0x137/0x2c0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.447921] Code: 89 c6 e8 cc da a1 ff 45 84 ff 74 17 9c 58 0f 1f 44 00 00 f6 c4 02 0f 85 67 01 00 00 31 ff e8 10 9e a5 ff fb 66 0f 1f 44 00 00 <45> 85 ed 0f 88 c5 00 00 00 49 63 cd 4c 2b 75 d0 48 8d 04 49 48 c1
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.449229] RSP: 0018:ffffc90000093e68 EFLAGS: 00000246 ORIG_RAX: ffffffffffffff13
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.449761] RAX: ffff888277d9f740 RBX: ffffffff8224c8c0 RCX: 000000000000001f
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.450270] RDX: 0000009f1f3d9d70 RSI: 00000000401a41a4 RDI: 0000000000000000
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.450773] RBP: ffffc90000093e98 R08: 0000000000000002 R09: 000000000001efc0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.451283] R10: 0000013f07895a8a R11: 0000000000000000 R12: ffff88827615e800
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.451796] R13: 0000000000000003 R14: 0000009f1f3d9d70 R15: 0000000000000000
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.452309]  ? cpuidle_enter_state+0x114/0x2c0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.452632]  cpuidle_enter+0x2f/0x40
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.452900]  call_cpuidle+0x1e/0x40
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.453158]  do_idle+0xf4/0x1c0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.453391]  cpu_startup_entry+0x18/0x20
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.453677]  start_secondary+0x145/0x170
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.453965]  secondary_startup_64+0xa4/0xb0
Sun Feb 28 20:56:47 2021 kern.warn kernel: [  683.454267] ---[ end trace 9c1109184821deb5 ]---
Sun Feb 28 20:56:47 2021 kern.info kernel: [  683.946111] r8125: eth0: link up
Sun Feb 28 20:56:48 2021 kern.info kernel: [  684.959635] r8125: eth0: link down
Sun Feb 28 20:56:48 2021 daemon.notice netifd: Network device 'eth0' link is down
Sun Feb 28 20:56:48 2021 kern.info kernel: [  685.156027] br-lan: port 1(eth0) entered disabled state
Sun Feb 28 20:56:49 2021 daemon.notice netifd: bridge 'br-lan' link is down
Sun Feb 28 20:56:49 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
[... 3 lines cut ...]
Sun Feb 28 20:56:52 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Feb 28 20:56:52 2021 kern.info kernel: [  689.415300] br-lan: port 1(eth0) entered blocking state
Sun Feb 28 20:56:52 2021 kern.info kernel: [  689.415693] br-lan: port 1(eth0) entered forwarding state
Sun Feb 28 20:58:17 2021 kern.info kernel: [  773.764459] r8125: eth0: link up
Sun Feb 28 20:58:18 2021 kern.info kernel: [  774.777326] r8125: eth0: link down
Sun Feb 28 20:58:18 2021 daemon.notice netifd: Network device 'eth0' link is down
Sun Feb 28 20:58:18 2021 kern.info kernel: [  775.142647] br-lan: port 1(eth0) entered disabled state
Sun Feb 28 20:58:19 2021 daemon.notice netifd: bridge 'br-lan' link is down
Sun Feb 28 20:58:19 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Sun Feb 28 20:58:22 2021 kern.info kernel: [  779.392440] r8125: eth0: link up
[... cut ...]
Sun Feb 28 20:59:39 2021 kern.info kernel: [  856.347776] r8125: eth0: link up
Sun Feb 28 20:59:39 2021 kern.info kernel: [  856.348116] br-lan: port 1(eth0) entered blocking state
Sun Feb 28 20:59:39 2021 daemon.notice netifd: Network device 'eth0' link is up
Sun Feb 28 20:59:39 2021 daemon.notice netifd: bridge 'br-lan' link is up
Sun Feb 28 20:59:39 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Sun Feb 28 20:59:39 2021 kern.info kernel: [  856.348582] br-lan: port 1(eth0) entered forwarding state

EnnawYang, 2021-02-28:
似乎和我同一型号，我的没有这个问题，如果开了qbittorrent等下载软件 尝试把连接数开小一点

liuyanantj, 2021-02-28:
其实下载的速度还不快  或者是可能速度还没上来了就断了重新开始导致的   原来的软路由也是螃蟹网卡  但是也没遇到这个问题

coolsnowwolf, 2021-03-01:
看起来是连接表问题，尝试关闭 SFE 看是否有改进

liuyanantj, 2021-03-01:
> 看起来是连接表问题，尝试关闭 SFE 看是否有改进

关闭了还是存在问题  但是从外网这边访问现在基本不怎么感觉断网了  原来是连控制台都进不去
Mon Mar  1 13:44:16 2021 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  1 13:44:16 2021 kern.info kernel: [61127.843685] br-lan: port 1(eth0) entered disabled state
Mon Mar  1 13:44:17 2021 daemon.notice netifd: bridge 'br-lan' link is down
Mon Mar  1 13:44:17 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Mar  1 13:44:20 2021 kern.info kernel: [61132.102522] r8125: eth0: link up
Mon Mar  1 13:44:20 2021 kern.info kernel: [61132.102820] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 13:44:20 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 13:44:20 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Mar  1 13:44:20 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Mar  1 13:44:20 2021 kern.info kernel: [61132.103226] br-lan: port 1(eth0) entered forwarding state
Mon Mar  1 13:45:21 2021 kern.info kernel: [61192.677131] r8125: eth0: link up
Mon Mar  1 13:45:22 2021 kern.info kernel: [61193.689993] r8125: eth0: link down
Mon Mar  1 13:45:22 2021 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  1 13:45:22 2021 kern.info kernel: [61193.886643] br-lan: port 1(eth0) entered disabled state
Mon Mar  1 13:45:23 2021 daemon.notice netifd: bridge 'br-lan' link is down
Mon Mar  1 13:45:23 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Mar  1 13:45:26 2021 kern.info kernel: [61198.144732] r8125: eth0: link up
Mon Mar  1 13:45:26 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 13:45:26 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Mar  1 13:45:26 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Mar  1 13:45:27 2021 kern.info kernel: [61198.145418] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 13:45:27 2021 kern.info kernel: [61198.145803] br-lan: port 1(eth0) entered forwarding state
Mon Mar  1 13:46:33 2021 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  1 13:46:33 2021 kern.info kernel: [61264.439833] br-lan: port 1(eth0) entered disabled state
Mon Mar  1 13:46:34 2021 daemon.notice netifd: bridge 'br-lan' link is down
Mon Mar  1 13:46:34 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Mar  1 13:46:37 2021 kern.info kernel: [61268.954524] r8125: eth0: link up
Mon Mar  1 13:46:37 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 13:46:37 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Mar  1 13:46:37 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Mar  1 13:46:37 2021 kern.info kernel: [61268.955117] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 13:46:37 2021 kern.info kernel: [61268.955515] br-lan: port 1(eth0) entered forwarding state
Mon Mar  1 13:48:03 2021 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  1 13:48:03 2021 kern.info kernel: [61354.288121] br-lan: port 1(eth0) entered disabled state
Mon Mar  1 13:48:04 2021 daemon.notice netifd: bridge 'br-lan' link is down
Mon Mar  1 13:48:04 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Mar  1 13:48:08 2021 kern.info kernel: [61359.218595] r8125: eth0: link up
Mon Mar  1 13:48:08 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 13:48:08 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Mar  1 13:48:08 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Mar  1 13:48:08 2021 kern.info kernel: [61359.219498] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 13:48:08 2021 kern.info kernel: [61359.219839] br-lan: port 1(eth0) entered forwarding state
Mon Mar  1 13:50:15 2021 kern.info kernel: [61486.539128] r8125: eth0: link up
Mon Mar  1 13:50:16 2021 kern.info kernel: [61487.552196] r8125: eth0: link down
Mon Mar  1 13:50:16 2021 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  1 13:50:16 2021 kern.info kernel: [61487.918648] br-lan: port 1(eth0) entered disabled state
Mon Mar  1 13:50:17 2021 daemon.notice netifd: bridge 'br-lan' link is down
Mon Mar  1 13:50:17 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Mar  1 13:50:21 2021 kern.info kernel: [61492.166885] r8125: eth0: link up
Mon Mar  1 13:50:21 2021 kern.info kernel: [61492.167183] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 13:50:21 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 13:50:21 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Mar  1 13:50:21 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Mar  1 13:50:21 2021 kern.info kernel: [61492.167607] br-lan: port 1(eth0) entered forwarding state
Mon Mar  1 13:51:35 2021 daemon.notice netifd: Network device 'eth0' link is down
Mon Mar  1 13:51:35 2021 kern.info kernel: [61566.237610] br-lan: port 1(eth0) entered disabled state
Mon Mar  1 13:51:36 2021 daemon.notice netifd: bridge 'br-lan' link is down
Mon Mar  1 13:51:36 2021 daemon.notice netifd: Interface 'lan' has link connectivity loss
Mon Mar  1 13:51:40 2021 kern.info kernel: [61571.167978] r8125: eth0: link up
Mon Mar  1 13:51:40 2021 kern.info kernel: [61571.168309] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 13:51:40 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 13:51:40 2021 daemon.notice netifd: bridge 'br-lan' link is up
[... cut ...]
Mon Mar  1 14:26:43 2021 kern.info kernel: [63674.695060] r8125: eth0: link up
Mon Mar  1 14:26:43 2021 kern.info kernel: [63674.695352] br-lan: port 1(eth0) entered blocking state
Mon Mar  1 14:26:43 2021 daemon.notice netifd: Network device 'eth0' link is up
Mon Mar  1 14:26:43 2021 daemon.notice netifd: bridge 'br-lan' link is up
Mon Mar  1 14:26:43 2021 daemon.notice netifd: Interface 'lan' has link connectivity
Mon Mar  1 14:26:43 2021 kern.info kernel: [63674.695805] br-lan: port 1(eth0) entered forwarding state

helianling, 2021-03-03:
可以尝试以下两种方法:

1. 在当前驱动下：把BIOS中的ASPM设置由默认的AUTO改成Disabled或L0s。
2. 不改BIOS，更新驱动。

可以尝试用r8125B最新的9.005.01或9.005.02驱动

https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software
[image]
[image]

liuyanantj, 2021-03-03:
> 看起来是连接表问题，尝试关闭 SFE 看是否有改进

大雕，楼下说 可以尝试用r8125B最新的9.005.01或9.005.02驱动  大雕能不能帮忙更新下  自己不会更新  主板的BIOS已经修改成Disabled的了

liuyanantj, 2021-03-03:
> 可以尝试以下两种方法:
> 
> 1. 在当前驱动下：把BIOS中的ASPM设置由默认的AUTO改成Disabled或L0s。
> 2. 不改BIOS，更新驱动。
> 
> 可以尝试用r8125B最新的9.005.01或9.005.02驱动
> 
> https://www.realtek.com/en/component/zoo/category/network-interface-controllers-10-100-1000m-gigabit-ethernet-pci-express-software
> [image]
> [image]

修改BIOS有改善 但是还是会有日志 估计得等大雕更新驱动看看效果了

coolsnowwolf, 2021-03-03:
已更新

liuyanantj, 2021-03-05:
> 已更新

感谢大佬 今天重新编译了  bios改成Disabled  已经没有原来的错误日志了

closed 2021-06-17 by coolsnowwolf