coolsnowwolf/lede #6450: mt7621，移动宽带断网后拨号老是失败

xiangfeigao, 2021-03-01:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [* ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题
mt7621，移动宽带是光纤桥接模式，通过路由器进行拨号，当移动网络断网后，看到路由器不断拨号，但就是拨不成功。
A：

### (2) 路由器型号和固件版本

A：B70 R21.2.1

### (3) 详细日志

A：

KevinChan8699, 2021-03-01:
网络，负载均衡，策略，默认（使用主路由表）

venpong, 2021-03-10:
同B70，遇到相同的问题，联通宽带断网后，路由器无法重新拨号，只有重启路由器或者在接口页面手动点击连接按钮才行

cyijun, 2021-03-12:
可以加一个脚本用cron定期执行，亲测有效
`#!/bin/sh
tries=0
echo --- my_watchdog start ---
while [[ $tries -lt 5 ]]
do
        if /bin/ping -c 1 223.5.5.5 >/dev/null
        then
            echo --- my_watchdog exit ---
            exit 0
        fi
        tries=$((tries+1))
        sleep 10
done

echo --- my_watchdog restart wan ---

ifup wan
#/etc/init.d/network restart
exit 0
`

xiangfeigao, 2021-03-18:
已修好，把lcp时间间隔改为15

closed 2021-03-18 by xiangfeigao