coolsnowwolf/lede #6453: 可以加入chelsio t422-cr驱动吗?性价比高的万兆网卡

lqhorochi, 2021-03-02:
可以加入chelsio t422-cr驱动吗?

性价比高的万兆网卡,双口万兆光口+双口千兆电口

esxi上就有网卡驱动,OpenWrt没有驱动,

Chelsio官网驱动下载
https://service.chelsio.com/

Aes64X, 2021-03-03:
> 可以加入chelsio t422-cr驱动吗?
> 
> 性价比高的万兆网卡,双口万兆光口+双口千兆电口
> 
> esxi上就有网卡驱动,OpenWrt没有驱动,
> 
> Chelsio官网驱动下载
> https://service.chelsio.com/

这。。。。OCP转的那票CX3系列不比这个高？驱动早就有了。单口一块全套才75

Aes64X, 2021-03-03:
还有啊。。。这家的卡散热片都是不够用的，需要加风冷。你说的T4系列还是PCIE 2.0

lqhorochi, 2021-03-03:
> 还有啊。。。这家的卡散热片都是不够用的，需要加风冷。你说的T4系列还是PCIE 2.0
> servethehome上有用T5系列，开机10分钟直接过热硬件离线的

现在esxi可以识别,可以直通,但是爱快,OpenWrt,秒开都没有驱动起来.

lqhorochi, 2021-03-03:
> > 可以加入chelsio t422-cr驱动吗?
> > 性价比高的万兆网卡,双口万兆光口+双口千兆电口
> > esxi上就有网卡驱动,OpenWrt没有驱动,
> > Chelsio官网驱动下载
> > https://service.chelsio.com/
> 
> 这。。。。OCP转的那票CX3系列不比这个高？驱动早就有了。单口一块全套才75，双口全套也就100+

这...你不懂...chelsio t422-cr是双光口万兆+双电口千兆.

Aes64X, 2021-03-03:
> > 还有啊。。。这家的卡散热片都是不够用的，需要加风冷。你说的T4系列还是PCIE 2.0
> > servethehome上有用T5系列，开机10分钟直接过热硬件离线的
> 
> 现在esxi可以识别,可以直通,但是爱快,OpenWrt,秒开都没有驱动起来.
> 你有相关的卡吗?能不能正常使用?还是直接虚拟化?

指路：你可以试试直接在内核启用驱动，实际上内核自带很多驱动，只是都没启用，常见的是用了kmod内核模块形式加载。直接编译进内核至少可保证两个万兆口使用，至于那两个千兆口。。不好说，目前没资料显示那两个千兆口是怎么弄出来的，用的什么芯片，是否涉及到PCIE拆分。（就像DELL服务器部件那种双千兆双万兆的OCP卡，OCP转接后在不支持PCIE拆分的主板上智能识别到两个万兆口）

lqhorochi, 2021-03-03:
> 指路：你可以试试直接在内核启用驱动，实际上内核自带很多驱动，只是都没启用，常见的是用了kmod内核模块形式加载。直接编译进内核至少可保证两个万兆口使用，至于那两个千兆口。。不好说，目前没资料显示那两个千兆口是怎么弄出来的，用的什么芯片，是否涉及到PCIE拆分。（就像DELL服务器部件那种双千兆双万兆的OCP卡，OCP转接后在不支持PCIE拆分的主板上智能识别到两个万兆口）

2个千兆口用不到,esxi识别不到2个千兆口

Aes64X, 2021-03-04:
> 2个千兆口用不到,esxi识别不到2个千兆口

esxi自带驱动都识别不到千兆口那就是需要主板PCIE拆分，就这一条门槛就已经一大堆主板不能用了，这网卡就毫无性价比。。。。。（较新的AMD芯片组和intel Z系列芯片组，服务器芯片组支持PCIE拆分）

lqhorochi, 2021-03-04:
> esxi自带驱动都识别不到千兆口那就是需要主板PCIE拆分，就这一条门槛就已经一大堆主板不能用了，这网卡就毫无性价比。。。。。（较新的AMD芯片组和intel Z系列芯片组，服务器芯片组支持PCIE拆分）

b450m自带拆分...我居然发现不到这个问题,..这网卡需要拆分的吗?

Aes64X, 2021-03-04:
> b450m自带拆分...我居然发现不到这个问题,..这网卡需要拆分的吗?

你打开拆分试试，八成是这样的。如果不需要拆分一定会是单芯片，T420单芯片集成双万兆双千兆这个不现实，至少我没见过这种方案。如果不需要拆分，那一定要有PLX片上桥接，我看T420那个外观。。。。感觉不大可能有PLX芯片。

lqhorochi, 2021-03-04:
> 你打开拆分试试，八成是这样的。如果不需要拆分一定会是单芯片，T420单芯片集成双万兆双千兆这个不现实，至少我没见过这种方案。如果不需要拆分，那一定要有PLX片上桥接，我看T420那个外观。。。。感觉不大可能有PLX芯片。

可惜退货了,我也想知道如何玩这卡.应该x4+x4来玩,

closed 2021-03-10 by lqhorochi