coolsnowwolf/lede #6459: BUG open-vm-tools有内存泄漏问题，上游已修复了，配置里的版本还是老的

0987363, 2021-03-03:
open-vm-tools有内存泄漏问题，上游的issue
https://github.com/vmware/open-vm-tools/issues/452

在11.2.0以后已经修复
https://github.com/vmware/open-vm-tools/releases

需要更新本地第三方库配置

x86下，内核崩溃日志
```
[607017.200530] sh invoked oom-killer: gfp_mask=0x400dc0(GFP_KERNEL_ACCOUNT|__GFP_ZERO), order=1, oom_score_adj=0
[607017.200770] CPU: 2 PID: 14785 Comm: sh Not tainted 5.4.95 #0
[607017.200890] Hardware name: VMware, Inc. VMware Virtual Platform/440BX Desktop Reference Platform, BIOS 6.00 05/28/2020
[607017.201105] Call Trace:
[607017.201168]  dump_stack+0x6d/0x8b
[607017.201245]  dump_header+0x4a/0x18a
[607017.201324]  oom_kill_process.cold.27+0xb/0x10
[607017.201422]  out_of_memory+0x1aa/0x230
[607017.201507]  __alloc_pages_slowpath+0xa65/0xbb0
[607017.201607]  __alloc_pages_nodemask+0x1ea/0x220
[607017.201706]  __get_free_pages+0x10/0x70
[607017.201793]  pgd_alloc+0x22/0x210
[607017.201869]  ? sched_clock+0x9/0x10
[607017.201950]  mm_init+0x177/0x220
[607017.202024]  dup_mm+0x54/0x450
[607017.202096]  ? kmem_cache_alloc+0x20/0x1a0
[607017.202187]  copy_process+0x10c6/0x1760
[607017.202274]  _do_fork+0x71/0x2b0
[607017.202349]  ? recalc_sigpending+0x16/0x40
[607017.202439]  ? __set_task_blocked+0x5e/0x70
[607017.202538]  __x64_sys_fork+0x26/0x30
[607017.202622]  do_syscall_64+0x50/0x2c0
[607017.202705]  entry_SYSCALL_64_after_hwframe+0x44/0xa9
[607017.202814] RIP: 0033:0x7f951b3ee426
[607017.202896] Code: Bad RIP value.
[607017.202970] RSP: 002b:00007fffad501220 EFLAGS: 00000246 ORIG_RAX: 0000000000000039
[607017.203131] RAX: ffffffffffffffda RBX: 0000000000000002 RCX: 00007f951b3ee426
[607017.203284] RDX: 00007fffad501220 RSI: 00007f951b41bc00 RDI: 0000000000000000
[607017.203436] RBP: 00007fffad501220 R08: 0101010101010101 R09: 0000000000000000
[607017.203587] R10: 0000000000000008 R11: 0000000000000246 R12: 00000000017de6e8
[607017.203739] R13: 00007f951b3ad6d0 R14: 00007f951b422360 R15: 00000000017ded78
[607017.203908] Mem-Info:
[607017.203967] active_anon:187456 inactive_anon:39054 isolated_anon:0
[607017.203967]  active_file:40 inactive_file:51 isolated_file:0
[607017.203967]  unevictable:0 dirty:0 writeback:0 unstable:0
[607017.203967]  slab_reclaimable:1312 slab_unreclaimable:7114
[607017.203967]  mapped:103 shmem:55580 pagetables:518 bounce:0
[607017.203967]  free:5612 free_pcp:187 free_cma:0
[607017.204602] Node 0 active_anon:749824kB inactive_anon:156216kB active_file:160kB inactive_file:404kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:412kB dirty:0kB writeback:0kB shmem:222320kB writeback_tmp:0kB unstable:0kB all_unreclaimable? yes
[607017.205051] DMA free:4124kB min:256kB low:320kB high:384kB active_anon:8980kB inactive_anon:0kB active_file:4kB inactive_file:12kB unevictable:0kB writepending:0kB present:15992kB managed:15908kB mlocked:0kB kernel_stack:40kB pagetables:4kB bounce:0kB free_pcp:0kB local_pcp:0kB free_cma:0kB
[607017.205548] lowmem_reserve[]: 0 967 967 967
[607017.205642] DMA32 free:17316kB min:16124kB low:20152kB high:24180kB active_anon:740744kB inactive_anon:156216kB active_file:0kB inactive_file:236kB unevictable:0kB writepending:0kB present:1032064kB managed:995240kB mlocked:0kB kernel_stack:3232kB pagetables:2068kB bounce:0kB free_pcp:1236kB local_pcp:0kB free_cma:0kB
[607017.206186] lowmem_reserve[]: 0 0 0 0
[607017.206268] DMA: 44*4kB (UME) 21*8kB (UME) 8*16kB (UME) 20*32kB (UE) 19*64kB (UE) 6*128kB (UE) 2*256kB (E) 1*512kB (E) 0*1024kB 0*2048kB 0*4096kB = 4120kB
[607017.206548] DMA32: 810*4kB (UMH) 384*8kB (UMH) 142*16kB (UMEH) 58*32kB (UMEH) 21*64kB (UMEH) 22*128kB (UMEH) 10*256kB (UME) 2*512kB (UM) 0*1024kB 0*2048kB 0*4096kB = 18184kB
[607017.206857] 55824 total pagecache pages
[607017.206942] 0 pages in swap cache
[607017.207017] Swap cache stats: add 0, delete 0, find 0/0
[607017.207154] Free swap  = 0kB
[607017.207222] Total swap = 0kB
[607017.207289] 262014 pages RAM
[607017.207355] 0 pages HighMem/MovableOnly
[607017.207440] 9227 pages reserved
[607017.207511] Tasks state (memory values in pages):
[607017.207611] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
[607017.207792] [   2109]     0  2109      220       10    40960        0             0 ubusd
[607017.207960] [   2111]     0  2111      179        8    32768        0             0 askfirst
[607017.208134] [   2139]     0  2139      193        9    36864        0             0 urngd
[607017.208303] [   2598]     0  2598   152346   151148  1257472        0             0 vmtoolsd
[607017.208476] [   2676]     0  2676      239       32    36864        0             0 logd
[607017.208642] [   2724]     0  2724      266       25    36864        0             0 rpcd
[607017.208809] [   2985]     0  2985      535       25    40960        0             0 hostapd
[607017.208980] [   2986]     0  2986      535       25    40960        0             0 wpa_supplicant
[607017.209163] [   3088]     0  3088      345       76    45056        0             0 netifd
[607017.209332] [   3483]     0  3483      275       40    40960        0             0 uhttpd
[607017.209506] [   4040]     0  4040      313       33    45056        0             0 nlbwmon
[607017.209693] [   4111]     0  4111      720       79    36864        0             0 smbd
[607017.209879] [   4112]     0  4112      725       31    36864        0             0 nmbd
[607017.210069] [   4154]     0  4154      185        9    36864        0             0 vlmcsd
[607017.210259] [   4310]     0  4310      223        9    40960        0             0 dropbear
[607017.210704] [   5608]     0  5608      275       13    40960        0             0 sh
[607017.210889] [   5646]     0  5646      272       12    32768        0             0 ntpd
[607017.211089] [   5850]     0  5850      275        7    40960        0             0 sh
[607017.211273] [   5864]     0  5864      515      256    45056        0             0 awk
[607017.211458] [   5865]     0  5865      272       11    36864        0             0 tail
[607017.211644] [   5991]     0  5991      186        8    36864        0             0 wsdd2
[607017.211832] [   6046]     0  6046     3724     1083    73728        0             0 zerotier-one
[607017.212032] [  26812]     0 26812     2534      360    57344        0             0 haproxy
[607017.212224] [  26857]     0 26857     2897      584    61440        0             0 haproxy
[607017.212416] [  14785]     0 14785      360      103    45056        0             0 sh
[607017.212598] [  30634]     0 30634  1229835     5499   143360        0             0 v2ray
[607017.212786] [  30676]     0 30676  1230142     6802   155648        0             0 v2ray
[607017.212975] [  31467] 65534 31467      346       23    36864        0             0 pdnsd
[607017.213164] [  31576]     0 31576      315       54    40960        0             0 sh
[607017.213346] [  17818]     0 17818      273       12    36864        0             0 crond
[607017.213539] [  18077]   453 18077     3511     3259    65536        0             0 dnsmasq
[607017.213767] [  18263]     0 18263      811      236    40960        0             0 adbyby
[607017.213958] [  18264]     0 18264     1796      809    49152        0             0 adbyby
[607017.214149] [  27775]     0 27775      272        6    40960        0             0 sleep
[607017.214338] [  28262]     0 28262      315       54    40960        0             0 sh
[607017.214522] [  28266]     0 28266      316       54    40960        0             0 sh
[607017.214705] [  28267]     0 28267      316       54    40960        0             0 sh
[607017.214889] oom-kill:constraint=CONSTRAINT_NONE,nodemask=(null),cpuset=/,mems_allowed=0,global_oom,task_memcg=/,task=vmtoolsd,pid=2598,uid=0
[607017.215205] Out of memory: Killed process 2598 (vmtoolsd) total-vm:609384kB, anon-rss:604592kB, file-rss:0kB, shmem-rss:0kB, UID:0 pgtables:1228kB oom_score_adj:0
[607017.235926] oom_reaper: reaped process 2598 (vmtoolsd), now anon-rss:0kB, file-rss:0kB, shmem-rss:0kB
```

greennyreborn, 2021-03-08:
https://github.com/coolsnowwolf/packages

应该到这里去提PR

greennyreborn, 2021-03-15:
https://github.com/coolsnowwolf/packages/pull/135

closed 2021-06-17 by coolsnowwolf