coolsnowwolf/lede #6466: Rsapberry Pi3B  编译出错 missing go.sum entry; to add it:

AmibaWorld, 2021-03-04:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：编译出错

### (2) 路由器型号和固件版本

A：Rsapberry Pi3B 

### (3) 详细日志

Finding targets
go: v2ray.com/core@v4.19.1+incompatible (replaced by github.com/v2fly/v2ray-core@v0.0.0-20210104111944-a6efb4d60b86): missing go.sum entry; to add it:
	go mod download v2ray.com/core

Building targets
go: v2ray.com/core@v4.19.1+incompatible (replaced by github.com/v2fly/v2ray-core@v0.0.0-20210104111944-a6efb4d60b86): missing go.sum entry; to add it:
	go mod download v2ray.com/core

Makefile:74: recipe for target '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-plugin-1.8.0/.built' failed
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/v2ray-plugin-1.8.0/.built] Error 1
make[2]: *** [package/lean/v2ray-plugin/compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/package/lean/v2ray-plugin'
time: package/lean/v2ray-plugin/compile#0.17#0.10#0.27
package/Makefile:111: recipe for target 'package/lean/v2ray-plugin/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
make: *** [world] Error 2
Error: Process completed with exit code 2.

DIUDUANG, 2021-03-04:
我也碰到一样的问题

vincent0214, 2021-03-05:
同上

Pantheon-Paradoxes, 2021-03-05:
x86 +1

wang1zhen, 2021-03-05:
X86_64遇到了同样的问题，可能与()[https://github.com/coolsnowwolf/lede/commit/73fdfdf6aa62f0b28535b036313266f2ab06e993]该commit有关，对应不上正确的v2ray版本号。

Guxing0321, 2021-03-05:
x86 +1

Nicepaul, 2021-03-05:
同上

PR #6470 (2021-03-05): v2ray-plugin: update to 4.35.0
fix: #6466 , fix: #6469
closed 2021-03-05 by coolsnowwolf