coolsnowwolf/lede #6473: 20210305编译失败

ITdesk01, 2021-03-05:
看起来像go的问题，你们正常不

Finding targets
no required module provides package github.com/docker/libnetwork/cmd/proxy: working directory is not part of a module
no required module provides package github.com/docker/libnetwork/cmd/dnet: working directory is not part of a module

Building targets
go: cannot find main module, but found vendor.conf in /home/zhang/Openwrt/1/lede/build_dir/target-x86_64_musl/libnetwork-2020-06-17-026aabaa
	to create a module there, run:
	cd ../.. && go mod init

Makefile:49: recipe for target '/home/zhang/Openwrt/1/lede/build_dir/target-x86_64_musl/libnetwork-2020-06-17-026aabaa/.built' failed
make[3]: *** [/home/zhang/Openwrt/1/lede/build_dir/target-x86_64_musl/libnetwork-2020-06-17-026aabaa/.built] Error 1
make[3]: Leaving directory '/home/zhang/Openwrt/1/lede/feeds/packages/utils/libnetwork'

tianiue, 2021-03-05:
一样编译不了

owen-gu, 2021-03-05:
本地编译成功， ACTION编译失败

EnnawYang, 2021-03-05:
就是golang升级了的问题

ITdesk01, 2021-03-05:
> 看起来像go的问题，你们正常不
>
> Finding targets
> no required module provides package github.com/docker/libnetwork/cmd/proxy: working directory is not part of a module
> no required module provides package [...]

这个问题我解决了
svn checkout https://github.com/openwrt/packages/trunk/utils/libnetwork packages/utils

ITdesk01, 2021-03-05:
然后新问题又来了
# github.com/gorilla/websocket
../../../../../dl/go-mod-cache/github.com/gorilla/websocket@v1.4.2/client.go:308:10: undefined: cloneTLSConfig
../../../../../dl/go-mod-cache/github.com/gorilla/websocket@v1.4.2/client.go:317:10: undefined: doHandshakeWithTrace
../../../../../dl/go-mod-cache/github.com/gorilla/websocket@v1.4.2/conn.go:393:10: c.writeBufs undefined (type *Conn has no field or method writeBufs)

Makefile:143: recipe for target '/home/zhang/Openwrt/1/lede/build_dir/target-x86_64_musl/Xray-core-1.3.1/.built' failed
make[3]: *** [/home/zhang/Openwrt/1/lede/build_dir/target-x86_64_musl/Xray-core-1.3.1/.built] Error 2
make[3]: Leaving directory '/home/zhang/Openwrt/1/lede/feeds/helloworld/xray-core'

breakings, 2021-03-05:
看着像是golang和docker方面的问题，期待大佬们PR
[image]

nature2608, 2021-03-05:
一样，打开makefile看了下里面的url地址打开已经不存在了

zeroshine0, 2021-03-05:
对，同样的问题

nature2608, 2021-03-07:
> 一样，打开makefile看了下里面的url地址打开已经不存在了

已经修复好了，可以关闭issus了 @ITdesk01

ITdesk01, 2021-03-07:
> > 一样，打开makefile看了下里面的url地址打开已经不存在了
> 
> 已经修复好了，可以关闭issus了 @ITdesk01

嗯嗯，已经OK了

closed 2021-03-07 by ITdesk01