coolsnowwolf/lede #6475: 正常用一段时间后登陆luci出错：Failed to execute function dispatcher target for entry '/'.

iseeyou, 2021-03-06:
错误信息：
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute function dispatcher target for entry '/'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute firstchild dispatcher target for entry '/admin'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute function dispatcher target for entry '/admin/status'.
The called action terminated with an exception:
/usr/lib/lua/luci/dispatcher.lua:469: Failed to execute template dispatcher target for entry '/admin/status/overview'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'admin_status/index'.
A runtime error occured: /usr/lib/lua/luci/util.lua:588: attempt to index local 'pp' (a nil value)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:469: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

coolsnowwolf, 2021-03-07:
attempt to index local 'pp' (a nil value) ，看看是否用了其他的luci 主页，例如显示天气之类

iseeyou, 2021-03-07:
> attempt to index local 'pp' (a nil value) ，看看是否用了其他的luci 主页，例如显示天气之类

没有，就一个bootstrap主题。
这个错误是登陆后立刻出现，路由器一切正常我就没管，过一天登陆就好了。

closed 2021-06-17 by coolsnowwolf