coolsnowwolf/lede #6477: 编译docker出错

yonggedebaqi, 2021-03-06:
Q：是否搜索了issue (使用 "x" 选择)

搜索找不到

### (1) 具体问题

编译docker出错

### (2) 路由器型号和固件版本

N1 固件

### (3) 详细日志

no required module provides package github.com/docker/cli/cmd/docker: working directory is not part of a module
Makefile:138: recipe for target '/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/docker-ce-19.03.12/.built' failed
make[3]: *** [/workdir/openwrt/build_dir/target-aarch64_cortex-a53_musl/docker-ce-19.03.12/.built] Error 1
make[3]: Leaving directory '/workdir/openwrt/feeds/packages/utils/docker-ce'
time: package/feeds/packages/docker-ce/compile#0.13#0.05#0.18
make[2]: *** [package/feeds/packages/docker-ce/compile] Error 2
package/Makefile:111: recipe for target 'package/feeds/packages/docker-ce/compile' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile] Error 2
make[2]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

EnnawYang, 2021-03-06:
别着急 还没修完（

1715173329, 2021-03-06:
see coolsnowwolf/packages#129

closed 2021-03-06 by coolsnowwolf