coolsnowwolf/lede #6478: 新三路由报错, Cannot install package kmod-mt7603e.

vincent0214, 2021-03-06:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

### (2) 路由器型号和固件版本

A：Newifi D2(新三路由) 

### (3) 详细日志

A：   
2021-03-06T04:49:21.3063034Z Configuring kmod-mmc.
2021-03-06T04:49:21.3063641Z Configuring wireless-tools.
2021-03-06T04:49:21.3064108Z Configuring mt_wifi.
2021-03-06T04:49:21.3064515Z Configuring uci.
2021-03-06T04:49:21.4306053Z Collected errors:
2021-03-06T04:49:21.4307177Z Configuring kmod-ipt-offload.
2021-03-06T04:49:21.4331791Z  * check_data_file_clashes: Package kmod-mt7603e wants to install file /home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/build_dir/target-mipsel_24kc_musl/root-ramips/lib/modules/5.4.101/mt7603e.ko
2021-03-06T04:49:21.4334404Z    **But that file is already provided by package  * kmod-mt7603**
2021-03-06T04:49:21.4335078Z Configuring kmod-tcp-bbr.
2021-03-06T04:49:21.4337931Z  * opkg_install_cmd: **Cannot install package kmod-mt7603e.**
2021-03-06T04:49:21.4338850Z Configuring luci-app-flowoffload.
2021-03-06T04:49:21.4340836Z make[2]: *** [package/install] Error 255
2021-03-06T04:49:21.4341919Z Configuring luci-i18n-flowoffload-zh-cn.
2021-03-06T04:49:21.4343132Z Configuring strongswan-mod-hmac.
2021-03-06T04:49:21.4343772Z Configuring dropbear.
2021-03-06T04:49:21.4344378Z Configuring luci-theme-argon.
2021-03-06T04:49:21.4345619Z make[1]: *** [/home/runner/work/AutoBuild-OpenWrt/AutoBuild-OpenWrt/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_install] Error 2
2021-03-06T04:49:21.4347026Z Configuring libcurl4.
2021-03-06T04:49:21.4347417Z Configuring curl.

minlang112, 2021-03-06:
kmod-mt7603 和 kmod-mt7603 这两个内核模块冲突，检查一下menuconfig

acooler15, 2021-03-07:
kmod-mt76是开源驱动吧？若用开源驱动就把闭源驱动取消掉，可以参考这个 #6341

vincent0214, 2021-03-08:
多谢各位提醒,后来修改配置文件.删除以下代码可以解决问题
`
CONFIG_PACKAGE_kmod-mt76=y
CONFIG_PACKAGE_kmod-sdhci=y
CONFIG_PACKAGE_kmod-sdhci-mt7620=y
`

closed 2021-03-08 by vincent0214