coolsnowwolf/lede #6480: x86编译后的固件，用physdiskwrite写入硬盘，报错

HUiTHEME, 2021-03-06:
我也是从 google上 搜索 physdiskwrite 报错 writtenwrite error after 的问题，看到了有人反馈，是因为 自编译Lean固件 引起的，似乎和我的情况是一样的。所以，遇到这样的情况，应该如何解决鸭？

leungyauming, 2021-03-06:
有沒有報錯信息

HUiTHEME, 2021-03-06:
> 有沒有報錯信息

如这个帖子描述 https://www.right.com.cn/forum/thread-4065175-1-1.html

leungyauming, 2021-03-06:
試一下用.gz格式的鏡像

HUiTHEME, 2021-03-06:
编译后的包里好像也没有.gz的包，而且，以前我也是用 physdiskwrite 把.img 写进硬盘的，以前都好好的，就是这次，写了会报错。。。

wifiway520, 2021-03-06:
是的，我也是持续报错，写入U盘没问题，写入sd卡要么卡在openwrt界面，要么没有启动项

HUiTHEME, 2021-03-06:
> 是的，我也是持续报错，写入U盘没问题，写入sd卡要么卡在openwrt界面，要么没有启动项

对，很明确的，以前都是用 physdiskwrite 把.img 写进硬盘，就是最近编译的包，无法写入，报错。。。

wifiway520, 2021-03-06:
另外借地问下，是不支持sd卡启动了吗，我用koolshare的lede可以直接写盘成功和启动，最近的lean大的怎么都不能启动

HUiTHEME, 2021-03-06:
> 另外借地问下，是不支持sd卡启动了吗，我用koolshare的lede可以直接写盘成功和启动，最近的lean大的怎么都不能启动

我还没试过SD卡，就是  用 physdiskwrite 把 .img 写进硬盘  现在写不进去了。。。

leungyauming, 2021-03-06:
> 编译后的包里好像也没有.gz的包，而且，以前我也是用 physdiskwrite 把.img 写进硬盘的，以前都好好的，就是这次，写了会报错。。。

試一下把.img直接打包進一個.gz

mysteryman2045, 2021-03-06:
弃用physdiskwrite，改用dd if of 刷固件，免去你以上问题。

HUiTHEME, 2021-03-07:
dd if of 怎么刷固件？
硬盘是MSATA，得插软路由上，软路由先装个ubuntu系统来跑dd if of命令？那没有U盘跑 PE运行 physdiskwrite 方便啊。。。

dd if of命令，我大概想到的应该是  买个硬盘盒插 ubuntu机器上来做吧，那，还得去买个硬盘盒。。。。

ziyangdaoren, 2021-03-07:
同样的问题

EnnawYang, 2021-03-07:
https://github.com/coolsnowwolf/lede/issues/5284  
https://github.com/coolsnowwolf/lede/issues/5602  
建议使用DiskImage V1.2版本，或者换别的工具

HUiTHEME, 2021-03-07:
全网找不到 DiskImage V1.2，找了个v.16的，写入报错。。。我再找，非找到  v1.2的看看

HUiTHEME, 2021-03-07:
找到DiskImage V1.2了，同样报错，报错信息：Error #0 occured while writting to disk at sector ******

其实这根本就不是换个工具的问题，以前，用 physdiskwrite 把.img 写进硬盘，一点问题都没有。

31415926pai, 2021-03-07:
刷入以前的版本，然后新版本升级上去看怎么样。

martin0607, 2021-03-07:
虽然报错但是其实还是能正常启动的，当初写盘报Error #0 occured while writting to disk at sector ******，但是硬盘插上能进系统，用了两个月一点毛病都没有

mysteryman2045, 2021-03-07:
> dd if of 怎么刷固件？
> 硬盘是MSATA，得插软路由上，软路由先装个ubuntu系统来跑dd if of命令？那没有U盘跑 PE运行 physdiskwrite 方便啊。。。
> 
> dd if of命令，我大概想到的应该是 买个硬盘盒插 ubuntu机器上来做吧，那，还得去买个硬盘盒。。。。

dd if of非得在ubuntu下执行？你有U盘做PE，为啥不能用U盘先做openwrt。openwrt下用dd if of固件刷到MSATA。之后就一劳永逸了。用PE盘，每次还得接显示器，键盘，鼠标，这才是真的麻烦。

HUiTHEME, 2021-03-08:
> > dd if of 怎么刷固件？
> > 硬盘是MSATA，得插软路由上，软路由先装个ubuntu系统来跑dd if of命令？那没有U盘跑 PE运行 physdiskwrite 方便啊。。。
> > dd if of命令，我大概想到的应该是 买个硬盘盒插 ubuntu机器上来做吧，那，还得去买个硬盘盒。。。。
> 
> dd if of非得在ubuntu下执行？你有U盘做PE，为啥不能用U盘先做openwrt。openwrt下用dd if of固件刷到MSATA。之后就一劳永逸了。用PE盘，每次还得接显示器，键盘，鼠标，这才是真的麻烦。

其实吧，与其这么复杂，不如U盘和MSATA都不要，直接把openwrt刷进一张TF卡，插拔也简单。。。

HUiTHEME, 2021-03-08:
> 虽然报错但是其实还是能正常启动的，当初写盘报Error #0 occured while writting to disk at sector ******，但是硬盘插上能进系统，用了两个月一点毛病都没有

所以只能视而不见这个报错了哈。。。

HUiTHEME, 2021-03-08:
> 刷入以前的版本，然后新版本升级上去看怎么样。

也行，但之前的版本，我没备份。。。

wifiway520, 2021-03-08:
你可以试试刷进tf卡能不能启动

leungyauming, 2021-03-08:
> > 虽然报错但是其实还是能正常启动的，当初写盘报Error #0 occured while writting to disk at sector ******，但是硬盘插上能进系统，用了两个月一点毛病都没有
> 
> 所以只能视而不见这个报错了哈。。。

你先試一下，能用就好。

klever1988, 2021-03-08:
> > dd if of 怎么刷固件？
> > 硬盘是MSATA，得插软路由上，软路由先装个ubuntu系统来跑dd if of命令？那没有U盘跑 PE运行 physdiskwrite 方便啊。。。
> > dd if of命令，我大概想到的应该是 买个硬盘盒插 ubuntu机器上来做吧，那，还得去买个硬盘盒。。。。
> 
> dd if of非得在ubuntu下执行？你有U盘做PE，为啥不能用U盘先做openwrt。openwrt下用dd if of固件刷到MSATA。之后就一劳永逸了。用PE盘，每次还得接显示器，键盘，鼠标，这才是真的麻烦。

http://www.chrysocome.net/dd
http://gnuwin32.sourceforge.net/packages/gzip.htm

shunlongyang, 2021-03-24:
同样遇到了这个问题，不知道怎么解决

ziyangdaoren, 2021-04-07:
用dd命令把5.4内核版本的固件直接刷进硬盘会grub会识别不到，先把用4.19内核版本的刷到硬盘里，再用sysupgrade命令升级5.4内核版本的固件，就能正常识别，很奇怪。

closed 2021-06-17 by coolsnowwolf

shunzawow, 2021-12-20:
我进入PE用phydiskwrite也遇到几次这样的情况，但是我用rufus写同一个软路由固件却没问题。无意中发现用DiskGenius删除分区后（务必不要新建分区及格式化）保存即可，再用phydiskwrite就没错误提示了。

newsharpstar, 2022-05-03:
我刚刚根据上面的帖子解决了用disk image写入efi.img文件报错的问题。在make menuconfig里面选择镜像包的时候需要在默认的选项基础上再选择gzip那个选项，然后执行编译。在输出目录中找到生成的efi.img.gz的文件，在linux里面使用gzip -d 文件名，将文件解压，最后将解压出来的efi.img文件传到u盘里面，用disk image文件写入硬盘即可。

Astarry-1127, 2023-03-26:
如果写入进硬盘时错误，那就尝试删除分区，重新新建分区时选择ext4，不要选择ntfs

goutC, 2023-11-07:
> 我进入PE格式化磁盘后用phydiskwrite也遇到几次这样的情况，但是我用rufus写同一个软路由固件却没问题。无意中发现用DiskGenius删除所有分区后（务必不要新建分区及格式化）保存即可，再用phydiskwrite就没错误提示了。

不要新建分区及格式化后phydiskwrite确实可以正常写入了。（补仓：我用的是微PE，rufus用3.8或4.3版本都识别不到主板上的固态磁盘）

dhso, 2024-04-23:
> 我进入PE格式化磁盘后用phydiskwrite也遇到几次这样的情况，但是我用rufus写同一个软路由固件却没问题。无意中发现用DiskGenius删除所有分区后（务必不要新建分区及格式化）保存即可，再用phydiskwrite就没错误提示了。

你说的对！

liubili, 2024-09-20:
亲测 DiskGenius删除所有分区后再保存分区表确实能行