coolsnowwolf/lede #6486: Github Actions编译失败

wang1zhen, 2021-03-07:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

利用Github Actions在线编译失败

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：

本地编译成功，同样的`.config`文件利用github actions编译无法通过

[image]

### (2) 路由器型号和固件版本

A：x86-64

### (3) 详细日志

A：

`.config`文件：

```
CONFIG_TARGET_x86=y
CONFIG_TARGET_x86_64=y
CONFIG_TARGET_x86_64_DEVICE_generic=y
CONFIG_PACKAGE_6in4=y
# CONFIG_PACKAGE_UnblockNeteaseMusic is not set
# CONFIG_PACKAGE_UnblockNeteaseMusicGo is not set
# CONFIG_PACKAGE_autosamba is not set
CONFIG_PACKAGE_ip6tables=y
# CONFIG_PACKAGE_iptables-mod-ipsec is not set
CONFIG_PACKAGE_ipv6helper=y
# CONFIG_PACKAGE_kmod-crypto-cbc is not set
# CONFIG_PACKAGE_kmod-crypto-deflate is not set
# CONFIG_PACKAGE_kmod-crypto-des is not set
# CONFIG_PACKAGE_kmod-crypto-echainiv is not set
# CONFIG_PACKAGE_kmod-crypto-hmac is not set
# CONFIG_PACKAGE_kmod-crypto-md5 is not set
# CONFIG_PACKAGE_kmod-ipsec is not set
# CONFIG_PACKAGE_kmod-ipt-ipsec is not set
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-iptunnel=y
# CONFIG_PACKAGE_kmod-iptunnel6 is not set
CONFIG_PACKAGE_kmod-nf-nat6=y
CONFIG_PACKAGE_kmod-sit=y
# CONFIG_PACKAGE_libatomic is not set
CONFIG_PACKAGE_libcap=y
# CONFIG_PACKAGE_libcares is not set
# CONFIG_PACKAGE_libgmp is not set
# CONFIG_PACKAGE_libhttp-parser is not set
# CONFIG_PACKAGE_libnghttp2 is not set
CONFIG_PACKAGE_libwebsockets-full=y
# CONFIG_PACKAGE_luci-app-accesscontrol is not set
# CONFIG_PACKAGE_luci-app-ipsec-vpnd is not set
# CONFIG_PACKAGE_luci-app-jd-dailybonus is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_fuse-utils is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-ng is not set
# CONFIG_PACKAGE_luci-app-rclone_INCLUDE_rclone-webui is not set
# CONFIG_PACKAGE_luci-app-samba is not set
# CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server is not set
CONFIG_PACKAGE_luci-app-ttyd=y
# CONFIG_PACKAGE_luci-app-unblockmusic is not set
# CONFIG_PACKAGE_luci-app-uugamebooster is not set
CONFIG_PACKAGE_luci-app-watchcat=y
# CONFIG_PACKAGE_luci-app-xlnetacc is not set
CONFIG_PACKAGE_luci-i18n-ttyd-zh-cn=y
CONFIG_PACKAGE_luci-i18n-watchcat-zh-cn=y
CONFIG_PACKAGE_luci-proto-ipv6=y
CONFIG_PACKAGE_luci-theme-argon=y
CONFIG_PACKAGE_luci-theme-freifunk-generic=y
CONFIG_PACKAGE_luci-theme-material=y
CONFIG_PACKAGE_luci-theme-netgear=y
# CONFIG_PACKAGE_node is not set
CONFIG_PACKAGE_odhcp6c=y
CONFIG_PACKAGE_odhcp6c_ext_cer_id=0
CONFIG_PACKAGE_odhcpd-ipv6only=y
CONFIG_PACKAGE_odhcpd_ipv6only_ext_cer_id=0
# CONFIG_PACKAGE_samba36-server is not set
# CONFIG_PACKAGE_shadowsocksr-libev-server is not set
# CONFIG_PACKAGE_strongswan is not set
CONFIG_PACKAGE_ttyd=y
# CONFIG_PACKAGE_uugamebooster is not set
CONFIG_PACKAGE_vim-full=y
CONFIG_PACKAGE_watchcat=y
# CONFIG_PACKAGE_wsdd2 is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=10240
# CONFIG_UnblockNeteaseMusic_Go is not set
# CONFIG_UnblockNeteaseMusic_NodeJS is not set
```

EnnawYang, 2021-03-07:
日志？

wang1zhen, 2021-03-07:
```
2021-03-07T10:03:50.7891723Z  make[3] -C package/lean/luci-app-filetransfer compile
2021-03-07T10:03:59.7142936Z  make[2] package/install
2021-03-07T10:04:17.6781740Z  make[2] target/install
2021-03-07T10:04:19.8360792Z  make[3] -C target/linux install
2021-03-07T10:59:42.2589864Z ##[error]The operation was canceled.
2021-03-07T10:59:42.6598169Z Post job cleanup.
2021-03-07T10:59:44.1095021Z [command]/usr/bin/git version
2021-03-07T10:59:44.1120091Z git version 2.30.1
2021-03-07T10:59:44.1158060Z [command]/usr/bin/git config --local --name-only --get-regexp core\.sshCommand
2021-03-07T10:59:44.1159376Z [command]/usr/bin/git submodule foreach --recursive git config --local --name-only --get-regexp 'core\.sshCommand' && git config --local --unset-all 'core.sshCommand' || :
2021-03-07T10:59:44.1165167Z [command]/usr/bin/git config --local --name-only --get-regexp http\.https\:\/\/github\.com\/\.extraheader
2021-03-07T10:59:44.1166237Z http.https://github.com/.extraheader
2021-03-07T10:59:44.1169946Z [command]/usr/bin/git config --local --unset-all http.https://github.com/.extraheader
```

EnnawYang, 2021-03-07:
私有库？ 免费时间用完了？

wang1zhen, 2021-03-07:
应该不是的

wang1zhen, 2021-03-07:
actions显示的日志有问题，导致error的日志应该在这一段：
```
MODPOST vmlinux.o
  MODINFO modules.builtin.modinfo
  LD      .tmp_vmlinux.kallsyms1
  KSYM    .tmp_vmlinux.kallsyms1.o
  LD      .tmp_vmlinux.kallsyms2
  KSYM    .tmp_vmlinux.kallsyms2.o
  LD      vmlinux
  SORTEX  vmlinux
  SYSMAP  System.map
grep: write error: No space left on device
Makefile:1100: recipe for target 'vmlinux' failed
make[5]: *** [vmlinux] Error 2
make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.101'
make[4]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.101/.image] Error 2
make[3]: *** [install] Error 2
make[2]: *** [target/linux/install] Error 2
Makefile:30: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.101/.image' failed
make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
Makefile:13: recipe for target 'install' failed
make[3]: Leaving directory '/workdir/openwrt/target/linux'
time: target/linux/install#19.58#6.98#33.18
target/Makefile:23: recipe for target 'target/linux/install' failed
make[2]: Leaving directory '/workdir/openwrt'
target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install' failed
make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.
```

leungyauming, 2021-03-07:
> actions显示的日志有问题，导致error的日志应该在这一段：
> 
> ```
> MODPOST vmlinux.o
>   MODINFO modules.builtin.modinfo
>   LD      .tmp_vmlinux.kallsyms1
>   KSYM    .tmp_vmlinux.kallsyms1.o
>   LD      .tmp_vmlinux.kallsyms2
>   KSYM    .tmp_vmlinux.kallsyms2.o
>   LD      vmlinux
>   SORTEX  vmlinux
>   SYSMAP  System.map
> grep: write error: No space left on device
> Makefile:1100: recipe for target 'vmlinux' failed
> make[5]: *** [vmlinux] Error 2
> make[5]: Leaving directory '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.101'
> make[4]: *** [/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.101/.image] Error 2
> make[3]: *** [install] Error 2
> make[2]: *** [target/linux/install] Error 2
> Makefile:30: recipe for target '/workdir/openwrt/build_dir/target-x86_64_musl/linux-x86_64/linux-5.4.101/.image' failed
> make[4]: Leaving directory '/workdir/openwrt/target/linux/x86'
> Makefile:13: recipe for target 'install' failed
> make[3]: Leaving directory '/workdir/openwrt/target/linux'
> time: target/linux/install#19.58#6.98#33.18
> target/Makefile:23: recipe for target 'target/linux/install' failed
> make[2]: Leaving directory '/workdir/openwrt'
> target/Makefile:19: recipe for target '/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install' failed
> make[1]: *** [/workdir/openwrt/staging_dir/target-x86_64_musl/stamp/.target_install] Error 2
> make[1]: Leaving directory '/workdir/openwrt'
> make: *** [world] Error 2
> /workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
> Error: Process completed with exit code 2.
> ```
> 
> 此外actions页面有提示：
> 
> > You are running out of disk space. The runner will stop working when the machine runs out of disk space. Free space left: 99 MB
> 
> 我的`.config`文件中设置了rootfs partition为10240MB，是否与这个有关？

對 `grep: write error: No space left on device` 指的就是設備儲存空間不足。
應該跟你在 `.config` 文件中的 `rootfs partition` 設置無關。

[image]

GitHub Actions的機器只有 14GB SSD空間 可能你使用的空間超了。

leungyauming, 2021-03-07:
```
<details>
<summary>標題</summary>

a
b
c

</details>
```

下次可以將很長的log這樣收納，GitHub Issue沒法直接跳到最低，你log放太長每次都要滑到最下面很麻煩。

wang1zhen, 2021-03-07:
> 
> 
> ```
> <details>
> <summary>標題</summary>
> 
> a
> b
> c
> 
> </details>
> ```
> 
> 下次可以將很長的log/config這樣收納，GitHub Issue沒法直接跳到最低，你log/config放太長每次都要滑到最下面很麻煩。

感谢，已改。

leungyauming, 2021-03-08:
你試一下在workflow的最後加一個step，跑df指令看一下磁盤使用率。

wang1zhen, 2021-03-09:
从头试了一下，发现又成功了，但还是有两个疑问想请教一下：

- 我分别用了两位L大的源码，配置的项目几乎相同，除了两边不同的跳墙软件，编译出来的固件下载下来，`openwrt-x86-64-combined-squashfs.img`文件，Lienol的源码编译出来的大小为55.8M，Lean的源码编译出来的大小为554M，不过经过zip压缩后大小又都差不多了。两边我都设置了512M的root file system partition，是不是这个引起的？

- 编译后在`bin/target/x86/64`目录下的`config.buildinfo`能否直接作为下一次按照相同配置编译的`.config`文件使用？`lede`文件夹下的`.config`文件有6000多行，而`config.buildinfo`不到100行，换言之，在使用github actions编译的时候，能不能上传`config.buildinfo`来作为配置文件进行编译呢？

WYC-2020, 2021-03-09:
没用0填充吧，自己用16进制工具查看就知道了

Dimitri2020007, 2021-03-10:
造成失败的可能的原因：

1、分区空间设置太大了（我kernel和rootfs分区大小各2G）

2、编译的组件有冲突。

3、编译的组件已经更新了版本而源码未能及时收录更新版本的组件。

4、action脚本更新。

wang1zhen, 2021-03-10:
> 
> 
> 造成失败的可能的原因：
> 
> 1、分区空间设置太大了（我kernel和rootfs分区大小各2G）
> 
> 2、编译的组件有冲突。
> 
> 3、编译的组件已经更新了版本而源码未能及时收录更新版本的组件。
> 
> 4、action脚本更新。
> 
> 从你的日志可以看出来你的rootfs分区大小给的太大了，kernel分区给的太小了。kernel分区大小太小了容易无限重启，没有足够的空间解压内核。

谢谢，我着手调整一下。

closed 2021-03-27 by wang1zhen