coolsnowwolf/lede #6488: R4S  1G 最新版拨号无用，即PPPOE拨号无效

xiaodi8688, 2021-03-07:
R4S  1G 最新版PPPOE拨号无效

leungyauming, 2021-03-07:
給多一點資訊
還有沒有人有同樣的情況？

xiaodi8688, 2021-03-07:
友善官方的rk3399-sd-friendlywrt-5.10-20210305.img.zip这个固件没问题，换了五六个人的Lean源码固件，测试都不行，但官方的没问题

coolsnowwolf, 2021-03-08:
没日志

xiaodi8688, 2021-03-08:
我修改下配置文件，重新编译了，晚上再刷试一试，到时候一起吧日志搞上来！

xiaodi8688, 2021-03-08:
系统日志，我复制上来，大佬看看，是不是配置文件还是哪里冲突了！
Fri Jan 18 16:50:56 2013 daemon.info netdata[5832]: resources control: allowed file descriptors: soft = 1024, max = 4096
Fri Jan 18 16:50:56 2013 daemon.info netdata[5832]: Adjusted my Out-Of-Memory (OOM) score from 0 to 1000.
Fri Jan 18 16:50:56 2013 daemon.err netdata[5832]: Cannot adjust netdata scheduling policy to idle (5), with priority 0. Falling back to nice.
Fri Jan 18 16:50:56 2013 daemon.err netdata[5832]: Cannot get my current process scheduling policy.
Fri Jan 18 16:50:56 2013 daemon.info netdata[5832]: netdata started on pid 5832.
Fri Jan 18 16:50:56 2013 daemon.info netdata[5832]: Initializing spawn client.
Fri Jan 18 16:50:56 2013 daemon.info netdata[5832]: Executing /usr/lib/netdata/plugins.d/system-info.sh
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]: ttyd 1.6.3 (libwebsockets 3.2.0)
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]: tty configuration:
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]:   start command: /usr/libexec/login.sh
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]:   close signal: SIGHUP (1)
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]:   terminal type: xterm-256color
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]:  Using foreign event loop...
Fri Jan 18 16:50:56 2013 daemon.notice ttyd[6059]:  Listening on port: 7681

xiaodi8688, 2021-03-08:
需要单独添加RTL8211E（wan口）RTL8211H（lan口）的网卡驱动不？你的源码有单独这两个网卡的驱动没？有的话能否将RTL8211网卡驱动的配置文件名称发下？谢谢！！！

minlang112, 2021-03-08:
> 系统日志，我复制上来，大佬看看，是不是配置文件还是哪里冲突了！
> Fri Jan 18 16:50:56 2013 daemon.info netdata[5832]: resources control: allowed file descriptors: soft = 1024, max = 4096
> Fri Jan 18 16:50:56 2013 daemon.info [...]

麻烦日志里搜索一下，有没有pppoe相关的内容

xiaodi8688, 2021-03-08:
> 麻烦日志里搜索一下，有没有pppoe相关的内容

搜索了，木有哈，我怀疑是RTL8211网卡驱动的问题！

minlang112, 2021-03-08:
额，如果网卡没有驱动起来的话，你连WAN都无法设置的呀，更不用谈PPPOE拨号了

xiaodi8688, 2021-03-08:
> 额，如果网卡没有驱动起来的话，你连WAN都无法设置的呀，更不用谈PPPOE拨号了

友善官方的rk3399-sd-friendlywrt-5.10-20210305.img.zip这个固件可以拨号，除了这个，其他的都不行，不知道是什么问题，好郁闷！

xiaodi8688, 2021-03-08:
> 麻烦日志里搜索一下，有没有pppoe相关的内容

Fri Jan 18 16:51:15 2013 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Fri Jan 18 16:51:15 2013 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Fri Jan 18 16:51:15 2013 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Fri Jan 18 16:51:35 2013 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Fri Jan 18 16:51:35 2013 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Fri Jan 18 16:51:35 2013 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Fri Jan 18 16:51:55 2013 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Fri Jan 18 16:51:55 2013 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Fri Jan 18 16:51:55 2013 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Fri Jan 18 16:52:12 2013 daemon.notice netifd: wan (3997): udhcpc: received SIGTERM
Fri Jan 18 16:52:12 2013 daemon.notice netifd: wan (3997): udhcpc: entering released state
Fri Jan 18 16:52:12 2013 daemon.notice netifd: wan (3997): Command failed: Permission denied
Fri Jan 18 16:52:12 2013 daemon.notice netifd: Interface 'wan' is now down
Fri Jan 18 16:52:12 2013 daemon.notice netifd: Interface 'wan' has link connectivity loss
Fri Jan 18 16:52:12 2013 daemon.notice netifd: Interface 'wan' has link connectivity
Fri Jan 18 16:52:12 2013 daemon.notice netifd: Interface 'wan' is setting up now
Fri Jan 18 16:52:12 2013 daemon.err insmod: module is already loaded - slhc
Fri Jan 18 16:52:12 2013 daemon.err insmod: module is already loaded - ppp_generic
Fri Jan 18 16:52:12 2013 daemon.err insmod: module is already loaded - pppox
Fri Jan 18 16:52:12 2013 daemon.err insmod: module is already loaded - pppoe
Fri Jan 18 16:52:12 2013 daemon.notice netifd: wan (6205): uci: Entry not found
Fri Jan 18 16:52:12 2013 daemon.notice netifd: wan (6205): sh: out of range
Fri Jan 18 16:52:12 2013 daemon.info pppd[6256]: Plugin rp-pppoe.so loaded.
Fri Jan 18 16:52:12 2013 daemon.info pppd[6256]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Jan 18 16:52:12 2013 daemon.notice pppd[6256]: pppd 2.4.8 started by root, uid 0
Fri Jan 18 16:52:15 2013 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Fri Jan 18 16:52:15 2013 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Fri Jan 18 16:52:15 2013 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Fri Jan 18 16:52:27 2013 daemon.notice netifd: Interface 'wan' is now down
Fri Jan 18 16:52:27 2013 daemon.notice netifd: Interface 'wan' is setting up now
Fri Jan 18 16:52:27 2013 daemon.err insmod: module is already loaded - slhc
Fri Jan 18 16:52:27 2013 daemon.err insmod: module is already loaded - ppp_generic
Fri Jan 18 16:52:27 2013 daemon.err insmod: module is already loaded - pppox
Fri Jan 18 16:52:27 2013 daemon.err insmod: module is already loaded - pppoe
Fri Jan 18 16:52:27 2013 daemon.notice netifd: wan (6941): uci: Entry not found
Fri Jan 18 16:52:27 2013 daemon.notice netifd: wan (6941): sh: out of range
Fri Jan 18 16:52:27 2013 daemon.info pppd[6974]: Plugin rp-pppoe.so loaded.
Fri Jan 18 16:52:27 2013 daemon.info pppd[6974]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Jan 18 16:52:27 2013 daemon.notice pppd[6974]: pppd 2.4.8 started by root, uid 0
Fri Jan 18 16:52:35 2013 daemon.notice netifd: Interface 'wan' is now down
Fri Jan 18 16:52:35 2013 daemon.notice netifd: Interface 'wan' is setting up now
Fri Jan 18 16:52:35 2013 daemon.err insmod: module is already loaded - slhc
Fri Jan 18 16:52:35 2013 daemon.err insmod: module is already loaded - ppp_generic
Fri Jan 18 16:52:35 2013 daemon.err insmod: module is already loaded - pppox
Fri Jan 18 16:52:35 2013 daemon.err insmod: module is already loaded - pppoe
Fri Jan 18 16:52:35 2013 daemon.notice netifd: wan (7164): uci: Entry not found
Fri Jan 18 16:52:35 2013 daemon.notice netifd: wan (7164): sh: out of range
Fri Jan 18 16:52:35 2013 daemon.info pppd[7196]: Plugin rp-pppoe.so loaded.
Fri Jan 18 16:52:35 2013 daemon.info pppd[7196]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Jan 18 16:52:35 2013 daemon.notice pppd[7196]: pppd 2.4.8 started by root, uid 0
Fri Jan 18 16:52:35 2013 user.info : Does not have a configuration file inside `/etc/netdata/ebpf.conf. It will try to load stock file.
Fri Jan 18 16:52:35 2013 user.err : PROCFILE: Cannot open file '/etc/netdata/apps_groups.conf'
Fri Jan 18 16:52:35 2013 user.info : Cannot read process groups configuration file '/etc/netdata/apps_groups.conf'. Will try '/usr/lib/netdata/conf.d/apps_groups.conf'
Fri Jan 18 16:52:50 2013 daemon.warn pppd[7196]: Timeout waiting for PADO packets
Fri Jan 18 16:52:50 2013 daemon.err pppd[7196]: Unable to complete PPPoE Discovery
Fri Jan 18 16:52:50 2013 daemon.info pppd[7196]: Exit.
Fri Jan 18 16:52:50 2013 daemon.notice netifd: Interface 'wan' is now down
[... cut ...]
Fri Jan 18 16:53:05 2013 daemon.err insmod: module is already loaded - pppoe
Fri Jan 18 16:53:05 2013 daemon.notice netifd: wan (7559): uci: Entry not found
Fri Jan 18 16:53:05 2013 daemon.notice netifd: wan (7559): sh: out of range
Fri Jan 18 16:53:05 2013 daemon.info pppd[7593]: Plugin rp-pppoe.so loaded.
Fri Jan 18 16:53:05 2013 daemon.info pppd[7593]: RP-PPPoE plugin version 3.8p compiled against pppd 2.4.8
Fri Jan 18 16:53:05 2013 daemon.notice pppd[7593]: pppd 2.4.8 started by root, uid 0

xiaodi8688, 2021-03-08:
看看，能否看出什么问题，是1G版本不兼容吗？还是有其他问题？

leungyauming, 2021-03-09:
```
Fri Jan 18 16:53:05 2013 daemon.warn pppd[7409]: Timeout waiting for PADO packets
Fri Jan 18 16:53:05 2013 daemon.err pppd[7409]: Unable to complete PPPoE Discovery
Fri Jan 18 16:53:05 2013 daemon.info pppd[7409]: Exit.
```

這個部分就是説你 PPPoE 鏈接建立失敗
你可以試一下在 LuCi 把 `LCP echo failure threshold` 跟 `LCP echo interval` 調高一點

DanielChen008, 2021-03-09:
话说我也是用 L 大的源码编译的固件，把 R4S 1G塞在进门弱电箱用了几个月了，定期更新重新编译一次，还未遇到不能拨号的问题。只是更新固件后，不能自动重启，每次得拔电源重插这点比较麻烦~

xiaodi8688, 2021-03-09:
应该是没添加8211驱动的原因，我尝试吧8211的驱动加上去在尝试，由于最近家里有事要处理，后面我在慢慢找原因测试！

xiaodi8688, 2021-03-25:
> r4s 转接网卡 rtl8111h 驱动已经由 r8169 提供。
在配置文件里添加rtl8111h的驱动名称就可以是吗？