coolsnowwolf/lede #6489: x86/x64 开启 SFE 后有自动 KP/重启 问题调查

coolsnowwolf, 2021-03-08:
目前零星收到一些反馈，主要是 x86/x64 开启 SFE 后，在某些主板上会有 KP/重启的问题

现在需要收集下信息，遇到这种情况的 CPU/主板/网卡 都是些什么？ 用哪些插件为主？

NagaseKouichi, 2021-03-08:
J4205软路由，网卡是I211。似乎是开启SFE之后在负载比较大的时候，会重启。比如我用docker运行了JD薅羊毛脚本，晚上0点的时候跑的脚本比较多，如果软路由开了SFE，在0点左右就有可能会重启

lqhorochi, 2021-03-08:
没有出现过,i350t2虚拟化和i350t4直通,虚拟化都没有问题
还有bcm5719直通也没有问题

binge8, 2021-03-08:
3215U，i211网卡，主板未知，关闭SFE一个月都没问题，开启SFE几乎每天一次无规律重启。主要用KP.SSRPLUS+,JD薅羊毛,微信推送，dockerman等

Neverends4, 2021-03-08:
插个楼，我的MT7621 红米RM2100开了SFE/flowoffload+闭源驱动也会随机重启，而且不一定是在负载大的时候……关了就最长26天没重启……前后自己编译了多个版本。然后kernel crushlog好像没用，是空的。正在尝试是不是闭源/开源wifi驱动的问题。

NagaseKouichi, 2021-03-08:
同楼上，小娱路由器C1开了flowoffload也会随机重启

hxlls, 2021-03-08:
小马V1 开启SFE 无重启无报警，开启flowoffload 会日志报错

dhso, 2021-03-08:
主板：J1900
插件：Docker，SS
内存：几乎空闲，无繁重任务
重启：每天不定时重启，有时重启失败无法使用，需要再次手动重启。

mjyhj, 2021-03-08:
信步26026
N2600
开了SFE，大概一星期重启一次，关了后十几天一直稳定

lbsystem, 2021-03-09:
give you a tip it's only about the X86/X64 architecture,and I have the same problem on the J4125cpu

trueboost, 2021-03-09:
CPU是奔腾4405U 网卡是82574L     SSP  PASSWALL  JD  插件为主

AwesomeSky, 2021-03-09:
[image]
[image]
J1900 intel 82583V 关闭SFE已稳定运行一个月 开启后平均每天重启一次

github-xihe, 2021-03-09:
CPU：i3-5010U
插件：docker、ssrp 
开启SFE后，一天会不定时几小时重启一次，关闭后运行几天未出现重启现象。后面取消编译SFE改用flowoffload，开启flowoffload后未出现自动重启现象。

llivenan, 2021-03-09:
设备:M93P  主板：Q87  CPU：I5-4590T  板载网卡英特尔219LM和MINIPCIE转英特尔I210网卡
OPENWRT版本R20.9.13-X64，R20.12.31-X64，R21.1.8-X64，三个版本都试过
ESXI虚拟OPENWRT，开启SFE，基本10个小时左右会死机，重启后恢复，故障重复，取消SFE或用FLOWOFFLOAD后测试十来天，网络正常，无死机，无重启

WYC-2020, 2021-03-09:
看来sfe还真是害了不少人哈哈

asushugo, 2021-03-09:
这个问题挺久的了，目前我遇到的是信步N2600，还有D525的，都可以复现这个问题，一般来说用一两天就会莫名其妙重启，只要关闭SFE或者切换到Flow就不会再出现了，插件就只运行了网易云解锁和SSRP，网卡是intel的，具体型号忘记了

db-one, 2021-03-09:
J1900  微软虚拟机环境  之前开启sfe一切正常，后来入了京东云，20M上行跑满上传，负载一高，就开始不定时系统重启，基本都在十来个小时左右，最多不超过一天，我装了外部日志服务器抓日志，也没看到有什么异常，后来是关闭了serverchan解决的，但不清楚和sfe有没有关系，不过症状倒是跟大家差不多，不过sfe好像是对京东云跑流量有点影响，也干脆一起关了，现在一切正常

z1790, 2021-03-09:
j1900 211网口 docker ssr 开sfe随机重启。。。有时候是几个小时。。有时候是几天。。关掉sfe就没重启过，不过sfe确实是降低不少cpu的使用率

xe5700, 2021-03-09:
歌华链 MT7621 SFE FLOWOFFLOAD都会重启，不过SFE更离谱，疯狂重启，有时候刚开机就重启，似乎最多没撑过12小时。
FLOWOFFLOAD+HWNAT倒是重启频率不是特别高，但偶尔也会发生。

jameyhhh, 2021-03-09:
8600T /Z390ITX /I211 219，物理机 群里大雕的X86 5.4内核固件。如openwrt-x64-R21.1.8.-generic-squashfs-combined.img 。所遇问题如下：
1、经常不定时重启。
2、SSR重启后是关闭状态，不会自动运行，得每次手动开启。
3、设置没变，GFW模式。但SSR导致部分国内网站打不开 (没记录具体网站)，关闭SSR正常。
4、设置没变，DDNS-阿里云- 重启后 IP长时间不变动。外网无法访问。
没法只能换回去年的4.19固件：openwrt-R20.4.8-x64-combined-squashfs(此版本之前稳定运行快1年没出任何问题)。暂时不敢再刷新固件了。望大佬解决。

sxx1314, 2021-03-09:
[#5349](https://github.com/coolsnowwolf/lede/issues/5349#issuecomment-708977359)
sfe导致重启最早起源于4.19最后几个版本然后一直延续5.4，不怎么需要这玩意就一直拖着没review，只有压测结果。目前情况看和硬件关系也不太大。

BTvalley, 2021-03-09:
老大 我的也重启 但不知道什么原因 来看了一下 原来是sfe的问题 flowoffload怎么切换也不会 大佬们  flowoffload 怎么切换呀

BTvalley, 2021-03-09:
我的是x86 虚拟机

cevinwu87, 2021-03-09:
J1900，网卡RTL8111，群里面最新的R21.1.8会莫名重启，之前的版本都没有遇到自己重启的情况

db-one, 2021-03-09:
> J1900 微软虚拟机环境 之前开启sfe一切正常，后来入了京东云，20M上行跑满上传，负载一高，就开始不定时系统重启，基本都在十来个小时左右，最多不超过一天，我装了外部日志服务器抓日志，也没看到有什么异常，后来是关闭了serverchan解决的，但不清楚和sfe有没有关系，不过症状倒是跟大家差不多，不过sfe好像是对京东云跑流量有点影响，也干脆一起关了，现在一切正常

补一个当时导出来的日志记录
2020-08-22T15:28:19+00:00 OpenWrt crond[16666]: crond (busybox 1.31.1) started, log level 8
2020-08-22T15:28:20+00:00 OpenWrt dnsmasq[16599]: Connected to system UBus
2020-08-22T20:00:00+00:00 OpenWrt crond[16666]: USER root pid 29039 cmd /usr/bin/serverchan/serverchan send &
2020-08-23T02:00:00+00:00 OpenWrt crond[16666]: USER root pid 25646 cmd lua /usr/share/passwall/rule_update.lua nil log > /dev/null 2>&1 &
2020-08-23T05:28:53+00:00 OpenWrt kernel: [162679.994484] ------------[ cut here ]------------
2020-08-23T05:28:53+00:00 OpenWrt kernel: [162679.998113] Kernel BUG at kfree+0x16a/0x1c0 [verbose debug info unavailable]
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.072437] invalid opcode: 0000 [#1] SMP PTI
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.106647] CPU: 1 PID: 22575 Comm: iptables Not tainted 5.4.51 #0
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.142131] Hardware name: Microsoft Corporation Virtual Machine/Virtual Machine, BIOS 090007  05/18/2018
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.190104] RIP: 0010:kfree+0x16a/0x1c0
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.222098] Code: ff ff c3 49 8b 04 24 31 db a9 00 00 01 00 74 06 41 0f b6 5c 24 51 49 8b 04 24 a9 00 00 01 00 75 0b 49 8b 44 24 08 a8 01 75 02 <0f> 0b 89 d9 ba ff ff ff ff be 06 00 00 00 48 c7 c7 80 42 28 82 d3
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.310087] RSP: 0018:ffffc9000973fcd0 EFLAGS: 00010246
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.350079] RAX: ffff88801faa2708 RBX: 0000000000000000 RCX: 0000000000000010
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.390076] RDX: 0000000000000000 RSI: 00000000002a0000 RDI: ffff88800a800000
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.423422] RBP: ffffc9000973fce8 R08: 0000000000000001 R09: 00000000f0000000
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.474085] R10: 0000000000000004 R11: 0000000000000000 R12: ffff88801faa0000
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.518082] R13: ffffffff82289d78 R14: ffff88800a800000 R15: ffffffff82279580
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.562101] FS:  00007efe12631d48(0000) GS:ffff88801f440000(0000) knlGS:0000000000000000
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.606080] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.646101] CR2: 0000000000c74000 CR3: 0000000014c7e000 CR4: 00000000001006e0
2020-08-23T05:28:55+00:00 OpenWrt kernel: [162680.694150] Call Trace:

22号8点的时候serverchan通知了一次信息，23号两点时候PW更新了一次规则，后面就没有任何日志了，然后到5.28分直接重启了，也可能重启时候日志没来得及输出到rsyslog，固件是LEAN+PW

[image]

icefruit-j, 2021-03-10:
开了DNS加速会导致科学谷歌打不开的问题

zhaoxp520, 2021-03-10:
i5 3337U
Intel 82574L网卡
只要不编译docker，怎么开都不会重启，编译了docker进去，无论容器多不多负载大不大，不关sfe就会不定时重启。
常开插件，openclash，微信推送，网易云解锁，kms……

PurpleMoon369, 2021-03-11:
Intel I7 5775c，华擎H97主板，I350 t4加rtl8125b，运行在proxmox ve虚拟机上，开启sfe后出现过一次直接卡死无法进入后台，无法ssh连接，vnc无法控制，虚拟机后台强制关闭无效，只能强行关闭物理机，关闭sfe后无类似情况

BOBINIUNIU, 2021-03-12:
个人觉得J1900以上级别的软路由就别开SFE了，转发能力已经足够了。SFE，BBR有时候会和其他插件冲突，成为一种负优化。
P.S
有人测试过开SFE和不开SFE包转发性能的对比吗？

Dimitri2020007, 2021-03-12:
> 个人觉得J1900以上级别的软路由就别开SFE了，转发能力已经足够了。SFE，BBR有时候会和其他插件冲突，成为一种负优化。
> P.S
> 有人测试过开SFE和不开SFE包转发性能的对比吗？

这个我会注意下的

lfdfxl, 2021-03-12:
小米R3G，MT7621的CPU开启HWNAT也有不定时重启现象，大佬也关注一下。

badcrazy, 2021-03-12:
hp t520 ,amd gx-212jc，主板型号不知道，rtl8151gh（板载）+rtl8111f（minipcie转接）双网卡，插件只开启了京东签到、kms和openclash，bt下载时两次遇到重启，当时cpu占用未知，基本肯定在50%以上，连接数在10%以上

guestid, 2021-03-14:
小娱，开HWNAT不定时会重启

eallion, 2021-03-14:
我直接没有编译 sfe

iso-lib, 2021-03-16:
PVE+华擎J3455 应该是rtl8111网卡吧
op只编译了passwall+frpc+sfe,还有mpd-full作为音乐服务器
很稳.
其实没特殊要求那个eSir的精品小包还不错
只不过网上的固件都没编译kmod-sound-core/kmod-usb-audio,我需要接一个usb声卡输出声音,只能自己编译.

MrEsquire, 2021-03-16:
[image]
esxi7.0 i5-10600 i350t4v2网卡直通 ,8125网卡直通
sfe开启会死机
cpu占用直接拉满,我esxi都进不去了

qpzr, 2021-03-17:
竞斗云
SFE+BBR  电脑斗鱼全屏+手机音频投屏，手机WIFI征战着大哥大的LOL排位赛，剑圣疯狗模式制霸全图的时候，重启断网了。 不知道这算不算？
没开BBR的时候，也是电脑斗鱼全屏手机WIFI征战大哥大完全没问题。
开了BBR不到一天，才玩了两三把。就开始出现卡延迟高，然后下一把就突然红了，我以为是宽带问题马上切出去开加速器，结果加速到80不动了，电脑斗鱼黑屏，右下角网线图标消失，地球仪图标出现，才意识到坏了竞斗云崩了重启0.0  流量打完对局后看系统日志，结果第一行Thu Mar 18 00:00:11 2021 kern.info kernel: [    0.000000] Booting Linux on physical CPU 0x0  很明显日志没了，因为电脑上微信朋友问我掉线了？是3点10分

iso-lib, 2021-03-18:
> 竞斗云 ipq4019
> SFE+BBR 电脑斗鱼全屏+手机音频投屏，手机WIFI征战着大哥大的LOL排位赛，剑圣疯狗模式制霸全图的时候，重启断网了。 不知道这算不算？
> 没开BBR的时候，也是电脑斗鱼全屏手机WIFI征战大哥大完全没问题。
> 开了BBR不到一天，才玩了两三把。就开始出现卡延迟高，然后下一把就突然红了电脑上也听不到队友说话，我以为是宽带问题马上切出去开加速器，结果加速到80不动了，电脑斗鱼黑屏，右下角网线图标消失，地球仪图标出现，才意识到坏了竞斗云崩了重启0.0 流量打完对局后看系统日志，结果第一行Thu Mar 18 00:00:11 2021 kern.info kernel: [ 0.000000] Booting Linux on physical CPU 0x0 很明显日志没了，因为电脑上微信朋友问我掉线了？是3点10分
---------------------------------------------------
竞斗云不是x86吧..........................

HiGarfield, 2021-03-19:
以前发现过一个问题，SFE开启的情况下，SQM QoS再开启，如果使用cake，则会随机导致负载跑满，然后看门狗超时而重启。关了SFE就一切正常。

shu0802, 2021-03-20:
3865u\i211网卡，开启sfe会偶尔断网和重启
[image]
[image]

qiuzhongqiang, 2021-03-21:
主板：3215U工控软路由。
插件：Docker，xray，dlna，aria2，网易云。（其他默认设置编译的功能，但只开启括号前的）
内存：内存剩余79%
其他：cpu温度40左右。活动连接500左右，大部分都是空闲任务，偶尔aria2限速7M下BT，平时大都看网页视频，手机应用等。
重启：每天不定时重启。

dolphinpaopao, 2021-03-21:
我有意测试观察过，对比出问题的配置，我没有编译docker、nwan。lan关闭桥接，不论是pve还是物理机安装，目前开启SFE都很稳定。距离上次编译固件已稳定运行30天。所以可能是组件冲突导致内核崩溃重启或者硬件有问题，比如内存。
[image]
[image]
[image]

shu0802, 2021-03-21:
> 我有意测试观察过，对比出问题的配置，我没有编译docker、nwan。lan关闭桥接，不论是pve还是物理机安装，目前开启SFE都很稳定。距离上次编译固件已稳定运行30天。所以可能是组件冲突导致内核崩溃重启或者硬件有问题，比如内存。
> [image]
> [image]
> [image]

> 我有意测试观察过，对比出问题的配置，我没有编译docker、nwan。lan关闭桥接，不论是pve还是物理机安装，目前开启SFE都很稳定。距离上次编译固件已稳定运行30天。所以可能是组件冲突导致内核崩溃重启或者硬件有问题，比如内存。
> [image]
> [image]
> [image]

> 我有意测试观察过，对比出问题的配置，我没有编译docker、nwan。lan关闭桥接，不论是pve还是物理机安装，目前开启SFE都很稳定。距离上次编译固件已稳定运行30天。所以可能是组件冲突导致内核崩溃重启或者硬件有问题，比如内存。
> [image]
> [image]
> [image]

内核版本可以看出来你的固件编译已经有一些时间了

dolphinpaopao, 2021-03-21:
> 内核版本可以看出来你的固件编译已经有一些时间了

大兄弟，就算是再新的固件，编译完成再观察两月也变成旧版本固件了，一个月前编译还只是调整增加了ipv6，没有其他变动。所以确切说来这个版本已经稳定运行三个月。以前编译的固件也没有遇到开启SFE导致每天重启问题，我都是默认关闭桥接开启SFE的。

shu0802, 2021-03-21:
> 大兄弟，就算是再新的固件，编译完成再观察两月也变成旧版本固件了，一个月前编译还只是调整增加了ipv6，没有其他变动。所以确切说来这个版本已经稳定运行三个月。以前编译的固件也没有遇到开启SFE导致每天重启问题，我都是默认关闭桥接开启SFE的。

之前的内核版本也正常，就是到101、105后才出现这种情况。不知与开启桥接模式是否有关。

dolphinpaopao, 2021-03-21:
> 之前的内核版本也正常，就是到101、105后才出现这种情况。不知与开启桥接模式是否有关。

老早前就有其他人上报出现过SFE重启问题，在我的网络500/50和使用环境（bt下载、远程、游戏和视频）下没有复现，所以可能是使用了不太成熟的组件比如docker，nwan冲突或者开启了有冲突的功能，比如桥接加速，上面明确提示到和vpn/桥接模式可能有冲突。openwrt在x86平台作为路由使用个人感觉设置好后是非常稳定的了，其他平台没用过，印象中驱动问题比较多。

Neverends4, 2021-03-21:
红米2100。之前wifi闭源驱动重启，试了编译开源驱动+sfe，只开启了sfe和ipv6加速，没用bbr，LS说的桥接模式加速也没开，还是随机重启了，开机时间4天多。
版本：OpenWrt R21.2.1 / LuCI Master (git-20.343.54716-6fc079f)
插件：
vlmcsd KMS服务器
upnp
dns=https-proxy
nlbwmon带宽监控
wireguard（应该和这个没关系，之前没加入编译也随机重启）
没有docker，但是有vlan看iptv，不过之前某次重启是午夜，iptv肯定没有用，看起来关系也不大？

stock169, 2021-03-22:
竞斗云一样开了SFE就各种死机重启关了就好，只能放弃

yjd, 2021-03-22:
`【2020.1.14op19.07-K2p最终版】openwrt-ramips-mt7621-k2p-squashfs-sysupgrade.bin`
我前几天从你网盘下这个刷了。然后前天下载正义联盟导演版，由于人多速度超快28M/s
然后就重启了一次。
而之前一直用`openwrt-ramips-R9.8.17-mt7621-k2p-5.0.0.2-squashfs-sysupgrade.bin`用了1,2年很正常。
当然速度没有新版快。新版打开网页明显响应快不少。

shjohnnyyu, 2021-03-26:
D2550 以前开SFE，各种高负荷运用一直稳定使用不重启，编译加了docker，就各种死机重启，关了SFE才太平。

He-Peter-Huang, 2021-03-27:
有一次重启的时候在ESXI上看到了这个
[image]

zhangj3, 2021-03-31:
我有十多台X86裸刷的OP，固件刷好后第一件事就是把sfe之类的关掉，今年年初吃了端口转发不成功的亏，后来还是KFER Mercer帮我定位到是sfe打开会造成端口转发有问题....sfe对x86提升到底有多大？

shu0802, 2021-03-31:
用了dockerman后就出现不停的死机或者是重启，不知是不是与dockerman有关系？

yyetss, 2021-04-03:
J900+i211，主板未知，跑了jd羊毛和小飞机，会出现不定时重启，在win0自带那个虚拟安装同样的固件跑同样的脚本和插件一样无规律重启

Dimitri2020007, 2021-04-12:
i5-10400f
主板b460m-d3h
系统windows server2019 
双软路由运行在hyper-v虚拟机（第二代）（两个软路由都是lede openwrt）
主路由网关192.168.1.2、ip192.168.1.1
旁路由网关192.168.1.1、ip192.168.1.2
两个网关的dns都是isp的。
（这样会产生环路吗？）

常用插件：ppp多拨、smartdns、kms激活、ssr+、turboacc加速有问题没有编译、upnp

掉线大约2天断一次，以下是掉线时的日志。

Mon Apr 12 14:46:55 2021 daemon.info procd: Instance sysntpd::instance1 s in a crash loop 28 crashes, 0 seconds since last crash
Mon Apr 12 14:46:56 2021 user.info mwan3track[64573]: Check (ping) success for target "www.baidu.com" on interface wan (pppoe-wan)
Mon Apr 12 14:46:56 2021 user.info mwan3track[67471]: Check (ping) success for target "www.baidu.com" on interface vwan2 (pppoe-vwan2)
Mon Apr 12 14:46:56 2021 user.info mwan3track[67471]: Lost 66 ping(s) on interface vwan2 (pppoe-vwan2)
Mon Apr 12 14:46:56 2021 user.notice mwan3track[67471]: Interface vwan2 (pppoe-vwan2) is online
Mon Apr 12 14:46:57 2021 user.info mwan3track[66070]: Check (ping) success for target "www.baidu.com" on interface vwan1 (pppoe-vwan1)
Mon Apr 12 14:46:57 2021 user.info mwan3track[66070]: Lost 66 ping(s) on interface vwan1 (pppoe-vwan1)
Mon Apr 12 14:46:57 2021 user.notice mwan3track[66070]: Interface vwan1 (pppoe-vwan1) is online
Mon Apr 12 14:46:57 2021 user.info mwan3track[62244]: Check (ping) success for target "www.baidu.com" on interface vwan3 (pppoe-vwan3)
Mon Apr 12 14:46:57 2021 user.info mwan3track[62244]: Lost 66 ping(s) on interface vwan3 (pppoe-vwan3)
Mon Apr 12 14:46:57 2021 user.notice mwan3track[62244]: Interface vwan3 (pppoe-vwan3) is online
Mon Apr 12 14:47:01 2021 daemon.err htpdate[4857]: www.yahoo.com no timestamp
Mon Apr 12 14:47:06 2021 daemon.info pppd[107758]: No response to 5 echo-requests
Mon Apr 12 14:47:06 2021 daemon.notice pppd[107758]: Serial link appears to be disconnected.
Mon Apr 12 14:47:06 2021 daemon.info pppd[107758]: Connect time 0.2 minutes.
Mon Apr 12 14:47:06 2021 daemon.info pppd[107758]: Sent 168 bytes, received 168 bytes.
Mon Apr 12 14:47:06 2021 daemon.notice netifd: Network device 'pppoe-vwan3' link is down
Mon Apr 12 14:47:06 2021 daemon.info pppd[107877]: No response to 5 echo-requests
Mon Apr 12 14:47:06 2021 daemon.notice pppd[107877]: Serial link appears to be disconnected.
Mon Apr 12 14:47:06 2021 daemon.info pppd[107877]: Connect time 0.3 minutes.
Mon Apr 12 14:47:06 2021 daemon.info pppd[107877]: Sent 168 bytes, received 168 bytes.
Mon Apr 12 14:47:06 2021 daemon.notice netifd: Network device 'pppoe-vwan1' link is down
Mon Apr 12 14:47:12 2021 daemon.notice netifd: vwan1 (109947): 2 processes have arrived, waiting for the left 2
Mon Apr 12 14:47:12 2021 daemon.info procd: Instance sysntpd::instance1 s in a crash loop 29 crashes, 0 seconds since last crash
Mon Apr 12 14:47:30 2021 daemon.err pppd[110515]: syncppp: sem_timewait time out
Mon Apr 12 14:47:30 2021 daemon.err pppd[110515]: syncppp sync fail
Mon Apr 12 14:47:31 2021 daemon.info procd: Instance sysntpd::instance1 s in a crash loop 30 crashes, 0 seconds since last crash

是pppd断线之后才出现procd崩溃的。

galaxyskyknight, 2021-04-13:
高负载高流量就有kernel warining,不断地打，10来秒一次。机器CPU I3865U, 网卡Intel I211 6口，这个问题之前报过。没人重视。现在日志里全是这个玩意，烦都烦死了了。观察到的现象就是网口软中断奇高，网卡有大量的overrun和一些drop. 启用irq balance也没怎么解决问题。
网卡的吞吐量应该是没有到达理论值，上下行加起来才200多Mb/s

Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.675936] ------------[ cut here ]------------
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.680788] WARNING: CPU: 1 PID: 0 at 0xffffffffa18ee22a
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.686271] Modules linked in: xt_FULLCONENAT ath9k iwl4965 iwl3945 ath9k_htc     ath9k_common xt_connlimit wl18xx wl12xx rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtl8821ae    rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800pci rt2800mmio rt2800lib rt2500usb      rt2500pci rt2400pci qcserial pppoe ppp_async option nf_conncount mt76x0u mt76x0e mt76x0_common l2tp_ppp iwlmvm iwlegacy iwldvm  ipw2200 ipw2100 iptable_nat gspca_m5602 gspca_konica gspca_jeilinj gspca_gl860 gspca_finepix gspca_etoms gspca_conex cdc_mbim   carl9170 btcoexist brcmsmac brcmfmac b43legacy b43 ath9k_hw ath6kl_usb ath6kl_core ath5k ath10k_pci ath10k_core ath adm8211     zd1211rw xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wlcore  uvcvideo usb8xxx usb_wwan ti_usb_3410_5052 sr9700 snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek           snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codec
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.686296]  smsc95xx sierra_net rtw88_pci rtw88_core rtlwifi rtl8xxxu rtl8821cu  rtl8812au rtl818x_pci rtl8187 rt73usb rt61pci rt2x00usb rt2x00pci rt2x00mmio rt2x00lib rsi_usb rsi_91x rndis_host qmi_wwan pwc  pptp pppox ppp_mppe ppp_generic plusb p54usb p54pci p54common orinoco_pci orinoco nft_redir nft_ct nf_nat_tftp                  nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table  nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink                nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mwl8k mwifiex_pcie  mwifiex mt7915e mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615e mt7615_common      mt7603e mt7601u mt76_usb mt76 mcs7830 mac80211 libipw libertas kalmia iwlwifi ipt_REJECT huawei_cdc_ncm gspca_zc3xx             gspca_vc032x gspca_tv8532 gspca_t613 gspca_sunplus gspca_stv06xx gspca_stk014
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.776049]  gspca_sq905c gspca_sq905 gspca_spca561 gspca_spca508 gspca_spca506   gspca_spca505 gspca_spca501 gspca_spca500 gspca_sonixj gspca_sonixb gspca_sn9c20x gspca_se401 gspca_pac7311 gspca_pac207        gspca_ov534_9 gspca_ov534 gspca_ov519 gspca_mr97310a gspca_mars gspca_main dm9601 cp210x ch341 cfg80211 cdc_subset cdc_ncm      cdc_ether cdc_eem ax88179_178a ath9k_pci_owl_loader asix amdgpu xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent    xt_quota xt_policy xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_esp xt_ecn xt_dscp xt_comment    xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY xr_usb_serial_common wlcore_sdio wireguard videobuf2_v4l2      videobuf2_common via_velocity via_rhine usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot           ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm solos_pci snd_usbmidi_lib snd_intel_nhlt    snd_hda_core slhc sky2 skge sis900 sis190 sch_cake rtl8150
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.865876]  regmap_core radeon r8712u(C) r8169 r8168 r8152 r8125 r6040 pegasus   pcnet32 niu nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota nft_objref nft_numgen        nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4              nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4    ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core macvlan lzo lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211    kvm_intel kvm kaweth irqbypass iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables igc i40e hso gpu_sched forcedeth exfat ethoc et131x e1000e e100 dmx_usb crc8 crc7 crc_itu_t crc_ccitt cpia2 cordic compat cdc_wdm cdc_acm brcmutil     br_netfilter bnx2x bnx2 be2net atl2 atl1e atl1c atl1 asn1_decoder alx 8390 8139too 8139cp fuse tcp_bbr sch_tbf sch_ingress      sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.955164]  cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred        configs videobuf2_vmalloc videobuf2_memops videodev evdev i2c_dev spi_ks8995 siit cryptodev xt_set ip_set_list_set              ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac              ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port              ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink rtl8306 mvsw61xx ip17xx nf_log_ipv6 nf_log_common ip6table_mangle         ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig nfsv4 nfsv3 nfsd nfs bonding 3c59x ip_gre gre          [... cut ...]
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.546404]  call_cpuidle+0x1e/0x40
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.550061]  do_idle+0xf4/0x1c0
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.553335]  cpu_startup_entry+0x18/0x20
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.557403]  start_secondary+0x145/0x170
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.561491]  secondary_startup_64+0xa4/0xb0
Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.565843] ---[ end trace 1c88e1cb8f9f7cc2 ]---

Dimitri2020007, 2021-04-13:
> #6114 #6089 这两个Issue. @coolsnowwolf 好久了，自从前几个月某个版本以后，就一直有这个问题
> 高负载高流量就有kernel warining,不断地打，10来秒一次。机器CPU I3865U, 网卡Intel I211 6口，这个问题之前报过。没人重视。现在日志里全是这个玩意，烦都烦死了了。观察到的现象就是网口软中断奇高，网卡有大量的overrun和一些drop. 启用irq balance也没怎么解决问题。
> 网卡的吞吐量应该是没有到达理论值，上下行加起来才200多Mb/s
> 
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.675936] ------------[ cut here ]------------
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.680788] WARNING: CPU: 1 PID: 0 at 0xffffffffa18ee22a
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.686271] Modules linked in: xt_FULLCONENAT ath9k iwl4965 iwl3945 ath9k_htc ath9k_common xt_connlimit wl18xx wl12xx rtw88_8822ce rtw88_8822c rtw88_8822be rtw88_8822b rtw88_8723de rtw88_8723d rtl8821ae rtl8192se rtl8192de rtl8192cu rtl8192ce rtl8192c_common rtl_usb rtl_pci rt2800usb rt2800pci rt2800mmio rt2800lib rt2500usb rt2500pci rt2400pci qcserial pppoe ppp_async option nf_conncount mt76x0u mt76x0e mt76x0_common l2tp_ppp iwlmvm iwlegacy iwldvm ipw2200 ipw2100 iptable_nat gspca_m5602 gspca_konica gspca_jeilinj gspca_gl860 gspca_finepix gspca_etoms gspca_conex cdc_mbim carl9170 btcoexist brcmsmac brcmfmac b43legacy b43 ath9k_hw ath6kl_usb ath6kl_core ath5k ath10k_pci ath10k_core ath adm8211 zd1211rw xt_state xt_nat xt_helper xt_conntrack xt_connmark xt_connbytes xt_REDIRECT xt_MASQUERADE xt_FLOWOFFLOAD xt_CT wlcore uvcvideo usb8xxx usb_wwan ti_usb_3410_5052 sr9700 snd_usb_audio snd_hda_intel snd_hda_codec_via snd_hda_codec_realtek snd_hda_codec_hdmi snd_hda_codec_generic snd_hda_codec
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.686296] smsc95xx sierra_net rtw88_pci rtw88_core rtlwifi rtl8xxxu rtl8821cu rtl8812au rtl818x_pci rtl8187 rt73usb rt61pci rt2x00usb rt2x00pci rt2x00mmio rt2x00lib rsi_usb rsi_91x rndis_host qmi_wwan pwc pptp pppox ppp_mppe ppp_generic plusb p54usb p54pci p54common orinoco_pci orinoco nft_redir nft_ct nf_nat_tftp nf_nat_snmp_basic nf_nat_sip nf_nat_pptp nf_nat_irc nf_nat_h323 nf_nat_ftp nf_nat_amanda nf_nat nf_flow_table_hw nf_flow_table nf_conntrack_tftp nf_conntrack_snmp nf_conntrack_sip nf_conntrack_rtcache nf_conntrack_pptp nf_conntrack_netlink nf_conntrack_irc nf_conntrack_h323 nf_conntrack_ftp nf_conntrack_broadcast nf_conntrack_amanda nf_conntrack mwl8k mwifiex_pcie mwifiex mt7915e mt76x2u mt76x2e mt76x2_common mt76x02_usb mt76x02_lib mt7663u mt7663_usb_sdio_common mt7615e mt7615_common mt7603e mt7601u mt76_usb mt76 mcs7830 mac80211 libipw libertas kalmia iwlwifi ipt_REJECT huawei_cdc_ncm gspca_zc3xx gspca_vc032x gspca_tv8532 gspca_t613 gspca_sunplus gspca_stv06xx gspca_stk014
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.776049] gspca_sq905c gspca_sq905 gspca_spca561 gspca_spca508 gspca_spca506 gspca_spca505 gspca_spca501 gspca_spca500 gspca_sonixj gspca_sonixb gspca_sn9c20x gspca_se401 gspca_pac7311 gspca_pac207 gspca_ov534_9 gspca_ov534 gspca_ov519 gspca_mr97310a gspca_mars gspca_main dm9601 cp210x ch341 cfg80211 cdc_subset cdc_ncm cdc_ether cdc_eem ax88179_178a ath9k_pci_owl_loader asix amdgpu xt_time xt_tcpudp xt_tcpmss xt_statistic xt_socket xt_recent xt_quota xt_policy xt_pkttype xt_owner xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_esp xt_ecn xt_dscp xt_comment xt_addrtype xt_TPROXY xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY xr_usb_serial_common wlcore_sdio wireguard videobuf2_v4l2 videobuf2_common via_velocity via_rhine usbserial usbnet usblp ums_usbat ums_sddr55 ums_sddr09 ums_karma ums_jumpshot ums_isd200 ums_freecom ums_datafab ums_cypress ums_alauda tulip ts_kmp ts_fsm ts_bm solos_pci snd_usbmidi_lib snd_intel_nhlt snd_hda_core slhc sky2 skge sis900 sis190 sch_cake rtl8150
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.865876] regmap_core radeon r8712u(C) r8169 r8168 r8152 r8125 r6040 pegasus pcnet32 niu nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject_bridge nft_reject nft_quota nft_objref nft_numgen nft_meta_bridge nft_log nft_limit nft_hash nft_fwd_netdev nft_dup_netdev nft_counter nf_tproxy_ipv6 nf_tproxy_ipv4 nf_tables_set nf_tables nf_socket_ipv6 nf_socket_ipv4 nf_reject_ipv4 nf_log_ipv4 nf_dup_netdev nf_defrag_ipv6 nf_defrag_ipv4 ne2k_pci mmc_spi mlx5_core mlx4_en mlx4_core macvlan lzo lib80211_crypt_wep lib80211_crypt_tkip lib80211_crypt_ccmp lib80211 kvm_intel kvm kaweth irqbypass iptable_raw iptable_mangle iptable_filter ipt_ah ipt_ECN ipheth ip_tables igc i40e hso gpu_sched forcedeth exfat ethoc et131x e1000e e100 dmx_usb crc8 crc7 crc_itu_t crc_ccitt cpia2 cordic compat cdc_wdm cdc_acm brcmutil br_netfilter bnx2x bnx2 be2net atl2 atl1e atl1c atl1 asn1_decoder alx 8390 8139too 8139cp fuse tcp_bbr sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32 cls_tcindex cls_route
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21339.955164] cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred configs videobuf2_vmalloc videobuf2_memops videodev evdev i2c_dev spi_ks8995 siit cryptodev xt_set ip_set_list_set ip_set_hash_netportnet ip_set_hash_netport ip_set_hash_netnet ip_set_hash_netiface ip_set_hash_net ip_set_hash_mac ip_set_hash_ipportnet ip_set_hash_ipportip ip_set_hash_ipport ip_set_hash_ipmark ip_set_hash_ip ip_set_bitmap_port ip_set_bitmap_ipmac ip_set_bitmap_ip ip_set nfnetlink rtl8306 mvsw61xx ip17xx nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables ip6t_REJECT x_tables nf_reject_ipv6 swconfig nfsv4 nfsv3 nfsd nfs bonding 3c59x ip_gre gre snd_via82xx snd_intel8x0 snd_mpu401_uart ixgbevf ixgbe igbvf igb e1000 snd_ac97_codec ac97_bus ifb dummy mdio l2tp_netlink l2tp_core udp_tun [... cut ...]
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.546404] call_cpuidle+0x1e/0x40
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.550061] do_idle+0xf4/0x1c0
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.553335] cpu_startup_entry+0x18/0x20
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.557403] start_secondary+0x145/0x170
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.561491] secondary_startup_64+0xa4/0xb0
> Sat Apr 10 17:44:20 2021 kern.warn kernel: [21340.565843] ---[ end trace 1c88e1cb8f9f7cc2 ]---

用的ESXI ?

WYC-2020, 2021-04-13:
@galaxyskyknight  那就不要更新，更新有bug修复都是随缘，有人看就不错了，sfe是第三方驱动，调试肯定要很久，不相等 那就不要启动sfe或者回归旧版本，不要催哈哈

galaxyskyknight, 2021-04-13:
> 用的ESXI ?

不是，软路由，物理机直刷

galaxyskyknight, 2021-04-13:
> @galaxyskyknight 那就不要更新，更新有bug修复都是随缘，有人看就不错了，sfe是第三方驱动，调试肯定要很久，不相等 那就不要启动sfe或者回归旧版本，不要催哈哈

我没有催，我只说fact和主观感受.我通篇没有提出说“必须”“马上”,"修复“之类的请求。

Dimitri2020007, 2021-04-13:
> 不是，软路由，物理机直刷

只用了sfe还是sfe和flowoffload一起编译的？

我没有试过直刷，一直都是虚拟机跑的

[6 more replies]

closed 2021-06-17 by coolsnowwolf