coolsnowwolf/lede #6492: micro sd卡启动不了

wifiway520, 2021-03-08:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述
最近编译openwrt写入micro sd卡抱错，偶尔能成功，但不论成功与否都是无法启动，有时还没有启动项，会出现error unknow  system..ahy  key  continue，，然后就跳到openwrt和openwrt(忘了)选项上，我的是uefi启动，采用legacy启动也是一样，添加了sdhcui和mmc驱动，也在make kernelconfig里添加mmc支持无果，，最后尝试koolshare的专用写盘固件顺利启动并进去系统，这期间，更换过tf卡，也用读卡器写盘成功并运行，也就是相同固件USB启动正常，而micro sd卡就是不成功，对比koolshare_lede的启动文件有四个，而openwrt有三个，不知道是不是这个原因！另外，就算用USBU盘启动openwrt后，插入micro  sd卡也不能识别出来，，最后没注意把硬盘都写没了，，，是驱动还是什么问题
### (1) 具体问题

A：microsd写盘不启动

### (2) 路由器型号和固件版本

A：x86

### (3) 详细日志

A：
无

wifiway520, 2021-03-08:
纠正一下，koolshare的lede是四个分区，而lean大的openwrt是三个分区

leungyauming, 2021-03-09:
有可能你的機器會挑卡，有沒有試過換一張 micro SD 卡？

wifiway520, 2021-03-09:
> 有可能你的機器會挑卡，有沒有試過換一張 micro SD 卡？

换过而且koolshare的lede可以开机进系统

EnnawYang, 2021-03-09:
可以试下TF卡完全擦除(用0填充)，然后再刷试试能不能启动

wifiway520, 2021-03-09:
基础的都做过了，而且相同的方法用koolshare的lede就可以启动，包括在koolshare里用dd命令

wifiway520, 2021-03-09:
再说明下，我有三个卡，三个卡我都可以安装识别Ubuntu和隔壁的lede，只要换了lean大的openwrt就是启动不起来

sxx1314, 2021-03-09:
自行修改 grub 使用uuid引导即可

closed 2021-06-17 by coolsnowwolf