coolsnowwolf/lede #6498: L大，建议源码中默认可以用Nginx替换掉uhttp

uosX, 2021-03-09:
1.L大，建议源码中默认可以用Nginx替换掉uhttp

uosX, 2021-03-09:
并且默认开启https，默认支持宝塔证书

WYC-2020, 2021-03-09:
最好不那么搞，很多人根本就不会玩nginx，如果用了nginx 那luci界面相关设置 web的都会无效，必须去nginx配置文件修改，要被小白喷死

minlang112, 2021-03-09:
为什么要改？

EnnawYang, 2021-03-09:
你可以自己改（大声

sxx1314, 2021-03-09:
n年前 默认就是Nginx  然后改回来了  为什么要改回来自行脑补。。。

qianxu2001, 2021-03-09:
局域网为什么要用 https

jun129, 2021-03-10:
去掉一些依赖APP 还有默认设置 可以不编译uhttpd，只编译nginx
不能去掉uhttpd就 / 看看哪个依赖，去掉即可

jun129, 2021-03-10:
但是nginx的ssr+有bug，更改配置会无法启动，必须SSH service shadowsocksr relaod 才行

Myoko, 2021-03-10:
不要NGINX最好

sxx1314, 2021-03-10:
> uhttpd轻量，够用，nginx对于MT7621等MIPS路由不友好，资源消耗太大。然后同意LS的，要改你自己改不就是了？

其实7621带nginx是足矣，仅仅管理等界面的日常cpu消耗接近于0-1%。但是flash还有没有空间塞者玩意就不一定了。

ihuaer, 2021-03-11:
大部分人用不上的功能

jun129, 2021-03-11:
我只是说一下我用的场景。。。

cyijun, 2021-03-12:
自己设计pcb做7621路由器的路过，要对路由器的功能有正确认识。。如果要玩花的大可以另开一个服务器，lede本来就是linux嵌入式开发环境，嵌入式用ngnix就比较少。。

uosX, 2021-04-28:
https://www.freebuf.com/articles/network/269696.html
我觉得有必要提醒大家，安全领域已经在研究uhttp的相关漏洞了

closed 2021-06-17 by coolsnowwolf