coolsnowwolf/lede #6501: 求助！如何选择wifi驱动？小米3g使用MTK驱动表现不佳，性能只有使用开源驱动时的3/2左右。

JohyC, 2021-03-09:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
求助，编译固件时如何选择WiFi驱动。
编译固件多次了，MTK驱动在米3g上表现实在不能令人满意。
500mbps带宽，wifi速度最快只能达到 300mbps。
下载文件或软件的速度只有 35mb/s。

### (2) 路由器型号和固件版本

A：
型号：Xiaomi Mi Router 3G
固件版本：OpenWrt R21.2.1 / LuCI Master (git-21.035.34289-13f96d1)
内核版本：5.4.101

### (3) 详细日志

A：wifi速度的日志怎么看啊 ？

Neverends4, 2021-03-10:
你需要SFE/Flowoffload开启硬件NAT，否则7621跑300M就是极限了。开了以后我的红米2100有线能跑到700M。然而SFE/Flowoffload貌似有导致不规则重启的毛病，见#6489

jxph1123, 2021-03-10:
我也是3G，只编译了Flowoffload，没有编译SFE，目前没有出现不规则重启。wifi用5G跑500应该没问题，跑700是不行的，毕竟是开源，但5G很稳定，不错了。

JohyC, 2021-03-10:
> 你需要SFE/Flowoffload开启硬件NAT，否则7621跑300M就是极限了。开了以后我的红米2100有线能跑到700M。然而SFE/Flowoffload貌似有导致不规则重启的毛病，见#6489

Turbo ACC的HWNAT开启了。 之前使用社区编译的老版本开源wifi固件，我的wifi速度能够达到 400mbps；换到MTK闭源驱动后速度一直在200到300之间。大多时间下载东西只有25mb/s左右。

jxph1123, 2021-03-10:
现在默认编译是闭源，你可以自己改成开源

JohyC, 2021-03-10:
> 现在默认编译是闭源，你可以自己改成开源

对！我现在的诉求就是想知道如何编译开源wifi！ 我不知道应该编译那些驱动。
之前自己尝试编译过，但编译出来的固件没有wifi。。。

jxph1123, 2021-03-10:
在/target/linux/ramips/image/mt7621.mk 内，
将  DEVICE_PACKAGES := kmod-mt7603e kmod-mt76x2e kmod-usb3 \
	kmod-usb-ledtrig-usbport luci-app-mtwifi uboot-envtools
改为
  DEVICE_PACKAGES := kmod-mt7603 kmod-mt76x2 kmod-usb3 \
	kmod-usb-ledtrig-usbport wpad-openssl uboot-envtools
就是开源了

jxph1123, 2021-03-10:
需要手动将luci-app-mtwifi去掉

JohyC, 2021-03-10:
> 在/target/linux/ramips/image/mt7621.mk 内， R3G里面对应的
> 将 DEVICE_PACKAGES := kmod-mt7603e kmod-mt76x2e kmod-usb3
> kmod-usb-ledtrig-usbport luci-app-mtwifi uboot-envtools
> 改为
> DEVICE_PACKAGES := kmod-mt7603 kmod-mt76x2 kmod-usb3
> kmod-usb-ledtrig-usbport wpad-openssl uboot-envtools
> 就是开源了

多谢！下次编译的时候我再试试。

cc-monkey, 2021-03-21:
你好，编译前修改以上的代码，去掉luci-app-mtwifi勾选，就可以编译了吗？还需不需要勾选其他选项。