coolsnowwolf/lede #6514: 如何调整编译固件的大小

cc-monkey, 2021-03-11:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述
现在编译小米3g固件，插件只是默认的插件，另外加了一个，为什么编译出来的固件有21.4M，之前编译才10.4的大小。
### (1) 具体问题

A：

### (2) 路由器型号和固件版本
小米3G
A：

### (3) 详细日志

A：

ihuaer, 2021-03-11:
清空配置后重新编译

jeasonliang, 2021-03-11:
试试 rm -rf  .config tmp
还有是否使用了upx 压缩？

cc-monkey, 2021-03-11:
清空配置，还是一样的，如何使用upx压缩呢，之前用虚拟机编译出来的固件都只有10兆左右，我是用物理机的Ubuntu编译的。

badcrazy, 2021-03-12:
这要看你加了什么插件，了哟徐诶插件本身体积就大，还要顺带安装一堆依赖

cc-monkey, 2021-03-13:
谢谢，是ssrplus这个插件依赖问题。还有给问题就是usb打印服务器的问题，luci-app-p910nd和luci-app-usb-print这两个都编译进固件了，好像p910nd没什么用，我用的东芝e-studio2006打印机，打印机在节能模式时打印文件无法唤醒打印机，之前用老毛子的固件不存在这个问题，这是什么原因呢？求解！

jeasonliang, 2021-03-15:
> 谢谢，是ssrplus这个插件依赖问题。还有给问题就是usb打印服务器的问题，luci-app-p910nd和luci-app-usb-print这两个都编译进固件了，好像p910nd没什么用，我用的东芝e-studio2006打印机，打印机在节能模式时打印文件无法唤醒打印机，之前用老毛子的固件不存在这个问题，这是什么原因呢？求解！

要不我按照你的要求帮你编译一个试试？ 反正我的机器闲着， 呵呵！

cc-monkey, 2021-03-18:
谢谢你，openwrt我编译了无数给版本还是不能自动唤醒打印机，老毛子的就没有这个问题

pjygithub, 2021-03-19:
> 谢谢你，openwrt我编译了无数给版本还是不能自动唤醒打印机，老毛子的就没有这个问题

打印可以试试 cups ，通过分享添加打印机到计算机上

closed 2021-06-17 by coolsnowwolf