coolsnowwolf/lede #6521: R21是否能自定义root filesystem partition size大小？

jobber2955, 2021-03-12:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ x] 没有类似的issue
* 已在Issue搜索 overlay和root filesystem partition size

## 2. 详细叙述
之前用过其他人的Openwrt，是支持自定义Root filesystem partition size，这样一来可以方便地充分利用TF卡的容量（例如R2S），但Lean大佬的R21貌似不支持自定义，无法在编译时更改。请问是否能支持自定义呢？若没有支持的打算，那是否我们能通过某些方法更改Root filesystem partition size的大小呢？
### (1) 具体问题
SDK中的Target Image --> Root filesystem partition size 无法自定义
A：

### (2) 路由器型号和固件版本
路由器：R2S
固件版本：最新版R21
A：

### (3) 详细日志
无
A：

Ljzkirito, 2021-03-14:
[image]
[image]
[image]
你再看看？

jobber2955, 2021-03-14:
> [image]
> [image]
> [image]
> 你再看看？

啊这，那估计是我的ssh客户端按键转发有问题了，我再试试，不好意思打扰了

jobber2955, 2021-03-14:
已解决：是垃圾网页版终端导致的按键转发问题。
测试：WIN10自带SSH、PuTTY、XShell三款都无该问题。

closed 2021-03-14 by jobber2955