coolsnowwolf/lede #6534: r4s lan wan 口无法启动

wyap-echo, 2021-03-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：用今天最新代码编译的（默认配置）刷入内存卡 启动后lan wan 口灯均不亮 换内存卡测试后情况依旧 刷入老版本固件可以启动系统 功能均正常

### (2) 路由器型号和固件版本

A：

### (3) 详细日志

A：编译配置为默认

liuyanantj, 2021-03-15:
我也遇到了相同的情况，现在貌似不分1g和4g版本了

MapesxGM, 2021-03-15:
默认Wan口和Lan调换了。

adamgonglj, 2021-03-15:
是啊，为啥要这样设置呢

adamgonglj, 2021-03-15:
是啊，为啥要这样设置呢

adamgonglj, 2021-03-15:
为啥要这样设置?

adamgonglj, 2021-03-15:
> 默认Wan口和Lan调换了。

为啥要这样设置呢？关键是更新日志中也没有讲呀。被坑死。

liuyanantj, 2021-03-15:
试了下 果然调换接口了  但是现在的两个网卡的指示灯都不亮了

adamgonglj, 2021-03-15:
可以自己配置led灯

adamgonglj, 2021-03-15:
@coolsnowwolf 
Lean，您好，如果有时间，在下个版本把wan lan口改回来呗，另外wan-lan灯可以配置一下

MapesxGM, 2021-03-15:
> > 默认Wan口和Lan调换了。
> 
> 为啥要这样设置呢？关键是更新日志中也没有讲呀。被坑死。

我编译固件刷机后以为是死机了，看了下更新源码才知道网口调换了，eth0是CPU出来的网口，eth1是通过pcie转的网口。LED灯他们已经提交了修复了，如果自己编译可以改一下源码就可以了。

adamgonglj, 2021-03-15:
> > > 默认Wan口和Lan调换了。
> > 
> > 
> > 为啥要这样设置呢？关键是更新日志中也没有讲呀。被坑死。
> 
> 我编译固件刷机后以为是死机了，看了下更新源码才知道网口调换了，eth0是CPU出来的网口，eth1是通过pcie转的网口。LED灯他们已经提交了修复了，如果自己编译可以改一下源码就可以了。

在源码那个位置改eth0 和 eth1 分别为 wan 和lan 呀，我可以自己去改源码。

MapesxGM, 2021-03-15:
> > > > 默认Wan口和Lan调换了。
> > > 
> > > 
> > > 为啥要这样设置呢？关键是更新日志中也没有讲呀。被坑死。
> > 
> > 
> > 我编译固件刷机后以为是死机了，看了下更新源码才知道网口调换了，eth0是CPU出来的网口，eth1是通过pcie转的网口。LED灯他们已经提交了修复了，如果自己编译可以改一下源码就可以了。
> 
> 在源码那个位置改eth0 和 eth1 分别为 wan 和lan 呀，我可以自己去改源码。

LED灯的问题刚刚源码中已经修复了，网口调换修改lede/target/linux/rockchip/armv8/base-files/etc/board.d/02_network这个文件中的第13行'eth1' 'eth0'调换下位置就可以了，不过我还要测试一下才能确定。

adamgonglj, 2021-03-15:
> LED灯的问题刚刚源码中已经修复了，网口调换修改lede/target/linux/rockchip/armv8/base-files/etc/board.d/02_network这个文件中的第13行'eth1' 'eth0'调换下位置就可以了，不过我还要测试一下才能确定。

等你消息。

我看了一下源码。我感觉02_network的41行也要改。

adamgonglj, 2021-03-15:
我把13和16行改了，我试一下。

adamgonglj, 2021-03-15:
我试了，不行。

MapesxGM, 2021-03-15:
> 我试了，不行。

只改第13行呢

adamgonglj, 2021-03-15:
刚刚试了，也不行。

adamgonglj, 2021-03-15:
> > 我试了，不行。
> 
> 只改第13行呢

刚刚试了，也不行。

这个改了之后不需要安装啥的吧，直接编译就可以的吧？

zgc, 2021-03-15:
-		ucidef_set_interfaces_lan_wan 'eth1' 'eth0'
+		ucidef_set_interfaces_lan_wan 'eth0' 'eth1'

adamgonglj, 2021-03-15:
> * ```
>     ucidef_set_interfaces_lan_wan 'eth1' 'eth0'
>   ```
> 
> * ```
>     ucidef_set_interfaces_lan_wan 'eth0' 'eth1'
>   ```

就是这样改的，不行。eth0和eth1对调了不行的。

adamgonglj, 2021-03-15:
我参考这个试一下。
https://github.com/friendlyarm/friendlywrt/blob/master-v19.07.1/target/linux/rockchip-rk3399/base-files/etc/board.d/02_network

把eth1 改成wlan0.

adamgonglj, 2021-03-15:
还是不行。
难不成要删除临时文件再编译？？
搞不懂了。

azgh, 2021-03-18:
R2S的固件3月7号正常，14号的固件编译后刷完无法访问lan口，我一直以为是固件有问题，看来R2S的网口应该也是对调了

PR #6566 (2021-03-19): rockchip: add execute permission for network setup script
Fixes: f4b1b358fe8b ("rockchip: sync latest codes from ImmortalWrt (#6522)")

Fixes: #6534
Fixes: #6548
Fixes: #6563
closed 2021-03-19 by coolsnowwolf