coolsnowwolf/lede #6536: Newifi-D1 5.4内核的版本没有交换机选项

eagle-lk, 2021-03-14:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：Newifi-D1，初始版本是R20 4.14.180，有两条外线，需要配置两个wan口，通过交换机菜单项下的VLAN功能可以很好的完成配置功能。最近自己编译了R21 5.4.101版本，发现没有交换机配置功能。通过把LAN1退出br-lan桥接接口，然后增加一个wan接口，虽然也能实现类似的功能，但是在硬件加速全开的情况下，转发速率只有先前的一半。

B: 用Newifi-D2顶替D1，使用了最近编译的R21 5.4.101版本，发现有交换机选项，原有配置都能正常实现。转发性能保持了原有的水平。

C: D1通过导入先前的/etc/config/network 文件，可以出现交换机选项，但提示“交换机拓扑未能确定”，vlan端口明显错误。重启后无法连接。

D: 修改编译环境中的 /target/linux/ramips/mt7621/base-files/etc/board.d/02_network，参照d2增加了switch0配置，编译后都无法启动。

E: 重新编译了#tag 200915，发现内核已经升级到了5.4。上一个4.14内核的版本就是tag 200320。

求教：
A：如何能在5.4内核上，让D1的性能恢复到比较正常的水平?
B：如果DSA架构导致的性能下降，是否能让D1也能像D2一样打开交换机选项？具体需要配置/修改哪些文件？
C:  如何能回退到最接近的4.14内核版本上去呢？

折腾了好几天了，就是搞不定，还请大神们不吝赐教，谢谢

### (2) 路由器型号和固件版本

A：
Newifi D1：R20 4.14.180 =》 R21 5.4.101
Newifi D2：OpenWrt 19.07.7 =》 R21 5.4.101

### (3) 详细日志

A：无

yulinsoft, 2021-03-15:
听说是5.0内核交换机采用了与以前不同的驱动，现有的luci无法配置。

eagle-lk, 2021-03-15:
@yulinsoft D2的版本也是5.4内核，但是有交换机选项。我很想探究一下究竟是怎么控制的。如果不配置vlan，性能会大打折扣。

coolsnowwolf, 2021-03-15:
D2 是特意移植了驱动和改写 DTS 实现的

eagle-lk, 2021-03-15:
@coolsnowwolf 谢谢您的亲自答复。恳请您能指教：如何在新架构下，如何配置D1，使其性能达到和4.14内核版本基本一致的水准；或者如何回退到一个恰当的4.14版本上。请您指点一下途径即可，我愿意自己多尝试一下。D2个头太大，没法放到弱电箱里，所以还是希望能继续让D1发挥余热。谢谢！

coolsnowwolf, 2021-03-15:
可以抄 D2 的 DTS 和 network 文件实现，驱动已经移植好

eagle-lk, 2021-03-15:
@coolsnowwolf 好的，我再去仔细核对一下差别。network文件是指/base-files/etc/board.d/02_network 吗？谢谢

AmadeusGhost, 2021-03-15:
https://github.com/AmadeusGhost/lede/commit/990cebb4a7b6e3dd39715655ecc401eed9af95b9 改好了，你可以照着改下试试

eagle-lk, 2021-03-15:
多谢，正在编译中，验证完了向大家汇报。

eagle-lk, 2021-03-15:
@AmadeusGhost 成功启动了，交换机菜单项出来了，VLAN的端口配置也是正常的。稍后再测试一下性能。

eagle-lk, 2021-03-15:
测试完毕，已经恢复到R20.5.20(4.14.180)版本的转发水平了。谢谢Lean老大的指点和AmadeuGhost老兄的相助。目前的是Lean最新的版本，R21.2.1(5.4.105)，Newifi-D1满血复活。

closed 2021-03-15 by eagle-lk

dukmonki, 2021-07-24:
能不能改下Hiwifi 5962的DTS？？？需要。谢谢。

Volmation, 2023-12-03:
@coolsnowwolf 小米4路由器 如何修改