coolsnowwolf/lede #6538: 3月15日最新的jd bonus提示“网络离线状态无法使用二维码获取Cookie，请刷新后重试！”

alepennyq, 2021-03-15:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题
3月15日最新的jd bonus提示“网络离线状态无法使用二维码获取Cookie，请刷新后重试！”，无法添加Cookie

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：如上

### (2) 路由器型号和固件版本

A：OpenWrt X64

### (3) 详细日志

A：

EnnawYang, 2021-03-15:
没有复现

EnnawYang, 2021-03-15:
可以试下以下步骤 

如果更新ipk后无法获取二维码
```
登录ssh后执行
rm -rf /tmp/luci-indexcache /tmp/luci-modulecache/*
```

alepennyq, 2021-03-15:
> 可以试下以下步骤
> 
> 如果更新ipk后无法获取二维码
> 
> ```
> 登录ssh后执行
> rm -rf /tmp/luci-indexcache /tmp/luci-modulecache/*
> ```

不行。。。还是一样的，弹窗“网络离线状态无法使用二维码获取Cookie，请刷新后重试！”

adamgonglj, 2021-03-15:
我也遇到同意的问题

adamgonglj, 2021-03-15:
> > 可以试下以下步骤
> > 如果更新ipk后无法获取二维码
> > ```
> > 登录ssh后执行
> > rm -rf /tmp/luci-indexcache /tmp/luci-modulecache/*
> > ```
> 
> 不行。。。还是一样的，弹窗“网络离线状态无法使用二维码获取Cookie，请刷新后重试！”

执行代码后，仍然报同样的错误。

EnnawYang, 2021-03-15:
建议你们去[上游](https://github.com/jerrykuku/luci-app-jd-dailybonus)看看，有错误日志的话请一起附上

hahach, 2021-03-15:
一样的问题，我补一个系统日志

Mon Mar 15 18:57:01 2021 daemon.notice procd: /etc/rc.d/S25jd-dailybonus: lua: /usr/share/jd-dailybonus/gen_cookieset.lua:18: bad argument #1 to 'pairs' (table expected, got nil)
Mon Mar 15 18:57:01 2021 daemon.notice procd: /etc/rc.d/S25jd-dailybonus: stack traceback:
Mon Mar 15 18:57:01 2021 daemon.notice procd: /etc/rc.d/S25jd-dailybonus: 	[C]: in function 'pairs'
Mon Mar 15 18:57:01 2021 daemon.notice procd: /etc/rc.d/S25jd-dailybonus: 	/usr/share/jd-dailybonus/gen_cookieset.lua:18: in main chunk
Mon Mar 15 18:57:01 2021 daemon.notice procd: /etc/rc.d/S25jd-dailybonus: 	[C]: ?
Mon Mar 15 18:57:02 2021 daemon.notice procd: /etc/rc.d/S25jd-dailybonus: Command failed: Not found

zxcdeyouxiang, 2021-03-15:
同样的问题，这个会不会和系统做旁路由有关

alepennyq, 2021-03-15:
> 同样的问题，这个会不会和系统做旁路由有关

我不是庞路由

sirfromprc, 2021-03-16:
修改 luci-app-jd-dailybonus/luasrc/view/jd-dailybonus/cookie_tools.htm 里，在<script src="https://cdn.jsdelivr.net/npm/qrcode_js@1.0.0/qrcode.min.js"></script>下增加一行jQuery，即下方代码即可使用
<script src="https://apps.bdimg.com/libs/jquery/2.1.4/jquery.min.js"></script>

sirfromprc, 2021-03-16:
在openwrt里的位置是/usr/lib/lua/luci/view/jd-dailybonus/cookie_tools.htm，增加上述一行就好了

alepennyq, 2021-03-16:
> 在openwrt里的位置是/usr/lib/lua/luci/view/jd-dailybonus/cookie_tools.htm，增加上述一行就好了

确实搞定了，3q

closed 2021-03-16 by alepennyq