coolsnowwolf/lede #6548: 3月14日编译R2S固件成功，但刷入设备后，无法正常启动

azgh, 2021-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：3月7日编译固件成功，刷入设备使用正常；3月14日，更改了插件的配置，重新编译，固件编译成功，但刷入设备，系统无法正常启动，重复编译多次，编译固件均成功，刷入设备均无法正常启动系统。
```
cd lede
git pull
./scripts/feeds update -a && ./scripts/feeds install -a
rm -rf ./tmp && rm -rf .config
make menuconfig
make -j8 download V=s
make -j1 V=s
```

### (2) 路由器型号和固件版本

A：nanopi r2s，3月7日编译固件使用正常。

### (3) 详细日志

A：日志未出现异常。

zgc, 2021-03-16:
好像是修复了tf卡启动的问题，导致在线升级有问题，我重新写卡就正常了，一共四台，三台r2s都是一样的状况，一台正常升级。

azgh, 2021-03-16:
> 好像是修复了tf卡启动的问题，导致在线升级有问题，我重新写卡就正常了，一共四台，三台r2s都是一样的状况，一台正常升级。

编译多次，刷tf卡多次，均不能正常启动，后来只能刷回7号编译的固件。
现在这个问题已经修复了吗？

zgc, 2021-03-16:
> > 好像是修复了tf卡启动的问题，导致在线升级有问题，我重新写卡就正常了，一共四台，三台r2s都是一样的状况，一台正常升级。
> 
> 编译多次，刷tf卡多次，均不能正常启动，后来只能刷回7号编译的固件。
> 现在这个问题已经修复了吗？

我编译的commit id: 4824ae9e8

buzz0416, 2021-03-16:
网口反了，r4s也是~

buzz0416, 2021-03-16:
网口反了吧？~今天编译的也还没改回来

azgh, 2021-03-19:
牛

PR #6566 (2021-03-19): rockchip: add execute permission for network setup script
Fixes: f4b1b358fe8b ("rockchip: sync latest codes from ImmortalWrt (#6522)")

Fixes: #6534
Fixes: #6548
Fixes: #6563
closed 2021-03-19 by azgh