coolsnowwolf/lede #6549: 编译的HiWiFi HC5962 SPI 32m固件有线lan口无法连接

lon91ong, 2021-03-16:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：刷入网络下载的16M闪存版编程器固件(内核版本4.14), 有线lan连接正常, 排除硬件问题
自行编译得到了5.4内核的32M闪存版固件, 有线lan连接无法DHCP分配IP, 手动配置静态IP也无法连接; 无线连接DHCP服务正常
怀疑内核版本的问题, 尝试编译4.14内核版, 得到的还是5.4内核

### (2) 路由器型号和固件版本

A：极路由B70 硬改32m spi闪存, 用git action编译的5.4内核版

### (3) 编译配置

A：本来怀疑是内核版本的问题, 想要编译4.14的试试, 但是编译完了发现还是5.4内核
[参考config-4.14](https://github.com/signigelchan/lede/blob/8b8d5c726a98e037086beb9b1424cab6ba060fd1/target/linux/ramips/mt7621/config-4.14)
```
# 内核版本选4.14
sed -i 's/5.4/4.14/g' target/linux/ramips/Makefile
sed -i 's/5.4/4.14/g' target/linux/ramips/image/mt7621.mk
# 32m spi flash
sed -i 's/<0x50000 0xf60000>/<0x50000 0x1fb0000>/g' target/linux/ramips/dts/mt7621_hiwifi_hc5962-spi.dts
cat >> ./target/linux/ramips/image/mt7621.mk <<EOF
define Device/hiwifi_hc5962-spi
  IMAGE_SIZE := 32128k
  DEVICE_VENDOR := HiWiFi
  DEVICE_MODEL := HC5962
  DEVICE_PACKAGES := kmod-mt7603 kmod-mt76x2 kmod-usb3 kmod-sdhci-mt7620 \
	kmod-usb-ledtrig-usbport wpad-openssl
endef
TARGET_DEVICES += hiwifi_hc5962-spi
EOF
sed -i 's/^[ \t]*//g' ./target/linux/ramips/image/mt7621.mk
# 
rm -f ./.config*
#touch ./.config
cp -f $GITHUB_WORKSPACE/config-4.14 ./.config
# 
# ==========固件定制部分============
# 
# 编译极路由B70固件:
cat >> .config <<EOF
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_hiwifi_hc5962-spi=y
EOF
# 常用LuCI插件选择: 添加外面的主题和应用，包是通过diy.sh 脚本进行下载。
cat >> .config <<EOF
CONFIG_PACKAGE_luci-app-wol=n
CONFIG_PACKAGE_luci-app-upnp=y
CONFIG_PACKAGE_luci-app-accesscontrol=n
CONFIG_PACKAGE_luci-app-filetransfer=y
CONFIG_PACKAGE_luci-app-unblockmusic=n
CONFIG_PACKAGE_luci-app-unblockneteasemusic-mini=n
CONFIG_PACKAGE_luci-app-vsftpd=y
CONFIG_PACKAGE_luci-app-vlmcsd=n
CONFIG_PACKAGE_luci-app-zerotier=n
CONFIG_PACKAGE_luci-app-koolproxyR=y
CONFIG_PACKAGE_luci-app-samba=y
CONFIG_PACKAGE_luci-theme-argon=y
EOF
# SSR Configuration
cat >> .config <<EOF
CONFIG_PACKAGE_luci-app-ssr-plus=n
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Trojan=n
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_Redsocks2=n
CONFIG_PACKAGE_luci-app-ssr-plus_INCLUDE_ShadowsocksR_Server=n
CONFIG_PACKAGE_luci-app-ssrserver-python=n
CONFIG_PACKAGE_shadowsocks-libev-ss-local=n
CONFIG_PACKAGE_shadowsocks-libev-ss-redir=n
CONFIG_PACKAGE_shadowsocksr-libev-alt=n
CONFIG_PACKAGE_shadowsocksr-libev-ssr-local=n
EOF
# 常用软件包:
cat >> .config <<EOF
CONFIG_PACKAGE_mount-utils=n
CONFIG_PACKAGE_automount=y
CONFIG_PACKAGE_autosamba=y
CONFIG_PACKAGE_kmod-fs-ext4=y
CONFIG_PACKAGE_curl=y
CONFIG_PACKAGE_htop=y
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_tree=y
CONFIG_PACKAGE_vim-fuller=y
CONFIG_PACKAGE_wget=y
EOF
# ===========固件定制部分结束=========
sed -i 's/^[ \t]*//g' ./.config
make defconfig
```

AmadeusGhost, 2021-03-16:
跟内核版本无关，你没有在 02_network 加上机子的 device config

mingwei900, 2021-03-17:
我遇到跟你一样的问题，说明下情况吧。
在02_network上添加
phicomm,k2p-nand)
		ucidef_add_switch "switch0" \
			"0:lan" "1:lan" "2:lan" "3:lan" "4:wan" "6@eth0"
和下面的
	phicomm,k2p-nand)
		lan_mac=$(cat /sys/class/net/eth0/address)
		wan_mac=$(mtd_get_mac_binary factory 0xe006)
		label_mac=$(mtd_get_mac_binary factory 0x4)

在mt7621.mk里面添加

define Device/phicomm_k2p-nand
  $(Device/uimage-lzma-loader)
  BLOCKSIZE := 128k
  PAGESIZE := 2048
  KERNEL_SIZE := 4096k
  IMAGE_SIZE := 124416k
  UBINIZE_OPTS := -E 5
  IMAGES += kernel1.bin rootfs0.bin
  IMAGE/kernel1.bin := append-kernel
  IMAGE/rootfs0.bin := append-ubi | check-size
  IMAGE/sysupgrade.bin := sysupgrade-tar | append-metadata
  DEVICE_VENDOR := Phicomm
  DEVICE_MODEL := K2P-NAND
  SUPPORTED_DEVICES += k2p-nand
  DEVICE_PACKAGES := kmod-mt7615d_dbdc kmod-usb3
endef
TARGET_DEVICES += phicomm_k2p-nand

和DTS文件都弄好了，
mt7621_phicomm_k2p-nand.dts

可以正常编译，就是无法有线连接，WIFI能连接但是没有网络下面没有 交换机 哪一项。

lon91ong, 2021-03-17:
@AmadeusGhost 我切换4.14内核的方法有什么遗漏吗? 为什么编译出来的还是5.4内核?

AmadeusGhost, 2021-03-17:
4.14 已经不支持了

lon91ong, 2021-03-17:
@AmadeusGhost 最后用4.14内核的Commit大概是什么时候的?

AmadeusGhost, 2021-03-17:
你换这个库 https://github.com/coolsnowwolf/openwrt 就行了，还有别一直@我。。。

lon91ong, 2021-03-17:
好吧, 多谢指点!

lon91ong, 2021-03-17:
刚刚又编译了一遍, 问题依旧, 我是这么改02_network的
查了查, 里面已经有hc5962的定义了, 猜测可能是我把设备型号改成hc5962-spi的原因, 所以用下面的命令在设备`hiwifi,hc5962`处加入了设备型号`hiwifi,hc5962-spi`
`sed -i 's/hc5962/hc5962|\\\n\thiwifi,hc5962-spi/g' ./target/linux/ramips/mt7621/base-files/etc/board.d/02_network`

结果是...问题依旧~!

mingwei900, 2021-03-17:
> 
> 
> 刚刚又编译了一遍, 问题依旧, 我是这么改02_network的
> 查了查, 里面已经有hc5962的定义了, 猜测可能是我把设备型号改成hc5962-spi的原因, 所以用下面的命令在设备`hiwifi,hc5962`处加入了设备型号`hiwifi,hc5962-spi`
> `sed -i 's/hc5962/hc5962|\\\n\thiwifi,hc5962-spi/g' ./target/linux/ramips/mt7621/base-files/etc/board.d/02_network`
> 
> 结果是...问题依旧~!

我跟你一样，我也这样改过，但是没效果。
我以前是这么改的可以用。不知道为什么现在不行了。

lon91ong, 2021-03-17:
> 你换这个库 https://github.com/coolsnowwolf/openwrt 就行了，还有别一直@我。。。

用这个库编译两次, 总是报错
`/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_hiwifi_hc5962-spi.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"`

AmadeusGhost, 2021-03-17:
> `/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_hiwifi_hc5962-spi.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"`

你 dts 写错了，4.14 上是 gpio0 /gpio1 之类的，比如：`gpios = <&gpio0 15 GPIO_ACTIVE_LOW>;`

lon91ong, 2021-03-18:
> > `/workdir/openwrt/build_dir/target-mipsel_24kc_musl/linux-ramips_mt7621/image-mt7621_hiwifi_hc5962-spi.dtb: ERROR (phandle_references): Reference to non-existent node or label "gpio"`
> 
> 你 dts 写错了，4.14 上是 gpio0 /gpio1 之类的，比如：`gpios = <&gpio0 15 GPIO_ACTIVE_LOW>;`

多谢指点, 不然小白得在坑边转悠好久!

今天又编译了两遍, 同配置一个4.14内核, 一个5.4内核
4.14内核没有毛病, 无线有线都正常. 5.4内核问题依旧, 有线无法连接

mingwei900, 2021-03-18:
这个问题我已经解决了。希望对你有帮助！
https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4093027&page=1&extra=#pid11038104

lon91ong, 2021-03-18:
> 这个问题我已经解决了。希望对你有帮助！
> https://www.right.com.cn/forum/forum.php?mod=viewthread&tid=4093027&page=1&extra=#pid11038104

我没有复制到Windows下编辑造，我直接用sed命令改，然后直接用Github Action云编译的，所以我们的问题只是表象相同，本质还是有差的

lon91ong, 2021-03-21:
折腾了几天, 好歹是整出来一个可用的带passwall的固件了
**总结:**
有线功能不好使是`./target/linux/ramips/mt7621/base-files/etc/board.d/02_network`配置问题的经验**仅**对4.14内核有效
ssrplus不知原因的无法启动，*可能是依赖的问题*
passwall+adbyby的基本需求固件需要至少32M闪存，16M闪存无解！

lon91ong, 2021-04-03:
新的折腾经验:
[02_network.patch](https://github.com/lon91ong/OpenWrt/blob/hiwifi_hc5962-spi/devices/hiwifi-hc5962/patches/02_network.patch)
原来的配置针对的是nand固件, spi闪存固件需要对应修改

closed 2021-04-05 by lon91ong