coolsnowwolf/lede #6555: ttyd使用zsh终端

weifile, 2021-03-18:
ttyd使用zsh终端

默认的ttyd使用的ash的shell,
修改那里能让其启动为zsh的shell呢？
或是启动ash能让其自动变为zsh呢。

jjit, 2021-03-19:
固件里修改这个文件 /etc/config/ttyd
文件里面的 option command '/usr/libexec/login.sh'
改成 option command 'zsh'
编译的话，改源码的这个文件
https://github.com/coolsnowwolf/packages/blob/master/utils/ttyd/files/ttyd.config

weifile, 2021-03-20:
改了之后用ttyd登陆后确实是zsh，
但是只是zsh，它没用应用Oh My Zsh.有加行命令让他运行的吗？

jjit, 2021-03-20:
有安装过oh-my-zsh吗，没有的话安装一下
SSH登入，运行
opkg update && opkg install wget unzip ca-certificates
sh -c "$(wget -O- https://raw.githubusercontent.com/felix-fly/openwrt-ohmyzsh/master/install.sh)"
修改一下路由这个文件 /usr/share/zsh/5.8/scripts/newuser
文件里最下面添加一行
source /root/.zshrc
然后重启一下

weifile, 2021-03-20:
可以进入zsh与显示oh-my-zsh.不过他进入后直接是/目录，不是家目录。另外cd ~无效。但是cd root可以。进入root/目录显示的是~
[image]

weifile, 2021-03-20:
如果ttyd用ash终端 。然后再zsh 又是正常的。能正常识别
[image]

closed 2021-06-17 by coolsnowwolf