coolsnowwolf/lede #6563: 麻烦把R4Swan口和lan口改回默认的。

MapesxGM, 2021-03-18:
麻烦把R4Swan口和lan口改回默认的，即eth0为wan口，eth1为lan口，目前编译的固件网口刚好反过来，还没找到方法修改。

azgh, 2021-03-18:
R2S好像也有同样的问题

ericsnow911, 2021-03-18:
这样可以有更好的性能表现，为什么还要改回去？

MapesxGM, 2021-03-18:
> 这样可以有更好的性能表现，为什么还要改回去？

的确是像你说的这样，只是强迫症。。。。。。

db-one, 2021-03-18:
我软路由也是lan是0口，wan是1口，我比较习惯把第一个口当lan口

ericsnow911, 2021-03-18:
> > 这样可以有更好的性能表现，为什么还要改回去？
> 
> 的确是像你说的这样，只是强迫症。。。。。。

那不就行了，别扭的话自己改源码

MapesxGM, 2021-03-18:
> > > 这样可以有更好的性能表现，为什么还要改回去？
> > 
> > 
> > 的确是像你说的这样，只是强迫症。。。。。。
> 
> 那不就行了，别扭的话自己改源码

这个版本试了好几次都不行

ericsnow911, 2021-03-18:
> > > > 这样可以有更好的性能表现，为什么还要改回去？
> > > 
> > > 
> > > 的确是像你说的这样，只是强迫症。。。。。。
> > 
> > 
> > 那不就行了，别扭的话自己改源码
> 
> 这个版本试了好几次都不行

按理来说修改target/linux/rockchip/armv8/base-files/etc/board.d/02_network 这个文件就行，
但是和我之前fork的源码对比，这个文件并没有涉及到网口定义的变动。

badcrazy, 2021-03-18:
我也发现最近的版本，软路由编译出来板载网卡是eth2，pcie网卡占了0和1，记得原来板载室eth0

skefy, 2021-03-19:
修改 /etv/config/network

``` shell
config interface 'lan'
        option type 'bridge'
        option ifname 'eth0' // 《---- 修改成自己想要的网口
        option proto 'static'
        option ipaddr '192.168.1.1'
        option netmask '255.255.255.0'
        option ip6assign '60'

config interface 'wan'
        option ifname 'eth1'  // 《---- 修改成自己想要的网口
        option proto 'dhcp' 
```

AmadeusGhost, 2021-03-19:
https://github.com/AmadeusGhost/lede/commit/083c2c53761e8df7f941dad2f8d22681a3fcd6e1 可以试试这个 commit 能不能解决问题

PR #6566 (2021-03-19): rockchip: add execute permission for network setup script
Fixes: f4b1b358fe8b ("rockchip: sync latest codes from ImmortalWrt (#6522)")

Fixes: #6534
Fixes: #6548
Fixes: #6563
closed 2021-03-19 by coolsnowwolf