coolsnowwolf/lede #6570: upnp打开却无法生效

coLATin, 2021-03-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [x] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：L大openwrt单路由系统。upnp功能已经打开
[image]

但是设备实际未生效
[image]

挂甜糖或者网心云也是upnp未生效

电信和移动双宽带叠加，未多拨，分流正常。

### (2) 路由器型号和固件版本

A：X86 软路由

### (3) 详细日志

A：

Ljzkirito, 2021-03-20:
个人给你两个建议：
1.勾选 **启用IGDv1模式** ，重启路由再试试。
2.试试miniupnpd 2.2.1版本，
```
cd lede
rm -fr feeds/packages/net/miniupnpd
svn co https://github.com/Ljzkirito/openwrt-packages/trunk/miniupnpd feeds/packages/net/miniupnpd
```

ihuaer, 2021-03-27:
> 个人给你两个建议：
> 1.勾选 **启用IGDv1模式** ，重启路由再试试。
> 大多数软件仍仅支持IGDv1，所以最好编译时不选IGDv2，保留勾选启用IGDv1模式
> 2.试试miniupnpd 2.2.1版本，
> 
> ```
> cd lede
> rm -fr feeds/packages/net/miniupnpd
> svn co https://github.com/Ljzkirito/openwrt-packages/trunk/miniupnpd feeds/packages/net/miniupnpd
> ```

感谢分享 试试看效果

xyskwjy, 2021-05-12:
> 个人给你两个建议：
> 1.勾选 **启用IGDv1模式** ，重启路由再试试。
> 大多数软件仍仅支持IGDv1，所以最好编译时不选IGDv2，保留勾选启用IGDv1模式
> 2.试试miniupnpd 2.2.1版本，
> 
> ```
> cd lede
> rm -fr feeds/packages/net/miniupnpd
> svn co https://github.com/Ljzkirito/openwrt-packages/trunk/miniupnpd feeds/packages/net/miniupnpd
> ```

根据命令，如果是去编绎的，没有效果，依然还是2.0版本的upnp

Ljzkirito, 2021-05-24:
> 根据命令，如果是云编绎的，没有效果，依然还是2.0版本的upnp

我自己也是云编绎的，在`diy-part2.sh`中保存以下命令：
```
rm -fr feeds/packages/net/miniupnpd
svn co https://github.com/Ljzkirito/openwrt-packages/trunk/miniupnpd feeds/packages/net/miniupnpd
```
可参考我的[GitHub Actions](https://github.com/Ljzkirito/Actions-Lean-RM2100/blob/main/diy-part2.sh#L29)

xyskwjy, 2021-05-25:
> 我自己也是云编绎的，在`diy-part2.sh`中保存以下命令：
>
> ```
> rm -fr feeds/packages/net/miniupnpd
> svn co https://github.com/Ljzkirito/openwrt-packages/trunk/miniupnpd feeds/packages/net/miniupnpd
> ```
>
> 可参考我的[GitHub [...]

感谢，upnp搞定了。
另外请教一下，内核报36010.340787] TCP: request_sock_TCP: Possible SYN flooding on port 5000. Sending cookies.  Check SNMP counters.
这是攻击还是并发连接数太多了？

closed 2021-06-17 by coolsnowwolf

shunlongyang, 2021-12-09:
> 感谢，upnp搞定了。 另外请教一下，内核报36010.340787] TCP: request_sock_TCP: Possible SYN flooding on port 5000. Sending cookies. Check SNMP counters. 这是攻击还是并发连接数太多了？

请问一下这个upnp搞定了是按上面说的换了miniupnpd 2.2.1版本，还是改成了IGDv1好的啊