coolsnowwolf/lede #6571: 编译shadowsocks-rust 报错

JohyC, 2021-03-20:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ ] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：
编译最新固件报错 
报错位置在新添加的 shadowsocks-rust

### (2) 路由器型号和固件版本

A：
型号：Xiaomi Mi Router 3G
固件版本：OpenWrt R21.2.1 / LuCI Master (git-21.035.34289-13f96d1)

### (3) 详细日志

A：
/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocks-rust-1.9.2/sslocal
                       Ultimate Packer for eXecutables
                          Copyright (C) 1996 - 2018
UPX 3.95        Markus Oberhumer, Laszlo Molnar & John Reiser   Aug 26th 2018

        File size         Ratio      Format      Name
   --------------------   ------   -----------   -----------
upx: /workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocks-rust-1.9.2/sslocal: CantPackException: bad e_shstrndx

Packed 0 files.
make[3]: *** [/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocks-rust-1.9.2/.built] Error 1
Makefile:110: recipe for target '/workdir/openwrt/build_dir/target-mipsel_24kc_musl/shadowsocks-rust-1.9.2/.built' failed
make[2]: *** [package/feeds/helloworld/shadowsocks-rust/compile] Error 2
make[3]: Leaving directory '/workdir/openwrt/feeds/helloworld/shadowsocks-rust'
make[1]: *** [/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile] Error 2
time: package/feeds/helloworld/shadowsocks-rust/compile#0.14#0.04#0.18
package/Makefile:111: recipe for target 'package/feeds/helloworld/shadowsocks-rust/compile' failed
make[2]: Leaving directory '/workdir/openwrt'
package/Makefile:107: recipe for target '/workdir/openwrt/staging_dir/target-mipsel_24kc_musl/stamp/.package_compile' failed
make[1]: Leaving directory '/workdir/openwrt'
make: *** [world] Error 2
/workdir/openwrt/include/toplevel.mk:222: recipe for target 'world' failed
Error: Process completed with exit code 2.

closed 2021-03-20 by JohyC

AmadeusGhost, 2021-03-20:
把 shadowsocks-rust 的 upx 关掉即可

JohyC, 2021-03-20:
> 把 shadowsocks-rust 的 upx 关掉即可

多谢！我现在已经取消编译 shadowsocks-rust  了。