coolsnowwolf/lede #6580: nginx.conf不见了,

lxwq, 2021-03-22:
反馈bug/问题模板，提建议请删除

## 1.关于你要提交的问题

Q：是否搜索了issue (使用 "x" 选择)
* [ X] 没有类似的issue

## 2. 详细叙述

### (1) 具体问题

A：/etc/nginx/nginx.conf文件不见了,使用了一个uci.conf的软链接到/var/lib/nginx/uci.conf,里面的内容与以前的nginx.conf差不多,
但在编辑uci.conf文件并保存,重启nginx,编辑过的uci.conf就会被恢复,这是什么状况啊?

### (2) 路由器型号和固件版本

A：X86_64

### (3) 详细日志

A：

LGA1150, 2021-03-22:
https://openwrt.org/docs/guide-user/services/webserver/nginx

lxwq, 2021-03-22:
> https://openwrt.org/docs/guide-user/services/webserver/nginx

这个是版本升级后,配置文件改变了吗?
记得以前没有/etc/config/nginx这个文件,
我2月26号编译的好像还没有出现/etc/nginx/uci.conf之类的文件,

lxwq, 2021-03-23:
> https://openwrt.org/docs/guide-user/services/webserver/nginx

看着文件档搞了搞,只能HTTPS访问,而且还出403错误,
头大

closed 2021-06-17 by coolsnowwolf